linrunner/TLP #539: Thinkpad x270 dual battery calibration aborted

angeloc, 2020-10-25:
```
$ sudo tlp recalibrate BAT1
Setting temporary charge thresholds for BAT1:
  start =  96 (no change)
  stop  = 100 (no change)
Initiating discharge of battery BAT1 ...............
Error: discharge BAT1 malfunction.
Battery recalibration aborted.
```

```$ sudo tlp-stat --psup
--- TLP 1.4.0-alpha.0 --------------------------------------------

+++ Power supply diagnostic
/sys/class/power_supply/AC/type:Mains
/sys/class/power_supply/AC/online:1
/sys/class/power_supply/AC/device/path:\_SB_.PCI0.LPCB.EC__.AC__
/sys/class/power_supply/BAT0/type:Battery
/sys/class/power_supply/BAT0/present:1
/sys/class/power_supply/BAT0/status:Unknown
/sys/class/power_supply/BAT0/device/path:\_SB_.PCI0.LPCB.EC__.BAT0
/sys/class/power_supply/BAT1/type:Battery
/sys/class/power_supply/BAT1/present:1
/sys/class/power_supply/BAT1/status:Charging
/sys/class/power_supply/BAT1/device/path:\_SB_.PCI0.LPCB.EC__.BAT1
```
```
sudo tlp-stat -T
--- TLP 1.4.0-alpha.0 --------------------------------------------

-- Logs begin at Thu 2020-10-22 14:34:00 CEST, end at Sun 2020-10-25 15:51:01 CET. --
ott 25 12:19:42 angelo-x270 tlp[1326]: Applying power save settings...done.
ott 25 12:19:43 angelo-x270 tlp[1326]: Setting battery charge thresholds...done.
```

linrunner, 2020-10-25:
Hi,

you may take a little more effort with the problem description, e.g. the outputs as described in the [issue template](https://github.com/linrunner/TLP/blob/master/.github/ISSUE_TEMPLATE/bug_report.md).

    tlp-stat -c -s -b

Also important when testing an alpha version: is this a regression? Please try again with the release version 1.3.1.

angeloc, 2020-10-25:
I tested with the alpha version because the 1.3.1 was exhibiting the same problem.

Log attached
[tlp-c-s-b.log](https://github.com/linrunner/TLP/files/5436134/tlp-c-s-b.log)

linrunner, 2020-10-26:
1. Does `tlp recalibrate BAT0` work?
2. Why did you try to recalibrate BAT1? The output indicates a defect – which would explain the "malfunction".

angeloc, 2020-10-26:
>     1. Does `tlp recalibrate BAT0` work?
I haven0t tried, I'll do and I'll post the results.
>     2. Why did you try to recalibrate BAT1? The output indicates a defect (11% capacity) – which would explain the "malfunction".
Well, was exactly to have a real indication of battery life because the battery looks like it's depleted. Are you saying that is its normal behavior?

linrunner, 2020-10-26:
It suffices to initiate the recalibratio, wait until it starts and then abort via Ctrl+C.

Defective batteries may show any kind of strange and unpredictable behavior, which seems to be the case here. 

That's why the [TLP Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md) says

>  **What not to report:** [...} Hardware issues e.g. worn out or malfunctioning batteries

Please ask for advice in a hardware oriented forum.

linrunner, 2020-11-01:
Did you check with BAT0 yet?

linrunner, 2020-12-17:
Since no further information is forthcoming, I will close the issue.

closed 2020-12-17 by linrunner