linrunner/TLP #540: STOP_CHARGE_THRESH_BAT0 and /sys/class/power_supply/BAT0/charge_stop_threshold do not match

juha-h, 2020-11-01:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

STOP_CHARGE_THRESH_BAT0 and /sys/class/power_supply/BAT0/charge_stop_threshold do not match:
```
# tlp-stat -c
...
/etc/tlp.conf L0425: START_CHARGE_THRESH_BAT0="75"
/etc/tlp.conf L0426: STOP_CHARGE_THRESH_BAT0="80"

# tlp-stat -b
...
/sys/class/power_supply/BAT0/charge_start_threshold         =     75 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     75 [%]
```

**Expected behavior**

I would expect this:
```
# tlp-stat -b
...
/sys/class/power_supply/BAT0/charge_start_threshold         =     75 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     80 [%]
```

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?

Yes.

2. Actions to reproduce the behaviour

Reboot host with the shown config.

3. Shell commands entered and their output

See above.

4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**

```
# tlp-stat
--- TLP 1.3.1 --------------------------------------------

+++ Configured Settings:
defaults.conf L0004: TLP_ENABLE="1"
defaults.conf L0005: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0006: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0008: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0010: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0012: SCHED_POWERSAVE_ON_AC="0"
defaults.conf L0013: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0014: NMI_WATCHDOG="0"
defaults.conf L0015: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0016: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0017: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0018: DISK_IOSCHED="keep keep"
defaults.conf L0019: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0020: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
defaults.conf L0021: AHCI_RUNTIME_PM_TIMEOUT="15"
[... 182 lines cut ...]

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 01AV487
/sys/class/power_supply/BAT0/cycle_count                    =     45
/sys/class/power_supply/BAT0/energy_full_design             =  42010 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  43080 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  40120 [mWh]
/sys/class/power_supply/BAT0/power_now                      =   3650 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

/sys/class/power_supply/BAT0/charge_start_threshold         =     75 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     75 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   93.1 [%]
Capacity                                                    =  102.5 [%]

+++ Recommendations
* Install smartmontools for disk drive health info
```

**Additional context**

Add any other context about the problem here.

linrunner, 2020-11-01:
Hi, 

your problem is well known and therefore mentioned in the [FAQ](https://linrunner.de/tlp/faq/battery.html#charge-thresholds-shown-by-tlp-stat-b-do-not-correspond-to-the-configured-ones).

closed 2020-11-01 by linrunner

juha-h, 2020-11-01:
OK, the issue then also exists in ThinkPad 11e (5th Gen).