linrunner/TLP #541: Cannot enable autosuspend on USB card reader through USB whitelist

phrxmd, 2020-11-02:
**Describe the bug**

My notebook (Lenovo Thinkpad X1 Yoga 3rd gen) has a built-in Realtek USB3 MicroSD card reader (USB ID `0bda:0328`) that I never use. My system is on OpenSUSE Tumbleweed.

TLP reports the card reader as `control=on`. In order to enable autosuspend on the card reader, I have whitelisted it with `USB_WHITELIST="0bda:0328"` in my `/etc/tlp.conf`. Nevertheless, `tlp-stat` still shows the reader as `control=on`. 

**Expected behavior**

`tlp-stat` should show the reader as `control=auto`

**To Reproduce**

The behaviour is the same on AC and battery power.

Steps to reproduce:
1. Put `USB_WHITELIST="0bda:0328"` in `/etc/tlp.conf`
2. Run `tlp-stat`, check for `0bda:0328` in the `+++ USB` section

- tlp-stat output on AC: https://gist.github.com/phrxmd/d2cdf7b1904bec09901ca167df7e78b7
- tlp-stat output on battery: https://gist.github.com/phrxmd/357e20445cc0fb6b4053097d5b45a4a2

**Additional context**

`powertop` also reports the card reader, and the (empty) `/dev/sdb` device attached to it, as "Bad" with autosuspend disabled.

linrunner, 2020-11-02:
Hi,

can you switch to "auto" with powertop? If not, it's a kernel driver or hardware issue.

phrxmd, 2020-11-02:
Yes, I can - I switched it in `powertop`, now there it shows as "Good", and `tlp-stat` now reports `control=auto` ([gist](https://gist.github.com/phrxmd/1e1d8a4f78a56322d3405721b0bd0fe0))

linrunner, 2020-11-03:
TLP handles USB devices with the help of udev events. I guess the kernel prevents writing `control` for the device in some early boot stage. Please enable [trace mode](https://linrunner.de/tlp/support/index.html#trace-mode) via configuration:

    TLP_DEBUG="ps run udev usb"

Reboot and show the trace output:

   tlp-stat -T

ps. on my X1C6 here (very similar hardware) with kernel 5.9 the reader device behaves as expected.
ps2. USB_WHITELIST is unnecessary, card readers are not blacklisted implicitly.

linrunner, 2020-11-03:
I just remembered, since you use OpenSUSE: 

https://github.com/linrunner/TLP/issues/537
https://bugzilla.opensuse.org/show_bug.cgi?id=1177313

Make shure the latest tlp package is installed, otherwise udev events won't work at all.

phrxmd, 2020-11-06:
Thank you. I put `TLP_DEBUG="ps run udev usb"` in my `/etc/tlp.conf` and rebooted. Here's the `tlp-stat -T` output:

```
--- TLP 1.3.1 --------------------------------------------

-- Logs begin at Fri 2020-11-06 09:24:18 CET, end at Fri 2020-11-06 10:24:17 CET. --
Nov 06 09:24:20 calypso tlp[723]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 06 09:24:20 calypso tlp[713]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 06 09:24:20 calypso tlp[723]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Nov 06 09:24:20 calypso tlp[713]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Nov 06 09:24:20 calypso tlp[713]: power_source=ac
Nov 06 09:24:20 calypso tlp[723]: power_source=ac
Nov 06 09:24:20 calypso tlp[723]: manual_mode=none
Nov 06 09:24:20 calypso tlp[713]: manual_mode=none
Nov 06 09:24:20 calypso tlp[713]: power_mode=ac
Nov 06 09:24:20 calypso tlp[723]: power_mode=ac
Nov 06 09:24:20 calypso tlp[713]: compare_and_save_power_state(0).different: old=
Nov 06 09:24:22 calypso tlp[713]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpdgowLc -> /run/tlp/run.conf
Nov 06 09:24:29 calypso tlp[1792]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 06 09:24:29 calypso tlp[1792]: +++ init start (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Nov 06 09:24:29 calypso tlp[1792]: power_source=ac
Nov 06 09:24:29 calypso tlp[1792]: manual_mode=none
Nov 06 09:24:29 calypso tlp[1792]: power_mode=ac
Nov 06 09:24:30 calypso tlp[1792]: compare_and_save_power_state(0).equal
Nov 06 09:24:30 calypso tlp[1792]: Applying power save settings...done.
Nov 06 09:24:30 calypso tlp[1792]: Setting battery charge thresholds...done.
Nov 06 09:24:30 calypso tlp[1792]: Disabling radios: wwan.
Nov 06 09:24:30 calypso tlp[1792]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpWAby18 -> /run/tlp/run.conf
Nov 06 09:25:01 calypso tlp[3336]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 06 09:25:01 calypso tlp[3336]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Nov 06 09:25:01 calypso tlp[3336]: power_source=ac
Nov 06 09:25:02 calypso tlp[3336]: manual_mode=none
Nov 06 09:25:02 calypso tlp[3336]: power_mode=ac
Nov 06 09:25:02 calypso tlp[3336]: compare_and_save_power_state(0).equal
Nov 06 09:25:52 calypso tlp[4330]: get_sys_power_supply(AC).ac_offline_remembered
Nov 06 09:25:52 calypso tlp[4330]: get_sys_power_supply(BAT0).bat_discharging: syspwr=1
Nov 06 09:25:52 calypso tlp[4330]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Nov 06 09:25:52 calypso tlp[4330]: power_source=bat
Nov 06 09:25:52 calypso tlp[4330]: manual_mode=none
Nov 06 09:25:52 calypso tlp[4330]: power_mode=bat
Nov 06 09:25:52 calypso tlp[4330]: compare_and_save_power_state(1).different: old=0
Nov 06 09:25:52 calypso tlp[4362]: get_sys_power_supply(AC).ac_offline_remembered
Nov 06 09:25:52 calypso tlp[4362]: get_sys_power_supply(BAT0).bat_discharging: syspwr=1
Nov 06 09:25:52 calypso tlp[4362]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Nov 06 09:25:53 calypso tlp[4362]: power_source=bat
Nov 06 09:25:53 calypso tlp[4362]: manual_mode=none
Nov 06 09:25:53 calypso tlp[4362]: power_mode=bat
Nov 06 09:25:53 calypso tlp[4330]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpECt14s -> /run/tlp/run.conf
Nov 06 09:25:58 calypso tlp[4587]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 06 09:25:59 calypso tlp[4587]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Nov 06 09:25:59 calypso tlp[4587]: power_source=ac
Nov 06 09:25:59 calypso tlp[4600]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 06 09:25:59 calypso tlp[4587]: manual_mode=none
Nov 06 09:25:59 calypso tlp[4600]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Nov 06 09:25:59 calypso tlp[4587]: power_mode=ac
Nov 06 09:25:59 calypso tlp[4600]: power_source=ac
Nov 06 09:25:59 calypso tlp[4600]: manual_mode=none
Nov 06 09:25:59 calypso tlp[4587]: compare_and_save_power_state(0).different: old=1
Nov 06 09:25:59 calypso tlp[4600]: power_mode=ac
Nov 06 09:26:00 calypso tlp[4587]: save_runconf.ok: /run/tlp/tlp-run.conf_tmp8dhAF6 -> /run/tlp/run.conf
```

It switches from battery to AC mode and vice versa just fine, so I think #537 does not affect me. I use TLP 1.3.1 with kernel 5.9.1.

linrunner, 2020-11-08:
The trace reveals no udev events at all. Strange.

Are there events in the trace when you (re)plug some USB device like your trackball?

I suggest you try to change to USB batch mode by configuring:

    X_TLP_USB_MODE=1     

Reboot and show the trace again.

linrunner, 2020-12-16:
I'm still missing your feedback on the workaround.

phrxmd, 2020-12-20:
Hi! Sorry for answering late, I had some work to do and it slipped my mind.

I've put the line you suggested (`X_TLP_USB_MODE=1`) in my config, now the card reader is shown as `auto`, and the trace contains events:

```
# tlp-stat
--- TLP 1.3.1 --------------------------------------------

+++ Configured Settings:
[...}
/etc/tlp.conf L0030: TLP_DEBUG="ps run udev usb"
/etc/tlp.conf L0032: X_TLP_USB_MODE="1"
[...]

+++ USB
Autosuspend         = enabled
Device whitelist    = 0bda:0328
Device blacklist    = (not configured)
Bluetooth blacklist = disabled
Phone blacklist     = disabled
WWAN blacklist      = disabled

Bus 004 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 3.0 root hub (hub)
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 002 Device 002 ID 0bda:0328 control = auto, autosuspend_delay_ms = 2000 -- Realtek Semiconductor Corp. USB3.0-CRW (usb-storage)
Bus 002 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 3.0 root hub (hub)
Bus 001 Device 006 ID 04f2:b61e control = auto, autosuspend_delay_ms = 2000 -- Chicony Electronics Co., Ltd Integrated Camera (uvcvideo)
Bus 001 Device 004 ID 8087:0a2b control = auto, autosuspend_delay_ms = 2000 -- Intel Corp. Bluetooth wireless interface (btusb)
Bus 001 Device 002 ID 056e:010d control = on,   autosuspend_delay_ms = 2000 -- Elecom Co., Ltd M-HT1DRBK HUGE Wireless Optical TrackBall (usbhid)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
```

If I do a `tlp-stat -T`, I get a trace that looks like this:

```
# tlp-stat -T
# tlp-stat -T
--- TLP 1.3.1 --------------------------------------------

-- Logs begin at Sun 2020-12-20 13:35:14 CET, end at Sun 2020-12-20 19:28:46 CET. --
Dez 20 13:35:17 calypso tlp[1039]: get_sys_power_supply(AC).ac_online: syspwr=0
Dez 20 13:35:17 calypso tlp[1047]: get_sys_power_supply(AC).ac_online: syspwr=0
Dez 20 13:35:17 calypso tlp[1079]: get_sys_power_supply(AC).ac_online: syspwr=0
Dez 20 13:35:17 calypso tlp[1039]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 20 13:35:17 calypso tlp[1047]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 20 13:35:17 calypso tlp[1079]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 20 13:35:17 calypso tlp[1047]: power_source=ac
Dez 20 13:35:17 calypso tlp[1039]: power_source=ac
Dez 20 13:35:17 calypso tlp[1079]: power_source=ac
Dez 20 13:35:17 calypso tlp[1039]: manual_mode=none
Dez 20 13:35:17 calypso tlp[1047]: manual_mode=none
Dez 20 13:35:17 calypso tlp[1079]: manual_mode=none
Dez 20 13:35:17 calypso tlp[1039]: power_mode=ac
Dez 20 13:35:17 calypso tlp[1047]: power_mode=ac
Dez 20 13:35:17 calypso tlp[1079]: power_mode=ac
Dez 20 13:35:17 calypso tlp[1039]: compare_and_save_power_state(0).different: old=
Dez 20 13:35:17 calypso tlp[1223]: get_sys_power_supply(AC).ac_online: syspwr=0
Dez 20 13:35:17 calypso tlp[1223]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 20 13:35:17 calypso tlp[1223]: power_source=ac
Dez 20 13:35:17 calypso tlp[1223]: manual_mode=none
Dez 20 13:35:17 calypso tlp[1223]: power_mode=ac
Dez 20 13:35:18 calypso tlp[1039]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpNk3WdV -> /run/tlp/run.conf
Dez 20 13:35:24 calypso tlp[1680]: get_sys_power_supply(AC).ac_online: syspwr=0
Dez 20 13:35:24 calypso tlp[1680]: +++ init start (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 20 13:35:24 calypso tlp[1680]: power_source=ac
Dez 20 13:35:24 calypso tlp[1680]: manual_mode=none
Dez 20 13:35:24 calypso tlp[1680]: power_mode=ac
Dez 20 13:35:25 calypso tlp[1680]: compare_and_save_power_state(0).equal
Dez 20 13:35:25 calypso tlp[1680]: usb_suspend_device.batch.on_hid_black.no_change: Bus 1 Dev 8 ID 056a:5146 /sys/bus/usb/devices/1-10 []
Dez 20 13:35:25 calypso tlp[1680]: usb_suspend_device.batch.on_hid_black.no_change: Bus 1 Dev 2 ID 056e:010d /sys/bus/usb/devices/1-2 []
[... 24 lines cut ...]
Dez 20 19:24:36 calypso tlp[18316]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 20 19:24:36 calypso tlp[18329]: get_sys_power_supply(BAT0).bat_discharging: syspwr=1
Dez 20 19:24:36 calypso tlp[18316]: power_source=bat
Dez 20 19:24:36 calypso tlp[18329]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 20 19:24:36 calypso tlp[18316]: manual_mode=none
Dez 20 19:24:36 calypso tlp[18329]: power_source=bat
Dez 20 19:24:36 calypso tlp[18316]: power_mode=bat
Dez 20 19:24:36 calypso tlp[18329]: manual_mode=none
Dez 20 19:24:36 calypso tlp[18329]: power_mode=bat
Dez 20 19:24:36 calypso tlp[18316]: compare_and_save_power_state(1).different: old=0
Dez 20 19:24:36 calypso tlp[18374]: get_sys_power_supply(AC).ac_offline_remembered
Dez 20 19:24:36 calypso tlp[18374]: get_sys_power_supply(BAT0).bat_discharging: syspwr=1
Dez 20 19:24:37 calypso tlp[18374]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 20 19:24:37 calypso tlp[18374]: power_source=bat
Dez 20 19:24:37 calypso tlp[18374]: manual_mode=none
Dez 20 19:24:37 calypso tlp[18374]: power_mode=bat
Dez 20 19:24:37 calypso tlp[18316]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpMZiY1c -> /run/tlp/run.conf
Dez 20 19:24:40 calypso tlp[18635]: get_sys_power_supply(AC).ac_offline_remembered
Dez 20 19:24:40 calypso tlp[18635]: get_sys_power_supply(BAT0).bat_discharging: syspwr=1
Dez 20 19:24:40 calypso tlp[18635]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 20 19:24:40 calypso tlp[18635]: power_source=bat
Dez 20 19:24:41 calypso tlp[18635]: manual_mode=none
Dez 20 19:24:41 calypso tlp[18635]: power_mode=bat
Dez 20 19:24:41 calypso tlp[18635]: compare_and_save_power_state(1).equal
```

phrxmd, 2020-12-20:
Note that it works only in batch mode. If I comment out the `X_TLP_USB_MODE=1` line, I get `control=auto` for the card reader in the `tlp-stat` output.

linrunner, 2020-12-27:
I still wonder about the missing udev events at boot without `X_TLP_USB_MODE=1`. Is there a trace entry when you plug in any USB device?

phrxmd, 2020-12-28:
No. Here is the `tlp-stat -T` output right after unplugging my USB trackball, and plugging in an USB-C external drive:

```
# tlp-stat -T
--- TLP 1.3.1 --------------------------------------------

-- Logs begin at Sun 2020-12-27 10:48:30 CET, end at Mon 2020-12-28 16:57:45 CET. --
Dez 27 10:48:32 calypso tlp[926]: get_sys_power_supply(AC).ac_online: syspwr=0
Dez 27 10:48:32 calypso tlp[935]: get_sys_power_supply(AC).ac_online: syspwr=0
Dez 27 10:48:32 calypso tlp[974]: get_sys_power_supply(AC).ac_online: syspwr=0
Dez 27 10:48:32 calypso tlp[926]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 27 10:48:32 calypso tlp[935]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 27 10:48:32 calypso tlp[974]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 27 10:48:32 calypso tlp[926]: power_source=ac
Dez 27 10:48:33 calypso tlp[935]: power_source=ac
Dez 27 10:48:33 calypso tlp[926]: manual_mode=none
Dez 27 10:48:33 calypso tlp[935]: manual_mode=none
Dez 27 10:48:33 calypso tlp[974]: power_source=ac
Dez 27 10:48:33 calypso tlp[926]: power_mode=ac
Dez 27 10:48:33 calypso tlp[935]: power_mode=ac
Dez 27 10:48:33 calypso tlp[974]: manual_mode=none
Dez 27 10:48:33 calypso tlp[974]: power_mode=ac
Dez 27 10:48:33 calypso tlp[926]: compare_and_save_power_state(0).different: old=
Dez 27 10:48:33 calypso tlp[1075]: get_sys_power_supply(AC).ac_online: syspwr=0
Dez 27 10:48:33 calypso tlp[1075]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 27 10:48:33 calypso tlp[1075]: power_source=ac
Dez 27 10:48:33 calypso tlp[1075]: manual_mode=none
Dez 27 10:48:33 calypso tlp[1075]: power_mode=ac
Dez 27 10:48:34 calypso tlp[926]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpHlx3qP -> /run/tlp/run.conf
Dez 27 10:49:08 calypso tlp[3136]: get_sys_power_supply(AC).ac_offline_remembered
Dez 27 10:49:08 calypso tlp[3136]: get_sys_power_supply(BAT0).bat_discharging: syspwr=1
Dez 27 10:49:08 calypso tlp[3149]: get_sys_power_supply(AC).ac_offline_remembered
Dez 27 10:49:08 calypso tlp[3136]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 27 10:49:08 calypso tlp[3149]: get_sys_power_supply(BAT0).bat_discharging: syspwr=1
Dez 27 10:49:09 calypso tlp[3136]: power_source=bat
Dez 27 10:49:09 calypso tlp[3149]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 27 10:49:09 calypso tlp[3136]: manual_mode=none
Dez 27 10:49:09 calypso tlp[3149]: power_source=bat
Dez 27 10:49:09 calypso tlp[3136]: power_mode=bat
Dez 27 10:49:09 calypso tlp[3149]: manual_mode=none
Dez 27 10:49:09 calypso tlp[3136]: compare_and_save_power_state(1).different: old=0
Dez 27 10:49:09 calypso tlp[3149]: power_mode=bat
Dez 27 10:49:09 calypso tlp[3194]: get_sys_power_supply(AC).ac_offline_remembered
Dez 27 10:49:09 calypso tlp[3194]: get_sys_power_supply(BAT0).bat_discharging: syspwr=1
Dez 27 10:49:09 calypso tlp[3194]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 27 10:49:09 calypso tlp[3194]: power_source=bat
Dez 27 10:49:09 calypso tlp[3194]: manual_mode=none
Dez 27 10:49:09 calypso tlp[3194]: power_mode=bat
Dez 27 10:49:10 calypso tlp[3136]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpp3WIdg -> /run/tlp/run.conf
Dez 27 10:49:13 calypso tlp[3459]: get_sys_power_supply(AC).ac_online: syspwr=0
Dez 27 10:49:13 calypso tlp[3459]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 27 10:49:13 calypso tlp[3469]: get_sys_power_supply(AC).ac_online: syspwr=0
Dez 27 10:49:13 calypso tlp[3459]: power_source=ac
Dez 27 10:49:13 calypso tlp[3469]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 27 10:49:13 calypso tlp[3459]: manual_mode=none
Dez 27 10:49:13 calypso tlp[3469]: power_source=ac
Dez 27 10:49:13 calypso tlp[3459]: power_mode=ac
Dez 27 10:49:13 calypso tlp[3469]: manual_mode=none
Dez 27 10:49:13 calypso tlp[3459]: compare_and_save_power_state(0).different: old=1
Dez 27 10:49:13 calypso tlp[3469]: power_mode=ac
Dez 27 10:49:13 calypso tlp[3490]: get_sys_power_supply(AC).ac_online: syspwr=0
Dez 27 10:49:13 calypso tlp[3490]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 27 10:49:13 calypso tlp[3490]: power_source=ac
Dez 27 10:49:13 calypso tlp[3490]: manual_mode=none
Dez 27 10:49:13 calypso tlp[3490]: power_mode=ac
Dez 27 10:49:14 calypso tlp[3459]: save_runconf.ok: /run/tlp/tlp-run.conf_tmppXIgdN -> /run/tlp/run.conf
Dez 27 10:49:15 calypso tlp[3651]: get_sys_power_supply(AC).ac_online: syspwr=0
Dez 27 10:49:15 calypso tlp[3651]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 27 10:49:15 calypso tlp[3651]: power_source=ac
Dez 27 10:49:15 calypso tlp[3651]: manual_mode=none
[... 70 lines cut ...]
Dez 27 22:40:18 calypso tlp[4815]: compare_and_save_power_state(1).equal
Dez 27 23:34:12 calypso tlp[5354]: get_sys_power_supply(AC).ac_offline_remembered
Dez 27 23:34:12 calypso tlp[5354]: get_sys_power_supply(BAT0).bat_discharging: syspwr=1
Dez 27 23:34:12 calypso tlp[5354]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 27 23:34:12 calypso tlp[5354]: power_source=bat
Dez 27 23:34:12 calypso tlp[5354]: manual_mode=none
Dez 27 23:34:12 calypso tlp[5354]: power_mode=bat
Dez 27 23:34:13 calypso tlp[5354]: compare_and_save_power_state(1).equal
Dez 28 10:49:58 calypso tlp[5733]: get_sys_power_supply(AC).ac_online: syspwr=0
Dez 28 10:49:58 calypso tlp[5733]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Dez 28 10:49:58 calypso tlp[5733]: power_source=ac
Dez 28 10:49:58 calypso tlp[5733]: manual_mode=none
Dez 28 10:49:58 calypso tlp[5733]: power_mode=ac
Dez 28 10:49:58 calypso tlp[5733]: compare_and_save_power_state(0).different: old=1
Dez 28 10:49:59 calypso tlp[5733]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpbqF3Vx -> /run/tlp/run.conf
```
I currently have this in my `/etc/tlp.conf`:
```
# Debugging
# Trace mode for USB to debug card reader quirks
TLP_DEBUG="ps run udev usb"
# USB batch mode
X_TLP_USB_MODE=1
```
Should there be more USB-related output visible?

linrunner, 2020-12-29:
There should be a line for every USB plugin event. By now i am quite sure that you have installed the defective package from #537.

Please show:

    locate tlp.rules

linrunner, 2020-12-29:
Additional output:

    udevadm test -a add /sys/bus/usb/devices/usb1

phrxmd, 2021-01-02:
I doubt it's the same issue as #537, because AC/BAT is working just fine, and that bug was fixed in the openSUSE repositories in October. But just in case here is the TLP version I'm using:
```
# zypper if tlp
Repository-Daten werden geladen...
Installierte Pakete werden gelesen...

Informationen zu Paket tlp:
---------------------------
Repository         : openSUSE:Factory
Name               : tlp
Version            : 1.3.1-2.1
Arch               : noarch
Anbieter           : openSUSE
Installierte Größe : 314,7 KiB
Installiert        : Ja
Status             : aktuell
Quellpaket         : tlp-1.3.1-2.1.src
Zusammenfassung    : Tools to save battery power on laptops
Beschreibung       : 
    TLP implements advanced power management for Linux.
    TLP is a pure command line tool with automated background tasks.
    It does not contain a GUI.
```
Here's the `locate tlp.rules` output:
```
> locate tlp.rules
/usr/lib/udev/rules.d/85-tlp.rules
```
And here's the output from `udevadm test -a add /sys/bus/usb/devices/1-7`:
```
> udevadm test -a add /sys/bus/usb/devices/1-7
This program is for debugging only, it does not run any program
specified by a RUN key. It may show incorrect results, because
some values may be different, or not available at a simulation run.

Load module index
Parsed configuration file /usr/lib/systemd/network/99-default.link
Created link configuration context.
Reading rules file: /usr/lib/udev/rules.d/00-dont-del-part-nodes.rules
Reading rules file: /usr/lib/udev/rules.d/01-md-raid-creating.rules
Reading rules file: /usr/lib/udev/rules.d/10-dm.rules
Reading rules file: /usr/lib/udev/rules.d/11-dm-lvm.rules
Reading rules file: /usr/lib/udev/rules.d/11-dm-mpath.rules
Reading rules file: /usr/lib/udev/rules.d/11-dm-parts.rules
Reading rules file: /usr/lib/udev/rules.d/13-dm-disk.rules
Reading rules file: /usr/lib/udev/rules.d/39-usbmuxd.rules
Reading rules file: /usr/lib/udev/rules.d/40-libgphoto2.rules
Reading rules file: /usr/lib/udev/rules.d/40-psm.rules
Reading rules file: /usr/lib/udev/rules.d/40-usb-blacklist.rules
Reading rules file: /usr/lib/udev/rules.d/40-usb-media-players.rules
Reading rules file: /usr/lib/udev/rules.d/40-usb_modeswitch.rules
Reading rules file: /usr/lib/udev/rules.d/42-usb-hid-pm.rules
Reading rules file: /usr/lib/udev/rules.d/50-avrdude.rules
Reading rules file: /usr/lib/udev/rules.d/50-iscsi-firmware-login.rules
Reading rules file: /usr/lib/udev/rules.d/50-udev-default.rules
Reading rules file: /usr/lib/udev/rules.d/54-before-scsi-sg3_id.rules
Reading rules file: /usr/lib/udev/rules.d/55-libsane.rules
Reading rules file: /usr/lib/udev/rules.d/55-scsi-sg3_id.rules
Reading rules file: /usr/lib/udev/rules.d/56-hpmud.rules
Reading rules file: /usr/lib/udev/rules.d/56-multipath.rules
Reading rules file: /usr/lib/udev/rules.d/56-sane-backends-autoconfig.rules
Reading rules file: /usr/lib/udev/rules.d/58-scsi-sg3_symlink.rules
Reading rules file: /usr/lib/udev/rules.d/59-fc-wwpn-id.rules
Reading rules file: /usr/lib/udev/rules.d/60-autosuspend.rules
Reading rules file: /usr/lib/udev/rules.d/60-block.rules
Reading rules file: /usr/lib/udev/rules.d/60-cdrom_id.rules
Reading rules file: /usr/lib/udev/rules.d/60-drm.rules
Reading rules file: /usr/lib/udev/rules.d/60-evdev.rules
Reading rules file: /usr/lib/udev/rules.d/60-fido-id.rules
Reading rules file: /usr/lib/udev/rules.d/60-input-id.rules
[... 111 lines cut ...]
DEVNUM=003
MAJOR=189
MINOR=2
ACTION=add
SUBSYSTEM=usb
ID_VENDOR=8087
ID_VENDOR_ENC=8087
ID_VENDOR_ID=8087
ID_MODEL=0a2b
ID_MODEL_ENC=0a2b
ID_MODEL_ID=0a2b
ID_REVISION=0010
ID_SERIAL=8087_0a2b
ID_BUS=usb
ID_USB_INTERFACES=:e00101:
ID_VENDOR_FROM_DATABASE=Intel Corp.
ID_MODEL_FROM_DATABASE=Bluetooth wireless interface
ID_PATH=pci-0000:00:14.0-usb-0:7
ID_PATH_TAG=pci-0000_00_14_0-usb-0_7
USEC_INITIALIZED=10374962
run: '/usr/lib/virtualbox/VBoxCreateUSBNode.sh 189 2 e0'
run: '/usr/libexec/udev/tlp-usb-udev /devices/pci0000:00/0000:00:14.0/usb1/1-7'
Unload module index
Unloaded link configuration context.
```

linrunner, 2021-01-02:
Let's look at the full package contents then:

    rpm -ql tlp

I would also like to see:

    ls -l /run/tlp

phrxmd, 2021-01-02:
Sure. Here are the package contents:
```
# rpm -ql tlp
/etc/tlp.conf
/etc/tlp.d
/etc/tlp.d/00-template.conf
/etc/tlp.d/README
/usr/bin/bluetooth
/usr/bin/run-on-ac
/usr/bin/run-on-bat
/usr/bin/tlp-stat
/usr/bin/wifi
/usr/bin/wwan
/usr/lib/systemd/system/system-sleep
/usr/lib/systemd/system/system-sleep/tlp
/usr/lib/systemd/system/tlp.service
/usr/lib/udev/rules.d/85-tlp.rules
/usr/lib/udev/tlp-usb-udev
/usr/sbin/rctlp
/usr/sbin/rctlp-sleep
/usr/sbin/tlp
/usr/share/bash-completion
/usr/share/bash-completion/completions
/usr/share/bash-completion/completions/bluetooth
/usr/share/bash-completion/completions/tlp
/usr/share/bash-completion/completions/tlp-rdw
/usr/share/bash-completion/completions/tlp-stat
/usr/share/bash-completion/completions/wifi
/usr/share/bash-completion/completions/wwan
/usr/share/doc/packages/tlp
/usr/share/doc/packages/tlp/AUTHORS
/usr/share/doc/packages/tlp/README.md
/usr/share/doc/packages/tlp/changelog
/usr/share/licenses/tlp
/usr/share/licenses/tlp/COPYING
/usr/share/licenses/tlp/LICENSE
/usr/share/man/man1/bluetooth.1.gz
/usr/share/man/man1/run-on-ac.1.gz
/usr/share/man/man1/run-on-bat.1.gz
/usr/share/man/man1/wifi.1.gz
/usr/share/man/man1/wwan.1.gz
/usr/share/man/man8/tlp-stat.8.gz
/usr/share/man/man8/tlp.8.gz
/usr/share/man/man8/tlp.service.8.gz
/usr/share/metainfo
/usr/share/metainfo/de.linrunner.tlp.metainfo.xml
/usr/share/tlp
/usr/share/tlp/defaults.conf
/usr/share/tlp/func.d
/usr/share/tlp/func.d/05-tlp-func-pm
/usr/share/tlp/func.d/10-tlp-func-cpu
/usr/share/tlp/func.d/15-tlp-func-disk
/usr/share/tlp/func.d/20-tlp-func-usb
/usr/share/tlp/func.d/25-tlp-func-rf
/usr/share/tlp/func.d/30-tlp-func-rf-sw
/usr/share/tlp/func.d/35-tlp-func-batt
/usr/share/tlp/func.d/40-tlp-func-bay
/usr/share/tlp/func.d/45-tlp-func-gpu
/usr/share/tlp/func.d/tlp-func-stat
/usr/share/tlp/tlp-func-base
/usr/share/tlp/tlp-pcilist
/usr/share/tlp/tlp-readconfs
/usr/share/tlp/tlp-usblist
/usr/share/tlp/tpacpi-bat
/var/lib/tlp
```
And the content of `/run/tlp`:
```
# ls -l /run/tlp
insgesamt 12
-rw-r--r-- 1 root root    2  1. Jan 17:51 last_pwr
-rw-rw-r-- 1 root root 1635  1. Jan 17:20 run.conf
-rw-r--r-- 1 root root    0 27. Dez 15:51 usb_done
-rw-r--r-- 1 root root    5  2. Jan 21:04 wlan0.itype
```

linrunner, 2021-01-02:
The install path /usr/lib/udev/tlp-usb-udev is fine. 

I dissected the .rpm myself and found that but 85-tlp.rules and 85-tlp-rdw.rules contain the wrong path:

    RUN+="/usr/libexec/udev/tlp-usb-udev %p"

The problem occurs when the package build calls TLP's Makefile with the TLP_ULIB parameter (tlp.spec Line 77):

      TLP_ULIB=%{_udevrulesdir}/.. \

This should give TLP_ULIB=/usr/lib/udev but seems to produce /usr/libexec/udev instead.

**I suggest you file a bug against the openSUSE package.** 

Btw, the same construct works in Fedora's tlp.spec.

One more thing to add to your bug report:  /usr/share/bash-completion/completions/tlp-rdw is installed by the tlp package although it belongs to tlp-rdw.

phrxmd, 2021-01-03:
Thank you. I've filed [bug 1180495](https://bugzilla.opensuse.org/show_bug.cgi?id=1180495) against openSUSE tw for the path in the udev rules, and [bug 1180496](https://bugzilla.opensuse.org/show_bug.cgi?id=1180496) for the bash completions in the wrong package.

linrunner, 2021-01-03:
Great, let's see.

phrxmd, 2021-01-10:
openSUSE has a new version of the tlp package (1.3.1-3.1) that shows udev events in the `tlp-stat -T`output, like this:
```
# tlp-stat -T
...
Jan 10 10:50:49 calypso tlp[19836]: +++ auto (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Jan 10 10:50:49 calypso tlp[19836]: power_source=ac
Jan 10 10:50:49 calypso tlp[19836]: manual_mode=none
Jan 10 10:50:49 calypso tlp[19836]: power_mode=ac
Jan 10 10:50:50 calypso tlp[19723]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpukIiZm -> /run/tlp/run.conf
Jan 10 10:56:19 calypso tlp[23504]: usb_suspend_device.udev.on_hid_black.no_change: Bus 1 Dev 6 ID 056e:010d /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2 []
```

The card reader is shown with `control=auto` in `tlp-stat` now, so I guess this is fixed.

closed 2021-01-10 by phrxmd

linrunner, 2021-01-10:
Thanks for your feedback.