linrunner/TLP #543: Some driver blacklist doesn't work

korongmambu, 2020-11-25:
Even the device blacklist doesn't work, and when on mode charging the driver or device is not all on.

This is my `tlp-stat --pcie` when discharging:
>  --- TLP 1.3.1 --------------------------------------------
> 
> +++ PCIe Active State Power Management
> /sys/module/pcie_aspm/parameters/policy = [default] performance powersave powersupersave (using BIOS preferences)
> 
> +++ Runtime Power Management
> Device blacklist = 00:02.0 00:03.0 00:16.0 00:1b.0 00:1c.0 00:1c.2 00:1c.3 00:1c.4 04:00.0
> Driver blacklist = i915 mei_me nouveau nvidia pcieport snd_hda_intel
> 
> /sys/bus/pci/devices/0000:00:00.0/power/control = auto (0x060000, Host bridge, bdw_uncore)
> /sys/bus/pci/devices/0000:00:02.0/power/control = on   (0x030000, VGA compatible controller, i915)
> /sys/bus/pci/devices/0000:00:03.0/power/control = auto (0x040300, Audio device, snd_hda_intel)
> /sys/bus/pci/devices/0000:00:04.0/power/control = auto (0x118000, Signal processing controller, proc_thermal)
> /sys/bus/pci/devices/0000:00:14.0/power/control = auto (0x0c0330, USB controller, xhci_hcd)
> /sys/bus/pci/devices/0000:00:16.0/power/control = on   (0x078000, Communication controller, mei_me)
> /sys/bus/pci/devices/0000:00:1b.0/power/control = auto (0x040300, Audio device, snd_hda_intel)
> /sys/bus/pci/devices/0000:00:1c.0/power/control = on   (0x060400, PCI bridge, pcieport)
> /sys/bus/pci/devices/0000:00:1c.2/power/control = auto (0x060400, PCI bridge, pcieport)
> /sys/bus/pci/devices/0000:00:1c.3/power/control = auto (0x060400, PCI bridge, pcieport)
> /sys/bus/pci/devices/0000:00:1c.4/power/control = auto (0x060400, PCI bridge, pcieport)
> /sys/bus/pci/devices/0000:00:1f.0/power/control = auto (0x060100, ISA bridge, lpc_ich)
> /sys/bus/pci/devices/0000:00:1f.2/power/control = auto (0x010601, SATA controller, ahci)
> /sys/bus/pci/devices/0000:00:1f.3/power/control = auto (0x0c0500, SMBus, i801_smbus)
> /sys/bus/pci/devices/0000:00:1f.6/power/control = auto (0x118000, Signal processing controller, intel_pch_thermal)
> /sys/bus/pci/devices/0000:02:00.0/power/control = auto (0x020000, Ethernet controller, r8168)
> /sys/bus/pci/devices/0000:03:00.0/power/control = auto (0x028000, Network controller, ath9k)
> /sys/bus/pci/devices/0000:04:00.0/power/control = auto (0x030200, 3D controller, nvidia)

This is my `tlp-stat --pcie` when charging:
> --- TLP 1.3.1 --------------------------------------------
> 
> +++ PCIe Active State Power Management
> /sys/module/pcie_aspm/parameters/policy = [default] performance powersave powersupersave (using BIOS preferences)
> 
> +++ Runtime Power Management
> Device blacklist = 00:02.0 00:03.0 00:16.0 00:1b.0 00:1c.0 00:1c.2 00:1c.3 00:1c.4 04:00.0
> Driver blacklist = i915 mei_me nouveau nvidia pcieport snd_hda_intel
> 
> /sys/bus/pci/devices/0000:00:00.0/power/control = on   (0x060000, Host bridge, bdw_uncore)
> /sys/bus/pci/devices/0000:00:02.0/power/control = on   (0x030000, VGA compatible controller, i915)
> /sys/bus/pci/devices/0000:00:03.0/power/control = auto (0x040300, Audio device, snd_hda_intel)
> /sys/bus/pci/devices/0000:00:04.0/power/control = on   (0x118000, Signal processing controller, proc_thermal)
> /sys/bus/pci/devices/0000:00:14.0/power/control = on   (0x0c0330, USB controller, xhci_hcd)
> /sys/bus/pci/devices/0000:00:16.0/power/control = on   (0x078000, Communication controller, mei_me)
> /sys/bus/pci/devices/0000:00:1b.0/power/control = auto (0x040300, Audio device, snd_hda_intel)
> /sys/bus/pci/devices/0000:00:1c.0/power/control = on   (0x060400, PCI bridge, pcieport)
> /sys/bus/pci/devices/0000:00:1c.2/power/control = auto (0x060400, PCI bridge, pcieport)
> /sys/bus/pci/devices/0000:00:1c.3/power/control = auto (0x060400, PCI bridge, pcieport)
> /sys/bus/pci/devices/0000:00:1c.4/power/control = auto (0x060400, PCI bridge, pcieport)
> /sys/bus/pci/devices/0000:00:1f.0/power/control = on   (0x060100, ISA bridge, lpc_ich)
> /sys/bus/pci/devices/0000:00:1f.2/power/control = on   (0x010601, SATA controller, ahci)
> /sys/bus/pci/devices/0000:00:1f.3/power/control = on   (0x0c0500, SMBus, i801_smbus)
> /sys/bus/pci/devices/0000:00:1f.6/power/control = on   (0x118000, Signal processing controller, intel_pch_thermal)
> /sys/bus/pci/devices/0000:02:00.0/power/control = on   (0x020000, Ethernet controller, r8168)
> /sys/bus/pci/devices/0000:03:00.0/power/control = on   (0x028000, Network controller, ath9k)
> /sys/bus/pci/devices/0000:04:00.0/power/control = auto (0x030200, 3D controller, nvidia)

linrunner, 2020-11-26:
Your issue cannot be processed without complete outputs as required by the the [issue template](https://github.com/linrunner/TLP/blob/master/.github/ISSUE_TEMPLATE/bug_report.md).

Provide them via https://gist.github.com/ please.

korongmambu, 2020-11-26:
Sorry, this is my first report on github. Also english is not my native language, I hope you understand.

linrunner, 2020-11-27:
No problem. I gather you are trying to force said devices to `on` by blacklisting them. 

This is not what `RUNTIME_PM{_DRIVER}_BLACKLIST` actually does and unfortunately what you want is not possible with the current TLP version.

I took your issue as an opportunity to improve the [documentation](https://linrunner.de/tlp/settings/runtimepm.html):

> * Exclusion of a device means that the value on or auto as initialized by the kernel default at system startup is used and TLP won’t touch it at all
> * A device initialized to auto by the kernel cannot be set to on by entering it into RUNTIME_PM_BLACKLIST

[TLP 1.4](https://github.com/linrunner/TLP/blob/master/changelog) will add `RUNTIME_PM_DISABLE` which implements your requirement. If you like to test the alpha version, you may build and install the package `tlp-git` from the AUR which tracks my master branch.

Remember to check the [changelog](https://github.com/linrunner/TLP/blob/master/changelog) for changed parameter names in 1.4.

Before you do that i suggest you clean up your configuration files:

- **/etc/default/tlp** is deprecated --> delete it
- I don't see **/etc/tlp.conf** -->  check if it's there, if not remove the package `tlp, remove everything in **/etc/tlp.d/**, then install `tlp` again`
- Dont' change **/etc/tlp.d/00-template.conf** but copy it to **01-my.conf** (for example) and enter your individual settings there

linrunner, 2020-12-02:
Do you have any news?

linrunner, 2020-12-16:
I am still interested in your test results, but I will close here for the time being.

closed 2020-12-16 by linrunner