linrunner/TLP #545: Apply USB autosuspend settings and rules upon resume

ValdikSS, 2020-12-15:
**Describe the bug**

When the laptop resumes from sleep or hibernation, USB autosuspend rules (blacklist/whitelist) are not applied.
I have tlp.conf configured with `USB_BLACKLIST="8087:07dc", USB_BLACKLIST_BTUSB=1`, but `8087:07dc` device is in `auto` suspend state when the laptop is resumed.
Running `tlp start` fixes the issue, disabling autosuspend for `8087:07dc`. However, `tlp resume`, which is executed upon resuming, does not configure USB autosuspend.

**Expected behavior**

USB autosuspend rules are applied upon resume, just as in `tlp start` call.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? **Both.**
2. Actions to reproduce the behaviour: **Suspend the laptop, resume the laptop, run `tlp-stat -u` to check `control` status of the blacklisted device**
3. Shell commands entered and their output

Before suspend:
```
$ sudo tlp-stat -u
--- TLP 1.3.1 --------------------------------------------

+++ USB
Autosuspend         = enabled
Device whitelist    = (not configured)
Device blacklist    = 8087:07dc
Bluetooth blacklist = enabled
Phone blacklist     = disabled
WWAN blacklist      = disabled

Bus 002 Device 003 ID 8087:07dc control = on,   autosuspend_delay_ms = 2000 -- Intel Corp. Bluetooth wireless interface (btusb)
Bus 002 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =    0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 004 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 3.0 root hub (hub)
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 001 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =    0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
```

After resume
```
$ sudo tlp-stat -u
--- TLP 1.3.1 --------------------------------------------

+++ USB
Autosuspend         = enabled
Device whitelist    = (not configured)
Device blacklist    = 8087:07dc
Bluetooth blacklist = enabled
Phone blacklist     = disabled
WWAN blacklist      = disabled

Bus 002 Device 003 ID 8087:07dc control = auto, autosuspend_delay_ms = 2000 -- Intel Corp. Bluetooth wireless interface (btusb)
Bus 002 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =    0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 004 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 3.0 root hub (hub)
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 001 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =    0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
```

4. Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.: **https://gist.github.com/ValdikSS/455b8f251c2e126636182a4d1443a79d**

**Additional context**

TLP 1.3.1 on Fedora 33.

ValdikSS, 2020-12-15:
Adding `X_TLP_USB_MODE=1` to the `tlp.conf` does nothing.

PaulVanSchayck, 2020-12-15:
Like I mention in #334 I also see even with `USB_BLACKLIST_BTUSB=1` or `USB_BLACKLIST="8087:0aaa"`, that my BT device is getting `control = auto`.

But unlike the ValdikSS, I see no relation with suspend. This already happens to me on first boot. Running `tlp usb` manually will always fix it.

Ubuntu 20.10. Linux 5.8.0-7630-generic. tlp 1.3.1. Lenovo X1 Carbon 7th gen. 

Happy to open separate bug report, if this is unrelated to this report.

linrunner, 2020-12-15:
@ValdikSS : that's right and works as designed(tm). TLP in normal operation relies solely on udev events to set `control` for USB devices. Your kernel seems not to preserve the setting, which i consider a kernel issue.

To achieve the desired behaviour, you may insert the line

    set_usb_suspend 0 auto

into **/usr/sbin/tlp** below line 417:  https://github.com/linrunner/TLP/blob/1.3.1/tlp.in#L417

ps. is it possible that i know your nickname because you develop for LineageOS?

linrunner, 2020-12-15:
@PaulVanSchayck : yours also sounds like a kernel issue, you may try the config workaround mentioned by @ValdikSS (also see #518 ).

ValdikSS, 2020-12-15:
@linrunner

>To achieve the desired behaviour, you may insert 

Yes, that works, that's what I've done before opening the issue. Should it work on older kernels? The older kernels fire new device event which is handled by udev?

>ps. is it possible that i know your nickname because you develop for LineageOS?

Not really a developer, but I've contributed Bluetooth SBC audio patch which is probably familiar to you. See https://habr.com/ru/post/456476/, https://lineageos.org/engineering/Bluetooth-SBC-XQ/

linrunner, 2020-12-15:
Imho the kernel never issued udev new device events for existing devices after resume. You may look into it yourself by enabling TLP's trace mode with `TLP_DEBUG="run udev usb"`.

Btw, i can confirm the problem for my X220. Even though i have a different bluetooth chip:

    Bus 001 Device 005 ID 0a5c:217f control = auto, autosuspend_delay_ms = 2000 -- Broadcom Corp. BCM2045B (BDC-2.1) (btusb)

areographe, 2021-01-31:
Hi, just confirming I have experienced this problem for the last few months, using kernels 5.4 and 5.10 (Manjaro) on the Thinkpad X1 Carbon (4th generation). I have used the workaround specified in https://github.com/linrunner/TLP/issues/545#issuecomment-745370611 for now. I will open a bug report for the Manjaro kernels re: the description of the issue in https://github.com/linrunner/TLP/issues/545#issuecomment-745370611 .

maxammann, 2021-02-05:
@areographe I can confirm. I'm using arch and had the problem that my mouse (MX Master) with a Logitech Unify Receiver goes to sleep every 2 seconds without movement. I first thought this would be a kernel driver issue with logitech.

Turns out it is a power management issue like described here. `tlp-stat -u` says that all devices have mode `auto` after boot. When executing `tlp usb` the mouse works and control is set to on. `X_TLP_USB_MODE=1` did not change anything for me.

So the kernel is somehow not executing the udev rules?

linrunner, 2021-02-05:
@maxammann : 

> So the kernel is somehow not executing the udev rules?

No. TLP's traces show that the rule and TLP's part is executed (btw: by the udevd not the kernel ) and results in control=on. But after that _something else_ reverts to control=auto.

linrunner, 2021-05-06:
Hi @all: I finally took the time to analyze this in more detail. Seems the kernel itself enforces autosuspend for all btusb devices. The responsible Kconfig is

    CONFIG_BT_HCIBTUSB_AUTOSUSPEND=y 

When i apply the boot option `btusb.enable_autosuspend=0` to my GRUB config, blacklisting in TLP works again. Also the opposite works: without blacklisting TLP activates autosuspend.

ValdikSS, 2021-08-01:
I've been using `btusb.enable_autosuspend=0` without any issues for months.
@linrunner, did you do documentation changes or should I add it to docs?

closed 2021-08-01 by ValdikSS