linrunner/TLP #547: [ThinkPad X230] Unable to get recalibrate to work

jaquer, 2021-01-03:
**Describe the bug**

I am unable to get the recalibrate feature to work on my ThinkPad X230. Here's the output of `tlp recalibrate`:

    Setting temporary charge thresholds for BAT0:
      start =  96 (no change)
      stop  = 100 (no change)
    Error: battery discharge/recalibrate not available.

Here's the relevant section of `tlp-stat -b` (full output of `tlp-stat` below):

    +++ Battery Features: Charge Thresholds and Recalibrate
    natacpi    = active (data, thresholds)
    tpacpi-bat = inactive (superseded by natacpi)
    tp-smapi   = inactive (ThinkPad not supported)

I have made made some software modifications to the machine, namely flash the EC to allow third party batteries, and replace the stock BIOS (see **Additional context** section below). My apologies in advance if those invalidate this issue.

**Expected behavior**

According to the [documentation](https://linrunner.de/tlp/faq/battery.html#which-kernel-module-do-i-need-for-my-thinkpad) I would expect the natacpi/tpacpi-bat combination to be enable TLP to calibrate the battery.

**To Reproduce**

Run `tlp recalibrate` as root.

`tlp-stat` output while on battery: https://gist.github.com/jaquer/a3b1a077013845923e0b87a73b9c965d - [raw](https://gist.githubusercontent.com/jaquer/a3b1a077013845923e0b87a73b9c965d/raw/e7fdf6fd167313a4f99a5b4d5a6061267eb02fe5/tlp-stat%2520%255Bbattery%255D) 
`tlp-stat` output while on ac: https://gist.github.com/jaquer/5e5ce204e6081ce70fbe3e39e8fb741a - [raw](https://gist.githubusercontent.com/jaquer/5e5ce204e6081ce70fbe3e39e8fb741a/raw/1da654612578f1b6a3e574dcb2a04dd8b28c0899/tlp-stat%2520%255Bac%255D)

**Additional context**

Checked for updates to all packages before submitting.

 - Arch Linux
 - Kernel: 5.10.4-arch2-1
 - tlp 1.3.1-2
 - acpi_call 1.2.1-15
 - Using a third party battery, but having the exact same issue with the original battery.
 - Flashed the EC using [thinkpad-ec](https://github.com/hamishcoleman/thinkpad-ec) in order to use the third party battery.
 - Replaced stock BIOS with [skulls](https://github.com/merge/skulls) using [1vyrain](https://github.com/n4ru/1vyrain/).

Please let me know if you need any more information. Thanks in advance for your help.

linrunner, 2021-01-04:
Hi,

please show the output of (as ROOT);

     /usr/share/tlp/tpacpi-bat -v -g FD 1; echo $?

jaquer, 2021-01-04:
Thanks for your help. Here it is:

    [root@snowball ~]# /usr/share/tlp/tpacpi-bat -v -g FD 1; echo $?
    Error: AE_NOT_FOUND for ASL base: \_SB.PCI0.LPCB.EC.HKEY
    Error: AE_NOT_FOUND at /usr/share/tlp/tpacpi-bat line 424.
    255

linrunner, 2021-01-04:
Furthermore:

     /usr/share/tlp/tpacpi-bat -v -g ST 1
     /usr/share/tlp/tpacpi-bat -v -g SP 1

jaquer, 2021-01-04:
[root@snowball ~]# /usr/share/tlp/tpacpi-bat -v -g ST 1
    Call    : \_SB.PCI0.LPCB.EC.HKEY.BCTG 0x1
    Response: 0x343
    67 (relative percent)

    [root@snowball ~]# /usr/share/tlp/tpacpi-bat -v -g SP 1
    Call    : \_SB.PCI0.LPCB.EC.HKEY.BCSG 0x1
    Response: 0x300
    0 (default)

linrunner, 2021-01-04:
Interesting. Coreboot seems to implement the ACPI calls for the start (BCTG) and the stop (BCSG) threshold, otherwise  natacpi would not work for them either. But it does not support the call for force_discharge (BDSG). Sad.

I suggest you ask the Coreboot devs about this.

jaquer, 2021-01-04:
Got it. I appreciate all your work. I will raise the issue upstream.

jaquer, 2021-01-05:
I did some more research and found out that coreboot also creates issues for other programs, such as [dockd](https://github.com/libthinkpad/dockd/issues/7#issuecomment-370502422), so I've decided not to pursue this further. I've switched to an 1vyrain patched BIOS and now TLP is working perfectly.

I will go ahead close this issue. Thanks again for your help.

closed 2021-01-05 by jaquer