linrunner/TLP #548: [ASUS FA706IU] laptop not supported

IgorFurmanov, 2021-01-07:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

Battery support for ASUS TUF A17 (FA706IU) is not available.
Recently I installed the TLP on my Xubuntu 20.04 and following configs do not have effect 
START_CHARGE_THRESH_BAT1=60
STOP_CHARGE_THRESH_BAT1=75

**Expected behavior**

STOP_CHARGE_THRESH_BAT should stop charging on specified level (e.g. 80%)

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
2. Actions to reproduce the behaviour
3. Shell commands entered and their output
4. Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.

**Additional context**
1. On both battery & AC
2. install tlp on ubuntu / xubuntu 20.04 and set STOP_CHARGE_THRESH_BAT0 / BAT1 config to 75%
3. - sudo tlp-stat -b

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = inactive (laptop not supported)
tpacpi-bat = inactive (laptop not supported)
tp-smapi   = inactive (laptop not supported)

- sudo tlp recalibrate

Error: battery charge thresholds not available.
Error: battery discharge/recalibrate not available.

4. https://gist.github.com/IgorFurmanov/1ec0cc5a00c567907908a596e6fe6578
https://gist.github.com/IgorFurmanov/4d09443831dcc3cd4792b2c638b7221c

Additional context:

Config:
---
START_CHARGE_THRESH_BAT0=60
STOP_CHARGE_THRESH_BAT0=75

START_CHARGE_THRESH_BAT1=60
STOP_CHARGE_THRESH_BAT1=75

linrunner, 2021-01-10:
Duplicate of #528

linrunner, 2021-01-10:
This is already in the works, but will take some time. Please sign up in #528 to be notified as soon as alpha testing begins.

closed 2021-01-10 by linrunner

linrunner, 2021-05-06:
Hi, this is now implemented. Please participate in the test --> https://gist.github.com/linrunner/2ead4b591eed33055cf86a38ccc73949