linrunner/TLP #549: Docs vs manpage on battery thresholds

redraw, 2021-01-14:
On the latest docs version (1.3) for `setcharge` it [says](https://linrunner.de/tlp/usage/tlp.html#change-battery-charge-thresholds-to-temporary-values),

> Sets the thresholds to the given values. Valid thresholds range from 1 to 100; START_THRESH must be below STOP_THRESH **- 3**. Without parameters the configured settings for the main battery (BAT0) are applied. **Upon reboot, thresholds are reset to the configured settings.**

and latest `tlp` (1.3) manpage says,

> Set  charge  thresholds  of  main  (BAT0)  or  auxiliary  (BAT1)  battery  temporarily.   Values must be between 1 and 100, START_CHARGE must not exceed
              STOP_CHARGE **- 4**.  **Configured thresholds are restored upon next system boot-up.** When called without arguments, configured thresholds are set.

-3/-4 difference is minor. I guess that -4 is the correct one, as defaults are 96/100.

However, as far as I've tested temporary thresholds are restored across system reboots, as the manpage says. So, I guess docs are old, right?

In that case, it might be better to update manpage's one to "~Configured~Temporary thresholds are restored upon next system boot-up.", so that it's clear that only when you run `sudo setcharge` then configured (from settings file) thresholds are actually restored.

By the way, where are those temporary thresholds stored?

linrunner, 2021-01-25:
Hi, 

TLP writes the thresholds to the ThinkPad's embedded controller. You may consider them to be permanent, but in fact they may be reset by removing the battery (on older models) or by another OS (the one with ThinkVantage software) in dualboot setups for example. 

When configured in tlp.conf TLP restores them upon boot if the hardware holds different values. Maybe i should change the website to "... restored ...".

redraw, 2021-01-25:
Oh, actually you're right. When running `setcharge <min> <max>`, the thresholds are actually written in the EC, not saved anywhere in the system. 
However, if they are uncommented and configured in tlp.conf, those are restored (and written to the EC) upon boot. That was my confusion, thanks!

linrunner, 2021-04-05:
Done.

closed 2021-04-05 by linrunner