linrunner/TLP #551: High power consumption after suspend/resume

pmj642, 2021-02-01:
**Describe the bug**

I've noticed on 3-4 occasions now that when my device is plugged and I close the lid to let it charge, if I remove the plug while in suspend then on resuming the power consumption is high. When I then run `sudo tlp-stat -s` then it becomes normal, which is strange as it's not even `sudo tlp start`.

**Expected behavior**

Power consumption should be normal in the above mentioned situation.

**Additional context**

Powertop output on starting 
[image]

tlp-stat -s output shows that its running in bat mode 
[image]
after tlp-stat -s command 
[image]

linrunner, 2021-02-02:
Please have a look at the issue template (again) to see what information is mandatory for an issue to be processed. In any case, screenshots are completely unsuitable.

I can assure you that `tlp-stat -s` does not make any changes to power saving settings. Maybe your laptop just needs some time to recognize the change of the power source after the resume.

I would start by checking the "tunables" tab in powertop for changes occuring.

pmj642, 2021-02-03:
Yes, I'm aware that `tlp-stat -s` doesn't make any changes. Doing `tlp start` works too. And initially when this issue occurred it went on for about an hour until I restarted tlp to reduce the power consumption.
 
I'll make note of the config next time the issue occurs and post an issue. As of now, I'm closing this issue. Thanks.

closed 2021-02-03 by pmj642

linrunner, 2021-02-08:
Feel free to reopen here when you have more evidence.