linrunner/TLP #552: tlp-stat - s gives notices about systemd-rfkill

DasJott, 2021-02-16:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

If I type in `tlp-stat -s` it returns some info, but also this:
```
Notice: systemd-rfkill.service is not masked -- invoke "systemctl mask systemd-rfkill.service" to correct this!
Notice: systemd-rfkill.socket is not masked -- invoke "systemctl mask systemd-rfkill.socket" to correct this!
```

**Expected behavior**

Everything should be set up correctly, no warnings or notices.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Install tlp via eopkg on Solus
2. Open Terminal and type `tlp-stat -s`
3. See notices about service not enabled and the notices above.

**Additional context**
I use Solus and installed tlp via its software center. Not impossible that it doesn't get setup correctly.  
It seems that a call so `systemctl enable tlp.service` is missing and maybe also the described ones.

I don't really know whether this is a bug, but it is a notice ;)

linrunner, 2021-02-16:
Would you mind posting the _full_ tlp-stat via gist output as required by the issue template?

DasJott, 2021-02-16:
[Here you are](https://gist.github.com/DasJott/dc0c9525a7d7a6ea4d77fc66da3c32a4)
Sorry, must have overseen that...

linrunner, 2021-02-16:
These messages are relevant only when you use TLP's settings for switching radio devices, i.e.:

- https://linrunner.de/tlp/settings/radio.html
- https://linrunner.de/tlp/settings/rdw.html

Then these standard services should be masked with the shown commands, to prevent conflicts with TLP's actions. Many distribution's packages of TLP already do that, but in the case of Solus obviously not.

Bottom line: your configuration does not use radio settings, so you may safely ignore the messages.

DasJott, 2021-02-17:
Thank you very much for your estimation!
Would those also be important if I wanted to use wifi power safe?

I didn't set up anything when noticing those messages... Now I am considering using wifi power safe mode on bat.

linrunner, 2021-02-17:
No. WIFI_PWR_ON_BAT is not among the settings whose descriptions I have linked above.

Since this is not a bug, but rather a problem of understanding, i allow myself to close here.

closed 2021-02-17 by linrunner

DasJott, 2021-02-17:
Just to put that right: "i.e." implied that those are examples and left open the possibility for more😉

But yes, I think this is no bug. At least not here, but rather on Solus packaging.

linrunner, 2021-02-18:
"i.e." means "in other words" (I am also not a native speaker).

> At least not here, but rather on **Solus packaging**.

Indeed.

linrunner, 2021-02-18:
ps. if you file a Solus bug, please share the link here. Thanks.

ps2. the Debain/Ubuntu package uses these scripts: https://linrunner.de/tlp/developers/scripts.html

oscarpax, 2021-10-02:
Starting with update to tlp 1.4 on Arch, I started to see same warnings as the op. I thought that you only have to mask rfkill.service and rfkill.socket IF you use tlp-rdw? I only install tlp, I don't install tlp-rdw nor can I use it since I don't use NetworkManager which I understand is a  prerequisite for tlp-drw. I'm confused..