linrunner/TLP #555: Add ThinkPad Extras release for Fedora 34

lakotamm, 2021-03-28:
I am testing Fedora 34 Beta and I ran into an issue when installing TLP on my T490. In your [Thinkpad only installation guide](https://linrunner.de/tlp/installation/fedora.html) it is suggested to add ThinkPad Extras repository using:

> dnf install https://repo.linrunner.de/fedora/tlp/repos/releases/tlp-release.fc$(rpm -E %fedora).noarch.rpm

This step at the moment results in an error, because there is no release file for Fedora 34.

> [MIRROR] tlp-release.fc34.noarch.rpm: Status code: 404 for https://repo.linrunner.de/fedora/tlp/repos/releases/tlp-release.fc34.noarch.rpm (IP: 85.13.161.171)

Would it be possible to add the release file (and the required packages) for Fedora 34? I am happy to help with testing to ensure that everything is running fine.

Thank you very much!

Thorashh, 2021-03-29:
You'll find all needed files in the tlp updates-testing Repo.

lakotamm, 2021-03-29:
Thank you! I added the Fedora 33 ThinkPad Extra repository and enabled the testing repository. Everything works fine!

closed 2021-03-29 by lakotamm