linrunner/TLP #556: TLP aggressively suspends active USB audio devices when on AC

flexiondotorg, 2021-04-14:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/flexiondotorg/337e3e077ec8dbc0e8debec2c83173db)

**Describe the bug**

When running on AC powered desktop hardware active USB audio devices that are playing/recording audio are suspended after a few seconds. Playing audio restores audio playback but the input devices for recording audio are not.

**Expected behaviour**

When running on AC powered desktop hardware active USB audio devices are not suspended after a few seconds.

**To Reproduce**

Steps to reproduce the unexpected behaviour:

1. AC power; I only have desktop hardware
2. Install Ubuntu 21.04. On first boot `sudo apt install tlp` and reboot. Connect a USB audio interface, listen to music or watch a video and after a few seconds, the device is suspended.
4. Here is the output of `tlp-stat`, although I have USB auto-suspend disabled to workaround the issue: <https://gist.github.com/flexiondotorg/337e3e077ec8dbc0e8debec2c83173db>

**Additional context**

I have raised a downstream bug in Launchpad:

  * https://pad.lv/1923737

TLP is shipped by default in Ubuntu MATE, which I maintain. I am going to disable USB suspend in Ubuntu MATE 21.04 as a workaround.

linrunner, 2021-04-14:
Hi Martin,

i wasn't aware that TLP is still in MATE's default install. If i were to defuse the default configuration too far, then TLP would no longer make sense. However, excluding audio devices by default seems reasonable.

Packages for testing are [here](https://download.linrunner.de/packages/).

linrunner, 2021-04-16:
@self: https://github.com/ubuntu-mate/ubuntu-mate-settings/commit/c2a8a00d676b34e09c242d964a2a4d3ed779dda6