linrunner/TLP #557: Can't set policy if using Performance gov.

PremiumUsername, 2021-04-14:
**Describe the bug**

Chosen policy for `CPU_ENERGY_PERF_POLICY` is not applied if using `performance` governor. If using `powersave` all is working as intended.
 
It is possible to set the right policy manually using x86_energy_perf_policy with:
`sudo x86_energy_perf_policy -all balance-power`

**Expected behavior**
policy choice is enacted. Now policy always becomes `performance`

**To Reproduce**

1. Does the problem occur on battery or AC or both? 

Both

2. Actions to reproduce the behaviour

Chose `performance` for gov and a policy other than `performance`

3. Shell commands entered and their output

4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**

https://gist.github.com/PremiumUsername/ca7cf8621f2b174e5f4383364e94d64f

Enviroment:
Skylake platform, Manjaro, K. 5.11

linrunner, 2021-04-14:
Sorry, but the template does *not* say to post *trace* output but a simple `tlp-stat`. In your case

    tlp-stat -c -s -p

suffices.

PremiumUsername, 2021-04-16:
Ok, I apologise... trying again:

https://gist.github.com/PremiumUsername/d5966f3d5aa28e0f14ea29a2e265ce87

linrunner, 2021-04-16:
Fine. Please try in a root shell:

    echo "balance_power" > /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference

On my system this throws "bash: echo: write error: Device or resource busy". So to me it seems this is just a limitation of the performance governor.

PremiumUsername, 2021-04-16:
My system responds the same way, but if I use:
`sudo x86_energy_perf_policy -all balance-power` it works:
```
[d-cave cpufreq]# echo "balance_power" > /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference
bash: echo: write error: Device or resource busy
[d-cave cpufreq]# x86_energy_perf_policy -all "balance-power"
[d-cave cpufreq]# cat /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference
balance_power
```

linrunner, 2021-04-17:
That may be, but TLP does not use x86_energy_perf_policy as of kernel 5.2. I deliberately want to reduce dependencies on external tools. there is no going back.

By the way, I find the limitation in the kernel understandable. Performance governor means full throttle, "balance_power" contradicts this.

So if you don't agree with that, you should raise the issue with the kernel developers.

PremiumUsername, 2021-04-18:
Ok, I can understand that. 

I think the separation of CPU gov. and policy is a bit confusing in this case, since `performance` is not _really_ full throttle. Both do freq. scaling and you can adjust the policy of that scaling so why do you need two types of gov?

closed 2021-09-24 by linrunner