linrunner/TLP #561: tlp should not override tunables set from "Bad" to "Good" by powertop

Pointedstick, 2021-05-23:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**
Manually running `sudo powertop --auto-tune` switches all power-related tunables to their "Good" state. However tlp overrides these every time the power cable is plugged in or unplugged.

**Expected behavior**

`tlp` should not reset or change the tunables set by `powertop`.

**To Reproduce**

Steps to Reproduce:
1. install both powertop and tlp
2. sudo systemctl enable tlp
3. sudo systemctl start tlp
4. sudo powertop --auto-tune
6. Unplug laptop's power cable

[tlp-stat output.txt](https://github.com/linrunner/TLP/files/6528253/tlp-stat.output.txt)

Pointedstick, 2021-05-23:
As an alternative, perhaps `tlp` could automatically set all the tunables to "Good" so that it is not necessary to use `powertop --auto-tune` to achieve this.

Pointedstick, 2021-05-23:
Never mind, I see now that the tunables (almost) all get set to good automatically when on battery power. So I guess running `powertop --auto-tune` is just totally unnecessary when using `tlp`.

closed 2021-05-23 by Pointedstick

linrunner, 2021-05-23:
Hi, to look in the [FAQ](https://linrunner.de/tlp/faq/powertop.html) before would have saved you effort ...