linrunner/TLP #563: Platform profile support

ErikReider, 2021-06-03:
To be able to set the preferred [platform_profile](https://www.kernel.org/doc/html/v5.13-rc3/userspace-api/sysfs-platform_profile.html) depending on power state (charging and ac). 

The profiles available on my Thinkpad X1 Carbon Gen 9 are `low-power balanced performance`.

Currently I have to run `echo performance | sudo tee /sys/firmware/acpi/platform_profile` manually to set the profile. Would be nice if TLP would do that automatically.

To get all of the profiles, I run `cat /sys/firmware/acpi/platform_profile_choices`.

At the moment some Lenovo laptops support this feature in 5.12 and some HP laptops in 5.13
https://www.phoronix.com/scan.php?page=news_item&px=Lenovo-Platform-Profile-For-512
https://www.phoronix.com/scan.php?page=news_item&px=Linux-5.13-HP-Platform-Profile

ErikReider, 2021-06-03:
Nvm. Looked through the repo and found support for it. Just hasn't been released yet

linrunner, 2021-06-04:
Wait! Don't leave right away. You are welcome to test. 

Packages are available [here](https://download.linrunner.de/packages/). 

Please enable the setting, check on AC and battery and report the output of

    tlp-stat --cdiff -s -p

for both. Thanks in advance.

ErikReider, 2021-06-04:
Haha. Didn't leave. Just installed it through the AUR and it works perfectly. I'll try to run that command and comment the output 😀

ErikReider, 2021-06-04:
This is the output:
```
--- TLP 1.4.0-alpha.2 --------------------------------------------

+++ Configured Settings (only differences to defaults):
/etc/tlp.conf L0038: TLP_WARN_LEVEL="1"
/etc/tlp.conf L0086: CPU_SCALING_GOVERNOR_ON_AC="performance"
/etc/tlp.conf L0087: CPU_SCALING_GOVERNOR_ON_BAT="powersave"
/etc/tlp.conf L0135: CPU_BOOST_ON_AC="1"
/etc/tlp.conf L0136: CPU_BOOST_ON_BAT="1"
/etc/tlp.conf L0144: CPU_HWP_DYN_BOOST_ON_AC="1"
/etc/tlp.conf L0145: CPU_HWP_DYN_BOOST_ON_BAT="1"
/etc/tlp.conf L0168: PLATFORM_PROFILE_ON_AC="performance"
/etc/tlp.conf L0169: PLATFORM_PROFILE_ON_BAT="balanced"
/etc/tlp.conf L0498: START_CHARGE_THRESH_BAT0="75"
/etc/tlp.conf L0499: STOP_CHARGE_THRESH_BAT0="80"

+++ System Info
System         = LENOVO ThinkPad X1 Carbon Gen 9 20XW002BMX
BIOS           = N32ET48W (1.24 )
OS Release     = Manjaro Linux
Kernel         = 5.12.9-1-MANJARO #1 SMP PREEMPT Thu Jun 3 14:56:42 UTC 2021 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.12-x86_64 root=UUID=1c942e93-5a0b-42ef-9750-6784a8b083fc rw quiet splash apparmor=1 security=apparmor resume=UUID=48eb6ffb-5e8d-47b9-89d6-5444657ed8d5 udev.log_priority=3
Init system    = systemd
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = not installed
Last run       = 21:14:40,    300 sec(s) ago
Mode           = battery
Power source   = battery

Warning: systemd-rfkill.service is not masked -- invoke "systemctl mask systemd-rfkill.service" to correct this!
Warning: systemd-rfkill.socket is not masked -- invoke "systemctl mask systemd-rfkill.socket" to correct this!

+++ Processor
CPU model      = 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  4200000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference = balance_power [HWP.EPP]
/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power

/sys/devices/system/cpu/cpu1..cpu7: omitted for clarity, use -v to show all

/sys/devices/system/cpu/intel_pstate/status            = active
/sys/devices/system/cpu/intel_pstate/min_perf_pct      =   9 [%]
/sys/devices/system/cpu/intel_pstate/max_perf_pct      = 100 [%]
/sys/devices/system/cpu/intel_pstate/no_turbo          =   0
/sys/devices/system/cpu/intel_pstate/hwp_dynamic_boost =   1
/sys/devices/system/cpu/intel_pstate/turbo_pct         =  47 [%]
/sys/devices/system/cpu/intel_pstate/num_pstates       =  39

/sys/module/workqueue/parameters/power_efficient       = Y
/proc/sys/kernel/nmi_watchdog                          = 0

+++ Platform Profile
/usr/share/tlp/func.d/tlp-func-stat: line 89: printf: balanced: invalid number
/sys/firmware/acpi/platform_profile                    = 0
/usr/share/tlp/func.d/tlp-func-stat: line 89: printf: low-power balanced performance: invalid number
/sys/firmware/acpi/platform_profile_choices            = 0
```
Seems to be an error at the bottom

linrunner, 2021-06-04:
Fixed. Rebuild the tlp-git package and show again. For battery and AC power please.

ErikReider, 2021-06-04:
Battery:
```
--- TLP 1.4.0-alpha.2 --------------------------------------------

+++ Configured Settings (only differences to defaults):

+++ System Info
System         = LENOVO ThinkPad X1 Carbon Gen 9 20XW002BMX
BIOS           = N32ET48W (1.24 )
OS Release     = Manjaro Linux
Kernel         = 5.12.9-1-MANJARO #1 SMP PREEMPT Thu Jun 3 14:56:42 UTC 2021 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.12-x86_64 root=UUID=1c942e93-5a0b-42ef-9750-6784a8b083fc rw quiet splash apparmor=1 security=apparmor resume=UUID=48eb6ffb-5e8d-47b9-89d6-5444657ed8d5 udev.log_priority=3
Init system    = systemd
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = not installed
Last run       = 22:32:55,    140 sec(s) ago
Mode           = battery
Power source   = battery

Warning: systemd-rfkill.service is not masked -- invoke "systemctl mask systemd-rfkill.service" to correct this!
Warning: systemd-rfkill.socket is not masked -- invoke "systemctl mask systemd-rfkill.socket" to correct this!

+++ Processor
CPU model      = 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  4200000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference = balance_power [HWP.EPP]
/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power

/sys/devices/system/cpu/cpu1..cpu7: omitted for clarity, use -v to show all

/sys/devices/system/cpu/intel_pstate/status            = active
/sys/devices/system/cpu/intel_pstate/min_perf_pct      =   9 [%]
/sys/devices/system/cpu/intel_pstate/max_perf_pct      = 100 [%]
/sys/devices/system/cpu/intel_pstate/no_turbo          =   0
/sys/devices/system/cpu/intel_pstate/hwp_dynamic_boost =   0
/sys/devices/system/cpu/intel_pstate/turbo_pct         =  47 [%]
/sys/devices/system/cpu/intel_pstate/num_pstates       =  39

/sys/module/workqueue/parameters/power_efficient       = Y
/proc/sys/kernel/nmi_watchdog                          = 0

+++ Platform Profile
/sys/firmware/acpi/platform_profile                    = balanced
/sys/firmware/acpi/platform_profile_choices            = low-power balanced performance
```

AC:
```
--- TLP 1.4.0-alpha.2 --------------------------------------------

+++ Configured Settings (only differences to defaults):
/etc/tlp.conf L0168: PLATFORM_PROFILE_ON_AC="performance"
/etc/tlp.conf L0169: PLATFORM_PROFILE_ON_BAT="balanced"

+++ System Info
System         = LENOVO ThinkPad X1 Carbon Gen 9 20XW002BMX
BIOS           = N32ET48W (1.24 )
OS Release     = Manjaro Linux
Kernel         = 5.12.9-1-MANJARO #1 SMP PREEMPT Thu Jun 3 14:56:42 UTC 2021 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.12-x86_64 root=UUID=1c942e93-5a0b-42ef-9750-6784a8b083fc rw quiet splash apparmor=1 security=apparmor resume=UUID=48eb6ffb-5e8d-47b9-89d6-5444657ed8d5 udev.log_priority=3
Init system    = systemd
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = not installed
Last run       = 22:37:45,      7 sec(s) ago
Mode           = AC
Power source   = AC

Warning: systemd-rfkill.service is not masked -- invoke "systemctl mask systemd-rfkill.service" to correct this!
Warning: systemd-rfkill.socket is not masked -- invoke "systemctl mask systemd-rfkill.socket" to correct this!

+++ Processor
CPU model      = 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  4200000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference = balance_performance [HWP.EPP]
/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power

/sys/devices/system/cpu/cpu1..cpu7: omitted for clarity, use -v to show all

/sys/devices/system/cpu/intel_pstate/status            = active
/sys/devices/system/cpu/intel_pstate/min_perf_pct      =   9 [%]
/sys/devices/system/cpu/intel_pstate/max_perf_pct      = 100 [%]
/sys/devices/system/cpu/intel_pstate/no_turbo          =   0
/sys/devices/system/cpu/intel_pstate/hwp_dynamic_boost =   0
/sys/devices/system/cpu/intel_pstate/turbo_pct         =  47 [%]
/sys/devices/system/cpu/intel_pstate/num_pstates       =  39

/sys/module/workqueue/parameters/power_efficient       = Y
/proc/sys/kernel/nmi_watchdog                          = 0

+++ Platform Profile
/sys/firmware/acpi/platform_profile                    = performance
/sys/firmware/acpi/platform_profile_choices            = low-power balanced performance
```

linrunner, 2021-06-05:
Very well. Thank you for testing.

Offtopic question: are you happy with the X1C9 running Linux?

ErikReider, 2021-06-05:
@linrunner I'm loving it! Sadly it isn't supported by lvfs yet but everything else works as expected, even the fingerprint reader 😄

linrunner, 2021-06-05:
Great. I read a rant about the keyboard being worse than the previous generations. What's your impression?

No LVFS, are you sure?
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN32ETXXW.firmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN32HT.firmware

ErikReider, 2021-06-05:
The only other laptop keyboards that I've used are MacBook(old air), some asus Chromebook, my old Dell latitude 5480 and a HP probook 470 (all school laptops). I quite like it. It's not as much travel as the dell but its a lot more precise and tactile.

Typical that I checked the lvfs status the day before it was released 😂

linrunner, 2021-06-07:
Thanks for your assessment. I'll close now.

closed 2021-06-07 by linrunner