linrunner/TLP #564: TLP conflicts with power-profiles-daemon package

mhadidg, 2021-07-06:
**Describe the bug**

After I upgraded to Ubuntu 21.04. I noticed that `tlp.service` was not activated on system startup. It turns out that the service unit `power-profiles-daemon.service` of the new [power-profiles-daemon](https://gitlab.freedesktop.org/hadess/power-profiles-daemon) package is the problem. It's declaring a conflict with `tlp.service` causing `tlp.service` to be effectively disabled.

Disabling `power-profiles-daemon.service` or removing power-profiles-daemon package resolves the issue for me.

This package is installed by default in [Ubuntu 21.04+](https://packages.ubuntu.com/search?keywords=power-profiles-daemon&searchon=names) and [some other distributions](https://www.reddit.com/r/Fedora/comments/mhd82q/suddenly_seeing_new_power_mode_options_in_settings/) as well.

_/lib/systemd/system/power-profiles-daemon.service_ (notice the _Conflicts_ directive)
```
[Unit]
Description=Power Profiles daemon
Conflicts=tuned.service tlp.service auto-cpufreq.service
Before=multi-user.target display-manager.target

...
```

**Expected behavior**

- The documentation may include a note under [Installation > Ubuntu](https://linrunner.de/tlp/installation/ubuntu.html#ubuntu) and [FAQ > Installation](https://linrunner.de/tlp/faq/installation.html) sections explaining that a user must either disable `power-profiles-daemon.service` or remove power-profiles-daemon package for TLP to work properly; or
- TLP service unit may include `Conflicts=power-profiles-daemon.service` directive. This will resolve the issue in the current setup of both services, as `tlp.service` is activated before `power-profiles-daemon.service`.
- Optionally, configure TLP service unit with `Before=power-profiles-daemon.service` to make sure that `tlp.service` activation will always take precedence over `power-profiles-daemon.service`.

**To Reproduce**

1. Install power-profiles-daemon package.
2. Reboot.
3. Inspect the runtime status of `tlp.service` unit.

linrunner, 2021-07-06:
Hi, thanks for the notification and your suggestions.

Would you please also open a Ubuntu bug about this? Canonical's oem team will be interested. They use TLP in hardware enablement for vendor specific power saving settings (afaik Dell and maybe others) and made TLP be included in the main repository some time ago.

In the meantime, I will consider which of your suggestions I consider best for "peaceful coexistence".

linrunner, 2021-07-06:
Btw: disabling tlp.service does not completely disable TLP. Only application of settings at boot is omitted (unfortunately also affects the charging thresholds). Next time the power source changes TLP's settings will take effect.

See https://linrunner.de/tlp/introduction.html#how-it-works

mhadidg, 2021-07-07:
An issue ([bug #1934944](https://bugs.launchpad.net/ubuntu/+source/tlp/+bug/1934944)) has been reported to Ubuntu TLP package maintainer.

linrunner, 2021-07-22:
> TLP service unit may include Conflicts=power-profiles-daemon.service directive. 

Finally took the time to setup up a 21.04 system and can confirm this works as expected.

> This will resolve the issue in the current setup of both services, as tlp.service **is activated before** power-profiles-daemon.service.

But I wonder _why_ it works. tlp.service is (by design) started very late in the boot process, whereas power-profiles-daemon has no such dependencies.

**tlp.service**

> [Unit]
> Description=TLP system startup/shutdown
> Conflicts=power-profiles-daemon.service
> **After=multi-user.target NetworkManager.service**
> Before=shutdown.target

**power-profiles-daemon.service** 

> [Unit]
> Description=Power Profiles daemon
> Conflicts=tuned.service tlp.service auto-cpufreq.service
> Before=multi-user.target display-manager.target

linrunner, 2021-07-22:
CORRECTION: actually it only worked once. for whatever reason. Neither *Conflicts=power-profiles-daemon.service* nor an additional *Before=power-profiles-daemon.service* start tlp.service reliably.

alex-tu-cc, 2021-08-30:
Hi @linrunner,
Thanks for reminder of this conflict.
The conflict was introduced from new gnome version and indeed will impact users who are counting on TLP for power saving.

Excuse me, but did you discuss the topic of how to keep the existed power saving experience offered by TLP on power-profiles-daemon with the power-profiles-daemon maintainer?

I am wondering if the maintainer consulted with you when he was writing the known conflict on [the readme of project](https://gitlab.freedesktop.org/hadess/power-profiles-daemon).

And is it a plan to migrate the existed functions to power-profiles-daemon or the plan should be to make power-profiles-daemon and tlp daemon compatible? How do you think?

alex-tu-cc, 2021-08-31:
@linrunner How do you think to prevent tlp daemon changing the value of platform_profile? So that power-profiles-daemon and tlp daemon can co-exist.

linrunner, 2021-08-31:
Hi @alex-tu-cc , thank you for caring.

> Excuse me, but did you discuss the topic of how to keep the existed power saving experience offered by TLP on power-profiles-daemon with the power-profiles-daemon maintainer?

Not yet, but still on my list. I'm busy with 1.4 right now.

> I am wondering if the maintainer consulted with you when he was writing the known conflict on the readme of project.

I have not heard from Sebastien yet.

> And is it a plan to migrate the existed functions to power-profiles-daemon or the plan should be to make power-profiles-daemon and tlp daemon compatible? How do you think?

I'm not aware of Sebastien's plans. I myself have no plans to contribute to p-p-d at the moment. I think coexistence would be the best way in the medium term.

Last time I looked into the p-p-d sources it has controlled exactly two kernel features:

1. /sys/devices/system/cpu/cpufreq/energy_performance_preference - TLP 1.3.1: CPU_ENERGY_PERF_POLICY_ON_AC/BAT (activated in default.conf)
2. /sys/firmware/acpi/platform_profile - TLP 1.4: PLATFORM_PROFILE_ON_AC/BAT (not activated in default.conf, so  no immeditate conflict)

Imho p-p-d is currently geared towards limiting power consumption under high or medium CPU loads, while TLP also takes care of idle and low load situations.

linrunner, 2021-08-31:
> @linrunner How do you think to prevent tlp daemon changing the value of platform_profile? So that power-profiles-daemon and tlp daemon can co-exist.

1. Option: TLP detects the running p-p-d and doesn't touch the above mentioned settings
2. Option: p-p-d detects the situation and warns the user

Or a synthesis of both  ...

linrunner, 2021-08-31:
My current approach is to mask **power-profiles-daemon.service** in package postinst. I'll propose just that to the Debian maintainer: https://github.com/linrunner/TLP/commit/6a9388e1af95051a90a33b4014af1158dfa241f6

fthx, 2021-09-19:
Can confirm this problem on Ubuntu 21.10.
https://discourse.ubuntu.com/t/tlp-vs-power-profiles/24266

plumlis, 2021-09-19:
I'd like to know what power-profiles do to save battery.

Just like TLP to run some script kernel tweak? Or some hardware level tweak?

I tried power-profiles for a week( Removed TLP)  and found it doesn't works well, Only 3 hours on my laptop（ 4-5 huors with TLP）

It seems Gnome 41 will enable power-profiles by default.

fthx, 2021-09-19:
I tested PP without TLP and it does not activate power savings, says powertop.
So now I simply uninstalled PP and TLP runs fine as usual.

linrunner, 2021-09-19:
In the meantime, the [TLP installation guide](https://linrunner.de/tlp/installation/ubuntu.html) tells you how to deactivate p-p-d's service. TLP 1.4 packages will contain it.

p-p-d writes to the same sysfiles as all other PM tools, I have indicated them [above](https://github.com/linrunner/TLP/issues/564#issuecomment-909116794). Maybe powertop doesn't cover them yet.

linrunner, 2021-09-30:
> My current approach is to mask _power-profiles-daemon.service_ in the deb postinst. I'll propose just that to the Debian maintainer: [6a9388e](https://github.com/linrunner/TLP/commit/6a9388e1af95051a90a33b4014af1158dfa241f6)

This has been implemented in Debian Sid in the meantime and also in the Ubuntu PPA.

seb128, 2021-10-14:
No, I meant that it has been advertised in upstream release notes and documentation as a feature added which is closing a gap compared to windows for many user. Your screenshot confirm the point, the option is vanishing without giving users any clue to help understand why

seb128, 2021-10-14:
>  We run 0.8.1 in Impish.

0.9 came after feature freeze is includes some changes that didn't make it suitable. We could patch the polkit authorization improvements if needed but the bug didn't get flagged as a security issue nor assigned a CVE. Anyway the TLP github is the wrong place for that topic, please open rather a bug on launchpad if you think Ubuntu impish needs to be updated

fthx, 2021-10-14:
There was two parts in my message. The first one was about conflicting not only in TLP deb package.

(For the second part I raised:
https://bugs.launchpad.net/ubuntu/+source/power-profiles-daemon/+bug/1947152
set as private bug, but the source message is not...
I stop here with PPD-specific stuff, this package is anyway quite useless for my laptop power savings.)

linrunner, 2021-12-11:
> Those do conflict so the right thing to do would be for the packages to **use a Conflicts** so users are not able to install them at the same time which is what leads to the nondeterministic behavior, instead of having units and postinst script conflicting in the background.

I have now included this in the notes on packaging. Thanks for your feedback.

linrunner, 2022-01-04:
The author of _power-profiles-daemon_ does not consider compatibility with TLP to be feasible [1]. Therefore I have consequently included the following in the dependencies [2]:

> **power-profiles-daemon - conflicts** [from 1.5]
>   There can only be one power management tool at a time.

[1] https://gitlab.freedesktop.org/hadess/power-profiles-daemon/-/issues/54
[2] https://linrunner.de/tlp/developers/dependencies.html

closed 2022-01-04 by linrunner