linrunner/TLP #565: acpi_call module can't build in Kernel 5.13

plumlis, 2021-07-10:
Fedora 34 64 bit

Recently I upgrade kernel to 5.13 and found akmod_apci_call build faild.
Maybe should be a new version?

linrunner, 2021-07-14:
Would you like to show the error message? Unfortunately, the issue glass globe is defective atm.

ValdikSS, 2021-08-01:
@linrunner please switch to newer acpi_call fork maintained by Nix project: https://github.com/nix-community/acpi_call
Your (old) module builds fine for me, but it oopses because its procfs does not support seeking, which is now mandatory for 5.13. Nix project fork is already fixed.

linrunner, 2021-08-01:
There are updated packages provided by @Thorashh in the [_tlp-updates-testing_ repository](https://linrunner.de/tlp/installation/fedora.html#thinkpads-only). Please test and report.

ValdikSS, 2021-08-01:
@linrunner works fine for me, does not oops.

linrunner, 2021-08-01:
@ValdikSS : thanks.

closed 2021-08-04 by linrunner