linrunner/TLP #566: [ThinkPad T495s Coreboot] Unable to get recalibrate to work

pachadotdev, 2021-07-28:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

Hi

Yesterday I installed a new battery for the T495s as I got one under guarantee conditions.

After installing it, the technician told me to charge the laptop to 100%, and then recalibrate the battery.

I did that but I get

```
Currently discharging battery BAT0:
voltage            =  10523 [mV]
remaining capacity =    840 [mWh]
remaining percent  =      1 [%]
remaining time     =      2 [min]
power              =  22813 [mW]
state              = Discharging
force discharge    = 1
Press Ctrl+C to cancel.
Error: battery BAT0 was not discharged completely. Check your hardware.
```

The charger was connected during the process and all the BIOS checks at the service center were ok after installing the battery.

Is this wrong? can I just uninstall tlp? this is one of the certified models for Linux, and I'm not really a techie user, just a statistician trying to work without problems.

This is the output of tlp-stat now after not discharging completely (I took it ~15 minutes later):

```
+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = active (data, thresholds)
tpacpi-bat = active (recalibrate)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 5B10W139
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  57000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  57000 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  29800 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  32543 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

/sys/class/power_supply/BAT0/charge_start_threshold         =     96 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   52.3 [%]
Capacity                                                    =  100.0 [%]

00
comments
```

A clear and concise description of what the bug is.

**Expected behavior**

Discharge the battery completely

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? AC, the process requires AC
2. Actions to reproduce the behaviour. Turn on the laptop and run `sudo tlp recalibrate`
3. Shell commands entered and their output: `sudo tlp recalibrate` (the output is in the bug description) and then `sudo tlp-stat -b`
[... 256 lines cut ...]
natacpi    = active (data, thresholds)
tpacpi-bat = active (recalibrate)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 5B10W139
/sys/class/power_supply/BAT0/cycle_count                    =      1
/sys/class/power_supply/BAT0/energy_full_design             =  57000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  57000 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  42790 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  26238 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

/sys/class/power_supply/BAT0/charge_start_threshold         =     96 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   75.1 [%]
Capacity                                                    =  100.0 [%]
```

**Additional context**

Add any other context about the problem here.

linrunner, 2021-07-28:
First of all: this is not a TLP bug. The issue tracker is not a support forum and not intended for advice on (battery) hardware problems. Please consult a hardware oriented forum of your choice for further questions.

Background: TLP displays the message "Error: battery BAT0 was not discharged completely. Check your hardware." if the Lenovo firmware terminates the discharge process before reaching 0%. Which means that the recalibration of the charge gauge in the battery pack didn't complete. 

In your case it terminated at 1% which may well be a rounding error in the firmware. If the battery otherwise performs well, you may decide to ignore it.

linrunner, 2021-07-28:
Note @self: add to FAQ

linrunner, 2021-07-28:
I stand corrected: it could be that I was wrong in my initial assessment of your problem. See my [commit](https://github.com/linrunner/TLP/commit/c95568a433ed3f518b1298f5e8a47aaa653a1cab).

You can find packages on the [download page](https://download.linrunner.de/packages/). Please test and report.

linrunner, 2021-08-04:
@pachadotdev : did you test the fix?

erazemk, 2021-08-10:
This might be partially related to my issue (#571), where tlp can't recalibrate a Thinkpad T540p. What do you think would be the best way for me to test your fix? I'm on Fedora, so the packages aren't available yet, but I can build it from source if needed.

pachadotdev, 2021-08-10:
@linrunner hi, i shall test this week, the laptop ended with a replaced mainboard and battery, because it failed both on windows and ubuntu so warranty applied

pachadotdev, 2021-08-16:
@linrunner hi, same problem, but this unit has a factory problem or something that it reboots at random
```
Currently discharging battery BAT0:
voltage            =  10614 [mV]
remaining capacity =    890 [mWh]
remaining percent  =      1 [%]
remaining time     =      3 [min]
power              =  15050 [mW]
state              = Discharging
force discharge    = 1
Press Ctrl+C to cancel.
Error: battery BAT0 was not discharged completely i.e. terminated by the firmware -- check your hardware (battery, charger).
Battery recalibration aborted.
```

linrunner, 2021-08-17:
I see. The possibilities to change TLP's code are exhausted, if your hardware stops at 1% instead of 0% then so be it. 

I will however add an explanation of the message to the FAQ..

linrunner, 2021-09-15:
Done: https://linrunner.de/tlp/faq/battery.html#tlp-recalibrate-terminates-with-an-error-message

closed 2021-09-15 by linrunner