linrunner/TLP #567: [TLP 1.4 Beta] Call for testers

linrunner, 2021-08-01:
Hi @all,

as with every upcoming release i need your help to complete the test coverage.

**Test Objectives**
- Testing of new and changed features that interest you → see [issues which need a test](https://github.com/linrunner/TLP/issues?q=is%3Aopen+is%3Aissue+label%3A%22needs+testing%22), [release notes](https://github.com/linrunner/TLP/releases) and [changelog](https://github.com/linrunner/TLP/blob/main/changelog)
-  General regression testing (check if everything still works for you)
- Battery care
  -  Testing of charge thresholds for ASUS, Huawei, LG, Lenovo (non-ThinkPad series) and Samsung laptops → see [special instructions](https://gist.github.com/linrunner/2ead4b591eed33055cf86a38ccc73949)
  -  Regression testing of ThinkPad charge thresholds (check if everything still works for you)

**Beta Packages**
See the [Download page](https://download.linrunner.de/packages/).

**What do I need from the testers?**
After installing please start the beta with

    sudo tlp start

I'm expecting your unabbreviated outputs of

    sudo tlp-stat

on battery and AC (via [Gist](https://gist.github.com/) please).

**Bug Reports**
If something does not work as expected → please open an [issue](https://github.com/linrunner/TLP/issues).

Thanks for testing!

pflanzenandi, 2021-08-04:
[AC](https://gist.github.com/pflanzenandi/0f1f849c2fe47cd64f6fb2bb47116b5b)
[BAT](https://gist.github.com/pflanzenandi/938de7bec6c8fbdfd64f15c16bc39700)

linrunner, 2021-08-04:
@pflanzenandi : looks good. Thanks!

BarbUk, 2021-08-04:
[AC](https://gist.github.com/BarbUk/6db4f918cbbd8e30790e161d47a5f5a8)
[BAT](https://gist.github.com/BarbUk/fcf44d8e8ed6570ec46fdf7450f5e47e)

BarbUk, 2021-08-04:
x220

[AC](https://gist.github.com/BarbUk/c854a7764e645c925f08dab68a2d6603#file-ac) 
[BAT](https://gist.github.com/BarbUk/c854a7764e645c925f08dab68a2d6603#file-bat)

parkerlreed, 2021-08-04:
[AC](https://gist.github.com/531272da9a95a5235ce2a77f3c508796)
[BAT](https://gist.github.com/3e7caaf6d42aee80106c4b7bd41c2785)

Thinkpad T495

sodaplayer, 2021-08-05:
[AC](https://gist.github.com/sodaplayer/14a778f11efcfd5feee19d76658f93af)
[BAT](https://gist.github.com/sodaplayer/e1fbf1258bd48fb62ff1f91e7c3cbc12)

linrunner, 2021-08-05:
Thanks for your help!

pflanzenandi, 2021-08-06:
XPS 15 9500

[AC](https://gist.github.com/pflanzenandi/75ed42811699e72c129098c942ac3707)
[BAT](https://gist.github.com/pflanzenandi/1cec0a6f10690703a6e06378b6a14a7e)

Korna, 2021-08-08:
[AC](https://gist.github.com/Korna/21269dada0af031b46816c6f8d53a867)
[BAT}(https://gist.github.com/Korna/9960e3c5e9671843ddb9ff8e8c01c043)

morialo, 2021-09-03:
Gigabyte Aorus 17G YC

AC 
[https://gist.github.com/morialo/2180c6921c91e32f13def06b4295c713](url)

BAT
[https://gist.github.com/morialo/cc5ae18dac27dc76a0599e346b380934](url)

linrunner, 2021-09-05:
Thank you all for testing :+1:

linrunner, 2021-09-10:
Beta 2 is [out](https://linrunner.de/tlp/news.html#news-top-1). Enjoy!

pflanzenandi, 2021-09-12:
[AC](https://gist.github.com/pflanzenandi/d90bfe17166ef97f2334ea5069c4da4d)
[BAT](https://gist.github.com/pflanzenandi/0df940c8c816e97965ca6c16ed233a65)

linrunner, 2021-09-24:
Beta test is completed - and [TLP 1.4 finally released](https://linrunner.de/tlp/news.html#tlp-1-4-released). Have fun!

closed 2021-09-24 by linrunner

linrunner, 2021-12-21:
TLP 1.5 Beta 1 is out. Testers welcome -> https://github.com/linrunner/TLP/issues/609

linrunner, 2023-07-22:
Hi @ALL : TLP 1.6 Beta 1 is out. Testers wanted -> https://github.com/linrunner/TLP/issues/700