linrunner/TLP #568: 1.4.1 beta LG laptop (2021) battery care value not showing properly

sutram, 2021-08-01:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

/sys/devices/platform/lg-laptop/battery_care_limit          = 0 [%]

A clear and concise description of what the bug is.

/sys/devices/platform/lg-laptop/battery_care_limit          = 0 [%]

A clear and concise description of what you expected to happen.

1) Upgraded TLP on Manjaro Linux to 1.4.1 beta using tlp-git and tlp-rw-git in AUR
2) Uncommented STOP_CHARGE_THRESH_BAT0="80" in /etc/tlp.conf and saved the file
3) Ran tlp start
```
sudo tlp start                                                                                                                                                 ✔  4s  
Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
>>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
>>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
TLP started in battery mode (auto).

```
4) Ran sudo tlp-stat -b -s -c.

Expected /sys/devices/platform/lg-laptop/battery_care_limit          = 80 [%]

Got

/sys/devices/platform/lg-laptop/battery_care_limit          = 0 [%]

1. Does the problem occur on battery or AC or both? Both
2. Actions to reproduce the behaviour (see above)
3. Shell commands entered and their output
    Also checked that I had the lg_laptop driver installed
   ```
  sudo lsmod | grep lg_laptop                                                                                                                                           ✔ 
lg_laptop              20480  0
sparse_keymap          16384  1 lg_laptop
wmi                    36864  2 lg_laptop,wmi_bmof

   ```
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**

https://gist.github.com/sutram/5c1ac71a9d13da17b5bec23f9111956a

**Additional context**

Add any other context about the problem here.

linrunner, 2021-08-01:
Hi,

please configure trace mode with:

    TLP_DEBUG="bat ps run"

Run the command

    sudo tlp setcharge

show its full output and also show the trace output

     sudo tlp-stat -T

sutram, 2021-08-01:
```
TLP_DEBUG="bat ps run"                                                                                                                                                                              

sudo tlp setcharge                                                                                                                                                                                 
Setting temporary charge threshold for all batteries:
  care limit = 80

sudo tlp-stat -T                                                                                                                                                                                
--- TLP 1.4.0-beta.1 --------------------------------------------

-- Journal begins at Tue 2021-07-27 00:11:41 MDT, ends at Tue 2021-08-31 11:11:45 MDT. --
Jul 27 04:10:37 vajra tlp[1087]: Applying power save settings...done.
Jul 27 04:10:37 vajra tlp[1087]: Setting battery charge thresholds...done.
Aug 01 10:19:58 vajra tlp[2986]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:19:58 vajra tlp[2986]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:19:58 vajra tlp[2986]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:19:58 vajra tlp[2986]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:26:51 vajra tlp[3799]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:26:51 vajra tlp[3799]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:26:51 vajra tlp[3799]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:26:51 vajra tlp[3799]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:33:28 vajra tlp[4795]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:33:28 vajra tlp[4795]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:33:28 vajra tlp[4795]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:33:28 vajra tlp[4795]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:40:54 vajra tlp[5047]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:40:54 vajra tlp[5047]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:40:54 vajra tlp[5047]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:40:54 vajra tlp[5047]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:41:10 vajra tlp[5085]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:41:10 vajra tlp[5085]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:41:10 vajra tlp[5085]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:41:10 vajra tlp[5085]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:42:54 vajra tlp[5639]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:42:54 vajra tlp[5639]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:42:54 vajra tlp[5639]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:42:54 vajra tlp[5639]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:43:24 vajra tlp[5695]: Applying power save settings...done.
Aug 01 10:43:24 vajra tlp[5695]: Setting battery charge thresholds...done.
Aug 01 10:44:15 vajra tlp[5894]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:44:15 vajra tlp[5894]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:44:15 vajra tlp[5894]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:44:15 vajra tlp[5894]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:45:36 vajra tlp[6403]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:45:36 vajra tlp[6403]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:45:36 vajra tlp[6403]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:45:36 vajra tlp[6403]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:46:12 vajra tlp[6530]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:46:12 vajra tlp[6530]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:46:12 vajra tlp[6530]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:46:12 vajra tlp[6530]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:46:39 vajra tlp[6743]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
[... cut ...]
Aug 01 11:23:04 vajra tlp[10274]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 11:31:28 vajra tlp[10559]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 11:31:28 vajra tlp[10559]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 11:31:28 vajra tlp[10559]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 11:31:28 vajra tlp[10559]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
```

linrunner, 2021-08-01:
The trace shows nothing. Did you add the line 

    TLP_DEBUG="bat ps run" 

to your configuration?  A quick way is (using a root shell):

    sudo -s
    echo 'TLP_DEBUG="bat ps run" ' > /etc/tlp.d/00-debug.conf

sutram, 2021-08-01:
Sorry, here is the latest output. It has more information now. Let me know if you need anything else. Thanks.
```
--- TLP 1.4.0-beta.1 --------------------------------------------

-- Journal begins at Tue 2021-07-27 00:11:41 MDT, ends at Tue 2021-08-31 11:11:45 MDT. --
Jul 27 04:10:37 vajra tlp[1087]: Applying power save settings...done.
Jul 27 04:10:37 vajra tlp[1087]: Setting battery charge thresholds...done.
Aug 01 10:19:58 vajra tlp[2986]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:19:58 vajra tlp[2986]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:19:58 vajra tlp[2986]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:19:58 vajra tlp[2986]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:26:51 vajra tlp[3799]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:26:51 vajra tlp[3799]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:26:51 vajra tlp[3799]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:26:51 vajra tlp[3799]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:33:28 vajra tlp[4795]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:33:28 vajra tlp[4795]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:33:28 vajra tlp[4795]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:33:28 vajra tlp[4795]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:40:54 vajra tlp[5047]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:40:54 vajra tlp[5047]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:40:54 vajra tlp[5047]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:40:54 vajra tlp[5047]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:41:10 vajra tlp[5085]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:41:10 vajra tlp[5085]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:41:10 vajra tlp[5085]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:41:10 vajra tlp[5085]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:42:54 vajra tlp[5639]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:42:54 vajra tlp[5639]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:42:54 vajra tlp[5639]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:42:54 vajra tlp[5639]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:43:24 vajra tlp[5695]: Applying power save settings...done.
Aug 01 10:43:24 vajra tlp[5695]: Setting battery charge thresholds...done.
Aug 01 10:44:15 vajra tlp[5894]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:44:15 vajra tlp[5894]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:44:15 vajra tlp[5894]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:44:15 vajra tlp[5894]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:45:36 vajra tlp[6403]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:45:36 vajra tlp[6403]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:45:36 vajra tlp[6403]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:45:36 vajra tlp[6403]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:46:12 vajra tlp[6530]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:46:12 vajra tlp[6530]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:46:12 vajra tlp[6530]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:46:12 vajra tlp[6530]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:46:39 vajra tlp[6743]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:46:39 vajra tlp[6743]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 10:46:39 vajra tlp[6743]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 10:46:39 vajra tlp[6743]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 10:47:52 vajra tlp[6945]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 10:47:52 vajra tlp[6945]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
[... cut ...]
Aug 01 12:07:36 vajra tlp[11884]: batdrv_init.lg: batteries=; natacpi=0; thresh=natacpi; bf_stop=/sys/devices/platform/lg-laptop/battery_care_limit
Aug 01 12:07:36 vajra tlp[11884]: batdrv.lg.read_threshold(stop): bf_stop=/sys/devices/platform/lg-laptop/battery_care_limit; out=0; rc=0
Aug 01 12:07:36 vajra tlp[11884]: batdrv.lg.write_thresholds(80, 1, BAT0).write: old=0; new=80; rc=0
Aug 01 12:07:36 vajra tlp[11884]: Setting battery charge thresholds...done.
Aug 01 12:07:36 vajra tlp[11884]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpEsl1Pt -> /run/tlp/run.conf
```

linrunner, 2021-08-01:
The trace shows that the kernel consumes the threshold without error

> Aug 01 12:07:36 vajra tlp[11884]: batdrv.lg.write_thresholds(80, 1, BAT0).write: old=0; new=80; **rc=0**

but the firmware seems to silently discard it. 

The trace also shows the whole thing happened on battery power. Could you try again on AC power? You don't need to boot, just issue the setcharge command:

    sudo tlp setcharge 0 80
    sudo tlp-stat -T

I only need the new part of the trace (look at the timestamp).

sutram, 2021-08-01:
This is the new stuff in the log...
```
Aug 01 12:32:03 vajra tlp[13489]: +++ auto (1.4.0-beta.1) ++++++++++++++++++++++++++++++++++++++++
Aug 01 12:32:03 vajra tlp[13489]: get_sys_power_supply(ADP1).ac_online: syspwr=0
Aug 01 12:32:03 vajra tlp[13489]: power_source=ac
Aug 01 12:32:03 vajra tlp[13489]: manual_mode=none
Aug 01 12:32:03 vajra tlp[13489]: power_mode=ac
Aug 01 12:32:03 vajra tlp[13489]: Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
Aug 01 12:32:03 vajra tlp[13489]: >>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Aug 01 12:32:03 vajra tlp[13489]: Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
Aug 01 12:32:03 vajra tlp[13489]: >>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
Aug 01 12:32:03 vajra tlp[13489]: compare_and_save_power_state(0).different: old=1
Aug 01 12:32:04 vajra tlp[13489]: save_runconf.ok: /run/tlp/tlp-run.conf_tmp5CQJMY -> /run/tlp/run.conf
Aug 01 12:32:23 vajra tlp[13603]: +++ setcharge (1.4.0-beta.1) ++++++++++++++++++++++++++++++++++++++++
Aug 01 12:32:23 vajra tlp[13603]: get_sys_power_supply(ADP1).ac_online: syspwr=0
Aug 01 12:32:23 vajra tlp[13603]: power_source=ac
Aug 01 12:32:23 vajra tlp[13603]: manual_mode=none
Aug 01 12:32:23 vajra tlp[13603]: power_mode=ac
Aug 01 12:32:23 vajra tlp[13603]: check_thinkpad.not_a_thinkpad: model=0.1
Aug 01 12:32:23 vajra tlp[13603]: batdrv_init.thinkpad.not_a_thinkpad
Aug 01 12:32:23 vajra tlp[13603]: check_thinkpad.not_a_thinkpad: model=0.1
Aug 01 12:32:23 vajra tlp[13603]: batdrv_init.thinkpad-legacy.not_a_thinkpad
Aug 01 12:32:23 vajra tlp[13603]: batdrv_init.asus.no_match
Aug 01 12:32:23 vajra tlp[13603]: batdrv_init.huawei.no_match
Aug 01 12:32:23 vajra tlp[13603]: batdrv_init.lenovo.no_match
Aug 01 12:32:23 vajra tlp[13603]: batdrv_init.samsung.no_match
Aug 01 12:32:23 vajra tlp[13603]: batdrv_init.lg: batteries=; natacpi=0; thresh=natacpi; bf_stop=/sys/devices/platform/lg-laptop/battery_care_limit
Aug 01 12:32:23 vajra tlp[13603]: batdrv.lg.select_battery(DEF): bat_str=; bd_read=/sys/class/power_supply/;
Aug 01 12:32:23 vajra tlp[13603]: batdrv.lg.read_threshold(stop): bf_stop=/sys/devices/platform/lg-laptop/battery_care_limit; out=0; rc=0
Aug 01 12:32:23 vajra tlp[13603]: batdrv.lg.write_thresholds(80, 2, ).write: old=0; new=80; rc=0
```

linrunner, 2021-08-01:
Thanks. I forgot to request another

    sudo tlp-stat -s -b

sutram, 2021-08-01:
```
--- TLP 1.4.0-beta.1 --------------------------------------------

+++ System Info
System         = LG Electronics 0.1 16Z90P-K.AAC7U1
BIOS           = T2ZF0320 X64
OS Release     = Manjaro Linux
Kernel         = 5.10.53-1-MANJARO #1 SMP PREEMPT Mon Jul 26 07:18:28 UTC 2021 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.10-x86_64 root=UUID=7473c82f-0fa6-43f3-a68a-6a39a7f2c2ba rw quiet splash apparmor=1 security=apparmor udev.log_priority=3
Init system    = systemd 
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 01:24:46 PM,     21 sec(s) ago
Mode           = AC
Power source   = AC

Warning: systemd-rfkill.service is not masked, radio device switching may not work as configured.
>>> Invoke 'systemctl mask systemd-rfkill.service' to correct this.
Warning: systemd-rfkill.socket is not masked, radio device switching may not work as configured.
>>> Invoke 'systemctl mask systemd-rfkill.socket' to correct this.
+++ Battery Care
Plugin: lg
Supported features: charge threshold
Driver usage:
* vendor (lg_laptop) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0: 80(on), 100(off) -- battery care limit

/sys/devices/platform/lg-laptop/battery_care_limit          = 0 [%]

```

linrunner, 2021-08-01:
OK, the code now detects when the write is silently discarded plus it detects batteries named CMB[01]. Please rebuild the tlp-git AUR package to fetch the fixes for the LG plugin.

Then show:

    sudo tlp setcharge 0 80
    sudo tlp-stat -s -b 
    sudo tlp-stat -T # only the new part please

But the downside is: i can't fix the write problem in TLP, it's a kernel issue. 

You'll have to file a kernel bug about the write to **/sys/devices/platform/lg-laptop/battery_care_limit** being discarded on your hardware without error.

sutram, 2021-08-02:
Seems like writing the value failed instead of being silently discarded. Also, when I rebuilt tlp-git, it thought there was a conflict with tlp-rdw-git and wanted to remove that so I let it remove that package.

sudo tlp setcharge 0 80 
```                                                                                                                                                                          
Setting temporary charge threshold for all batteries:
  care limit = 80 (Error: write failed)
```
sudo tlp-stat -b -c -s 
```
--- TLP 1.4.0-beta.1 --------------------------------------------

+++ Configured Settings:
defaults.conf L0004: TLP_ENABLE="1"
defaults.conf L0005: TLP_WARN_LEVEL="3"
defaults.conf L0006: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0008: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0010: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0012: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0013: SCHED_POWERSAVE_ON_AC="0"
defaults.conf L0014: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0015: NMI_WATCHDOG="0"
defaults.conf L0016: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0017: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0018: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0019: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0020: DISK_IOSCHED="keep keep"
defaults.conf L0021: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0022: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
defaults.conf L0023: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0025: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0026: PCIE_ASPM_ON_AC="default"
defaults.conf L0027: PCIE_ASPM_ON_BAT="default"
defaults.conf L0028: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0029: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0030: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0031: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0032: WIFI_PWR_ON_AC="off"
defaults.conf L0033: WIFI_PWR_ON_BAT="on"
defaults.conf L0034: WOL_DISABLE="Y"
defaults.conf L0035: SOUND_POWER_SAVE_ON_AC="1"
defaults.conf L0036: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0037: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0038: BAY_POWEROFF_ON_AC="0"
defaults.conf L0039: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0040: BAY_DEVICE="sr0"
defaults.conf L0041: RUNTIME_PM_ON_AC="on"
defaults.conf L0042: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0043: RUNTIME_PM_DRIVER_DENYLIST="mei_me nouveau radeon"
defaults.conf L0044: USB_AUTOSUSPEND="1"
defaults.conf L0045: USB_EXCLUDE_AUDIO="1"
defaults.conf L0046: USB_EXCLUDE_BTUSB="0"
defaults.conf L0047: USB_EXCLUDE_PHONE="0"
defaults.conf L0048: USB_EXCLUDE_PRINTER="1"
defaults.conf L0049: USB_EXCLUDE_WWAN="0"
defaults.conf L0050: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"
defaults.conf L0051: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0052: RESTORE_THRESHOLDS_ON_BAT="0"
defaults.conf L0053: NATACPI_ENABLE="1"
defaults.conf L0054: TPACPI_ENABLE="1"
defaults.conf L0055: TPSMAPI_ENABLE="1"
/etc/tlp.d/00-debug.conf L0001: TLP_DEBUG="bat ps run"
/etc/tlp.conf L0500: STOP_CHARGE_THRESH_BAT0="80"

+++ System Info
System         = LG Electronics 0.1 16Z90P-K.AAC7U1
BIOS           = T2ZF0320 X64
[... 38 lines cut ...]
Capacity                                                    =   99.1 [%]

```

sudo tlp-stat -T  
```
Aug 01 18:50:52 vajra tlp[30339]: +++ setcharge (1.4.0-beta.1) ++++++++++++++++++++++++++++++++++++++++
Aug 01 18:50:52 vajra tlp[30339]: get_sys_power_supply(ADP1).remember_ac_offline_
Aug 01 18:50:52 vajra tlp[30339]: get_sys_power_supply(CMB0).bat_discharging: syspwr=1; wait=
Aug 01 18:50:52 vajra tlp[30339]: power_source=bat
Aug 01 18:50:52 vajra tlp[30339]: manual_mode=none
Aug 01 18:50:52 vajra tlp[30339]: power_mode=bat
Aug 01 18:50:52 vajra tlp[30339]: check_thinkpad.not_a_thinkpad: model=0.1
Aug 01 18:50:52 vajra tlp[30339]: batdrv_init.thinkpad.not_a_thinkpad
Aug 01 18:50:52 vajra tlp[30339]: check_thinkpad.not_a_thinkpad: model=0.1
Aug 01 18:50:52 vajra tlp[30339]: batdrv_init.thinkpad-legacy.not_a_thinkpad
Aug 01 18:50:52 vajra tlp[30339]: batdrv_init.asus.no_match
Aug 01 18:50:52 vajra tlp[30339]: batdrv_init.huawei.no_match
Aug 01 18:50:52 vajra tlp[30339]: batdrv_init.lenovo.no_match
Aug 01 18:50:52 vajra tlp[30339]: batdrv_init.samsung.no_match
Aug 01 18:50:52 vajra tlp[30339]: batdrv_init.lg: batteries=CMB0; natacpi=0; thresh=natacpi; bf_stop=/sys/devices/platform/lg-laptop/battery_care_limit
Aug 01 18:50:52 vajra tlp[30339]: batdrv.lg.select_battery(DEF): bat_str=CMB0; bd_read=/sys/class/power_supply/CMB0;
Aug 01 18:50:52 vajra tlp[30339]: batdrv.lg.read_threshold(stop): bf_stop=/sys/devices/platform/lg-laptop/battery_care_limit; out=0; rc=0
Aug 01 18:50:52 vajra tlp[30339]: batdrv.lg.write_thresholds(80, 2, ).write: old=0; new=80; rc=6
```

linrunner, 2021-08-02:
Whatever you call it, it's the kernel which fails. Try it without TLP:

    sudo -s # root shell
    cat /sys/devices/platform/lg-laptop/battery_care_limit
    echo 80 > /sys/devices/platform/lg-laptop/battery_care_limit
    cat /sys/devices/platform/lg-laptop/battery_care_limit

ps. you need to rebuild tlp-rdw-git too if you want to keep it.

sutram, 2021-08-02:
Ok, got it. Thanks.

I assume this bug is in the lg_laptop module? I have never created a kernel bug report. If you have, it would be great if you could guide me.

linrunner, 2021-08-02:
Yes, _lg_laptop_. Sorry, i do not file kernel bugs that often. I'd start here: https://bugzilla.kernel.org/

From what i see in the [lg_laptop.c code](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/platform/x86/lg-laptop.c#n475) it is also possible that the LG firmware does not provide the function for your model. Values returned by the ACPI call that are not equal to 80 or 100 are replaced there by 0:

	if (status != 80 && status != 100)
		status = 0;

linrunner, 2021-08-02:
ps. i added another check for battery_care_limit=0 to the code. If you rebuild again,

    tlp-stat -b

> * vendor (lg_laptop) = inactive (laptop not supported)

Would be helpful if you could verify that.

linrunner, 2021-08-02:
ps2. you may also look for corresponding messages after writing with **tlp setcharge**:

    dmesg

sutram, 2021-08-02:
Oh, that is too bad. If the firmware is not proving that function, not much that the driver can do. Thanks for looking into the source code. With the latest from tlp git your changes are working as expected but not that way I would like :)

You can close this issue if you like. Thanks for your help.

```
+++ Battery Care
Plugin: lg
Supported features: none available
Driver usage:
* vendor (lg_laptop) = inactive (laptop not supported)

+++ Battery Status: CMB0
/sys/class/power_supply/CMB0/manufacturer                   =  LG 
/sys/class/power_supply/CMB0/model_name                     = LGC-LGC
/sys/class/power_supply/CMB0/cycle_count                    =      6
/sys/class/power_supply/CMB0/energy_full_design             =  80000 [mWh]
/sys/class/power_supply/CMB0/energy_full                    =  79570 [mWh]
/sys/class/power_supply/CMB0/energy_now                     =  34490 [mWh]
/sys/class/power_supply/CMB0/power_now                      =   7848 [mW]
/sys/class/power_supply/CMB0/status                         = Discharging

Charge                                                      =   43.3 [%]
Capacity                                                    =   99.5 [%]

```

```
tlp setcharge 0 80
Error: battery charge thresholds not available.
```

linrunner, 2021-08-02:
Then there is still hope. If you decide to open a kernel bug, please link it here.

I leave open for now, maybe other users can contribute something to the solution.

linrunner, 2021-09-10:
Beta 2 is [out](https://linrunner.de/tlp/news.html#news-top-1) which includes above fix. 

Any news from the kernel front?

sutram, 2021-09-11:
> Beta 2 is [out](https://linrunner.de/tlp/news.html#news-top-1) which includes above fix.
> 
> Any news from the kernel front?

@linrunner Yes, there is. Thanks for reminding me. 

I contacted the developer of the lg_laptop module and he was able to send me a fix for newer LG laptop models as part of the latest source code. Unfortunately, that didn't work for me but I figured out how to tweak the source code so that it works for my specific model. I sent him my fix though he is not sure which version of the kernel will have latest source code included. 

Until that time, I have to compile and deploy the module each time there is a new kernel version. :-(

linrunner, 2021-09-17:
@sutram : is [this](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/platform/x86/lg-laptop.c?id=4c4a3d7cffb42da21ea8891fc7e6808ae05dbcb5) the said patch?

Sparsa, 2021-09-17:
> @linrunner Yes, there is. Thanks for reminding me.
>
> I contacted the developer of the lg_laptop module and he was able to send me a fix for newer LG laptop models as part of the latest source code. [...]

Hi, have does your model have the fan mode? And does it work with the new patch?

sutram, 2021-09-23:
> @sutram : is [this](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/platform/x86/lg-laptop.c?id=4c4a3d7cffb42da21ea8891fc7e6808ae05dbcb5) the said patch?

No, there is a newer patch by the same author.

sutram, 2021-09-23:
@Sparsa I haven't looked into fan mode. How do I check for it? Which patch are you referring to?

linrunner, 2021-09-24:
[TLP 1.4 is released](https://linrunner.de/tlp/news.html#tlp-1-4-released). Have fun!

Further testing is welcome.

alcercu, 2021-09-27:
> > @sutram : is [this](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/platform/x86/lg-laptop.c?id=4c4a3d7cffb42da21ea8891fc7e6808ae05dbcb5) the said patch?
> 
> No, there is a newer patch by the same author.

[This](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/platform/x86/lg-laptop.c?id=8983bfd58d61ab29ab4114089d1f42e1ee6103a8) one right?
Also, thanks for all your work and testing! @linrunner @sutram

sutram, 2021-10-01:
@alcercu Yes! This is the one. Do you know which version of the kernel will have it?

alcercu, 2021-10-05:
> @alcercu Yes! This is the one. Do you know which version of the kernel will have it?

It seems like it will go live on 5.15 ([This is the commit](https://github.com/torvalds/linux/commit/8983bfd58d61ab29ab4114089d1f42e1ee6103a8) currently on v5.15-rc)

alcercu, 2021-11-04:
Updated to v5.15 and everything seems to be working correctly 🙌🏻
```
--- TLP 1.4.0 --------------------------------------------

+++ Battery Care
Plugin: lg
Supported features: charge threshold
Driver usage:
* vendor (lg_laptop) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0: 80(on), 100(off) -- battery care limit

/sys/devices/platform/lg-laptop/battery_care_limit          = 80 [%]

+++ Battery Status: CMB0
/sys/class/power_supply/CMB0/manufacturer                   =  LG
/sys/class/power_supply/CMB0/model_name                     = LGC-LGC
/sys/class/power_supply/CMB0/cycle_count                    =     42
/sys/class/power_supply/CMB0/energy_full_design             =  80000 [mWh]
/sys/class/power_supply/CMB0/energy_full                    =  76040 [mWh]
/sys/class/power_supply/CMB0/energy_now                     =  59920 [mWh]
/sys/class/power_supply/CMB0/power_now                      =      0 [mW]
/sys/class/power_supply/CMB0/status                         = Idle

Charge                                                      =   78.8 [%]
Capacity                                                    =   95.0 [%]
```

linrunner, 2021-11-05:
@alcercu : thanks for your feedback
@sutram , @nishantkr18 : does 5.15 work for you too?

nishantkr18, 2021-11-05:
Yeah, it does! Thanks for this :)
```
--- TLP 1.4.0 --------------------------------------------

+++ Battery Care
Plugin: lg
Supported features: charge threshold
Driver usage:
* vendor (lg_laptop) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0: 80(on), 100(off) -- battery care limit

/sys/devices/platform/lg-laptop/battery_care_limit          = 80 [%]

+++ Battery Status: CMB0
/sys/class/power_supply/CMB0/manufacturer                   =  LG 
/sys/class/power_supply/CMB0/model_name                     = LGC-LGC
/sys/class/power_supply/CMB0/cycle_count                    =     50
/sys/class/power_supply/CMB0/energy_full_design             =  72770 [mWh]
/sys/class/power_supply/CMB0/energy_full                    =  65390 [mWh]
/sys/class/power_supply/CMB0/energy_now                     =  51000 [mWh]
/sys/class/power_supply/CMB0/power_now                      = (not available) 
/sys/class/power_supply/CMB0/status                         = Charging

Charge                                                      =   78.0 [%]
Capacity                                                    =   89.9 [%]
```

sutram, 2021-11-05:
@linrunner I haven't installed 5.15 yet. Just waiting for Manjaro Linux to release it into their repo. I did take a look at the module code and I think it will work for my laptop.

Somewhat off-topic, @alcercu and @nishantkr18 , my laptop had a thermal issue described here:

https://bbs.archlinux.org/viewtopic.php?id=268721

It seems to be fixed in 5.15 and may also be backported to 5.14. Will be trying that out as well without the workaround in that thread.

alcercu, 2021-11-09:
> @linrunner I haven't installed 5.15 yet. Just waiting for Manjaro Linux to release it into their repo. I did take a look at the module code and I think it will work for my laptop.
>
> Somewhat [...]

Not facing this issue 🤔 But since I updated to 5.15 the file manager pop up is laggy. Are any of you having that problem?

linrunner, 2021-11-09:
Please stay on topic. Thanks.

sutram, 2021-11-21:
@alcercu @nishantkr18 @linrunner I just switched to 5.15 on ManjaroLinux and the included lg_laptop module code doesn't work :(

linrunner, 2022-01-19:
@sutram : any news from the kernel?

sutram, 2022-01-29:
@linrunner I tried with kernel 5.16 and battery care limit works out of the box!

```
+++ Battery Care
Plugin: lg
Supported features: charge threshold
Driver usage:
* vendor (lg_laptop) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0: 80(on), 100(off) -- battery care limit

/sys/devices/platform/lg-laptop/battery_care_limit          = 80 [%]

+++ Battery Status: CMB0
/sys/class/power_supply/CMB0/manufacturer                   =  LG 
/sys/class/power_supply/CMB0/model_name                     = LGC-LGC
/sys/class/power_supply/CMB0/cycle_count                    =     20
/sys/class/power_supply/CMB0/energy_full_design             =  80000 [mWh]
/sys/class/power_supply/CMB0/energy_full                    =  78700 [mWh]
/sys/class/power_supply/CMB0/energy_now                     =  62490 [mWh]
/sys/class/power_supply/CMB0/power_now                      =      0 [mW]
/sys/class/power_supply/CMB0/status                         = Idle

Charge                                                      =   79.4 [%]
Capacity                                                    =   98.4 [%]

```

I am ok with closing this issue.

Thanks.

linrunner, 2022-01-29:
Great.

closed 2022-01-29 by linrunner