linrunner/TLP #569: Charge threshold not (re)set for second battery when using tlp fullcharge

Dongerrs, 2021-08-03:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**
On Thinkpad T470 (or possibly laptop with dual batteries) only BAT0 has threshold (re)set to 100% when trying to fully charge both batteries, using 'tlp fullcharge'
Both batteries have 75% threshold set, but only BAT0 gets  set to 100%, not BAT1 as well 

**Expected behavior**
Both batteries have temp thresholds set to 100%, to allow for full charge of both batteries

A clear and concise description of what you expected to happen.
Expected both batteries to have temp threshold set to 100%

Steps to reproduce the unexpected behavior:
run 'tlp fullcharge' on T470 with dual batteries

A clear and concise description of what the bug is
Bug does not allow full charge of bothh batteries unless i manually reset threshold for BAT1 to 100%, via /etc/tlp.conf, and run tlp start

1. Does the problem occur on battery or AC or both? AC
2. Actions to reproduce the behaviour - run tlp fullcharge on Thinkpad T470 (or thinkpad) with dual batteries

3. Shell commands entered and their output - tlp fullcharge
Place your finger on the fingerprint reader
Setting temporary charge thresholds for BAT0:
  stop  = 100
  start =  96
Charging starts now, keep AC connected.

4. https://gist.github.com/Dongerrs/db10c8768f24c31f01dbcbe9d43090b5

**Additional context**

Add any other context about the problem here.

linrunner, 2021-08-03:
Hi, your expectation results from a misunderstanding: the commands **tlp setcharge/fullcharge/discharge/recalibrate** work only on _one_ battery at a time. This is why manpage and online docs always use "battery" in singular. 

To reset the thresholds for all batteries of your T470 you must use:

    tlp fullcharge # or tlp fullcharge BAT0
    tlp fullcharge BAT1

Dongerrs, 2021-08-03:
Specifying the specific battery works correctly, thanks!

closed 2021-08-03 by Dongerrs