linrunner/TLP #570: [TLP 1.4 Beta] ACPI platform profile not applied

truhanen, 2021-08-03:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

I have observed that the PLATFORM_PROFILE_ON_AC/BAT setting must be applied *before* at least one intel_pstate setting, CPU_BOOST_ON_AC/BAT. TLP seems to probably do this the other way around, which causes the platform profile not to be fully applied. The platform profile setting is correctly set in */sys/firmware/acpi/platform_profile*, yes, but e.g. *scaling_max_freq* is not affected according to the platform profile.

The required order might be a bug/feature in the intel_pstate driver or somewhere else, but can possibly be worked around in TLP by reversing the order in which those two settings are applied.

Feel free to close this issue right away if it's not the responsibility of TLP. But if there already exist other similar workarounds, this could be one of those.

**Expected behavior**

When the *PLATFORM_PROFILE_ON_AC/BAT* is changed from *balanced* to *performance* while the CPU "turbo boost" feature is off, the value in */sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq* should change from 1200000 to 2800000 kHz.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. The problem occurs on both battery & AC.
2. Assert that CPU turbo boost is *off*, ACPI platform profile is *balanced*, & */sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq* shows 1200000 kHz, [AC](https://gist.github.com/truhanen/3081d2afd1f604fe461a54fa353d7d25#file-ac_1-txt)/[battery](https://gist.github.com/truhanen/3081d2afd1f604fe461a54fa353d7d25#file-bat_1-txt).
3. Change the TLP setting *PLATFORM_PROFILE_ON_AC/BAT* to *performance* & run `sudo tlp start`.
4. Assert that *scaling_max_freq* still shows 1200000 kHz even if it should be 2800000 kHz, [AC](https://gist.github.com/truhanen/3081d2afd1f604fe461a54fa353d7d25#file-ac_2-txt)/[battery](https://gist.github.com/truhanen/3081d2afd1f604fe461a54fa353d7d25#file-bat_2-txt).
5. Call `echo 1 > /sys/devices/system/cpu/intel_pstate/no_turbo` *or* `sudo tlp start` *again* to see that *scaling_max_freq* then shows 2800000 kHz, [AC](https://gist.github.com/truhanen/3081d2afd1f604fe461a54fa353d7d25#file-ac_3-txt)/[battery](https://gist.github.com/truhanen/3081d2afd1f604fe461a54fa353d7d25#file-bat_3-txt).

**Additional context**

Sometimes *immediately* after switching between battery & AC, I have observed that setting the *no_turbo* value doesn't fix the issue as described in the steps above. After a couple of minutes, however, the steps have been reproducible.

linrunner, 2021-08-04:
Hi, thanks for your insights.

Changing the write sequence is simple. If you want to experiment, you may change it yourself --> [**/usr/sbin/tlp** L. 32](https://github.com/linrunner/TLP/blob/main/tlp.in#L32).

But i wonder what might be the "native" or vendor-intended behavior. _intel_pstate_ processes input from the system firmware and _platform_profile_ is an interface to that firmware.

Please do the following:

Disable the active part of TLP via configuration

    TLP_ENABLE=0

Reboot and invoke a root shell

    sudo -s

Then show the outputs:

    tlp-stat -p
    echo performance > /sys/firmware/acpi/platform_profile
    tlp-stat -p
    echo balanced > /sys/firmware/acpi/platform_profile
    tlp-stat -p
    echo low-power > /sys/firmware/acpi/platform_profile
    tlp-stat -p

Reboot again and invoke root shell. Show the outputs:

    echo 1 > /sys/devices/system/cpu/intel_pstate/no_turbo
    tlp-stat -p
    echo performance > /sys/firmware/acpi/platform_profile
    tlp-stat -p
    echo balanced > /sys/firmware/acpi/platform_profile
    tlp-stat -p
    echo low-power > /sys/firmware/acpi/platform_profile
    tlp-stat -p

linrunner, 2021-08-04:
What i gather from your testing is: 

- After boot: writing 1 to _sys/devices/system/cpu/intel_pstate/no_turbo_ without prior touch of _/sys/firmware/acpi/platform_profile_ already creates the symptom
- Writing _platform_profile_ (once?) before _no_turbo_ prevents it

Meaning that the real issue - that should be reported to Lenovo or as a kernel bug -  is with _no_turbo_. Right?

linrunner, 2021-08-04:
ps. does

    echo 0 > /sys/devices/system/cpu/intel_pstate/no_turbo

also make the symptom disappear?

truhanen, 2021-08-05:
Yes it does. The max frequency raises to 47000000 when the turbo boost is turned on independent of the platform profile setting.

linrunner, 2021-08-05:
Here you are. Please test and report.

Please also report the underlying problem, which has nothing to do with TLP, in the Lenovo forum and try to get Mark Pearson's (Mark_Lenovo) attention for it.

truhanen, 2021-08-05:
> After boot: writing 1 to sys/devices/system/cpu/intel_pstate/no_turbo without prior touch of /sys/firmware/acpi/platform_profile already creates the symptom
Yes, if the "symptom" here is that the max frequency isn't affected by the platform_profile state.
> Writing platform_profile (once?) before no_turbo prevents it
I'd rather say that writing no_turbo *after* platform_profile prevents the problem, even if the no_turbo value doesn't change.
> Writing platform_profile doesn't immediately change any of the cpufreq or intel_pstate attributes
This is true. Changing the order of invocation in TLP helps *only* if the settings for *both* platform_profile & no_turbo are enabled.

truhanen, 2021-08-05:
Many thanks for your advice on this issue!

truhanen, 2021-08-05:
And of course for fixing it as far as possible in TLP, as you apparently already did :)

Do you mean that it would be good to report both as a kernel issue and in the Lenovo forums?

linrunner, 2021-08-05:
I'd try Lenovo forum first. Possibly the firmware sends wrong ACPI data to the kernel.

truhanen, 2021-08-05:
Ok, thanks again.

truhanen, 2021-08-06:
I installed the updated version & tried to reproduce the issue as I first reported. Here are the logs for the steps 2 & 4. Everything works as expected, and the issue seems to be fixed.
2. [AC](https://gist.github.com/truhanen/21524683c7093efdb2d702fc8cfee8dc#file-ac1-txt)/[battery](https://gist.github.com/truhanen/21524683c7093efdb2d702fc8cfee8dc#file-bat1-txt)
4. [AC](https://gist.github.com/truhanen/21524683c7093efdb2d702fc8cfee8dc#file-ac2-txt)/[battery](https://gist.github.com/truhanen/21524683c7093efdb2d702fc8cfee8dc#file-bat2-txt)

linrunner, 2021-08-06:
Very well.

linrunner, 2021-09-24:
[TLP 1.4 is released](https://linrunner.de/tlp/news.html#tlp-1-4-released). Have fun!

linrunner, 2021-09-26:
Problem added to the [FAQ](https://linrunner.de/tlp/faq/processor.html#cpu-boost-on-bat-ac-0-decreases-scaling-max-freq). 

What did Lenovo answer?

closed 2021-09-26 by linrunner

Vargblood, 2022-10-28:
Hi all,

I can confirm that I still have this issue on my T490 (Arch Linux, linux hardened kernel 6.0.3, TLP 1.5.0). When I switch to the lts kernel 5.15.75 everything works as expected. The output of TLP as follows:

`
+++ Processor
CPU model      = Intel(R) Core(TM) i7-8565U CPU @ 1.80GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference = balance_performance [HWP.EPP]
/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power 
`

I hope to get some help here. Thank you.

Ph0rk0z, 2024-12-27:
Greetings from the future. X1 carbon gen9. Latest bios. 6.12 kernel. Totally not fixed. Goes to 1200mhz max when turbo is disabled. Doesn't matter what order you set them.

**I however did find a fix. If you set the turbo ratio to "auto" in MSR register 0x64c then it works as normal. The only downside is your turbo now only goes to 4.2ghz. Much horror, right? I am going to put a systemd service to set this to 0x0 at startup and call it a day.**

Worse thing is that /sys/devices/platform/thinkpad_acpi/dytc_lapmode is always 1 with my laptop on a wooden desk. Maybe it's because it's tilted. Have never seen it 0. I can still use turbo/performance and all that fun stuff, but I think it makes the fans blow. I also notice that PL1 on core is often set as an event when the CPU is stressed despite no thermal throttle.