linrunner/TLP #571: /usr/share/tlp/func.d/35-tlp-func-batt: line 25: xxxx Killed $TPACPIBAT -g FD 1 > /dev/null 2>&1

erazemk, 2021-08-08:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

TLP reports the following error on my Thinkpad T540p: `Error: battery discharge/recalibrate not available.`

I've used TLP many times before, with the calibration working normally, but since reinstalling Fedora 34 it doesn't seem to work anymore.

Bash also reports the following errors (process numbers change with every run):
```bash
/usr/share/tlp/func.d/35-tlp-func-batt: line 25:  8001 Killed                  $TPACPIBAT -g FD 1 > /dev/null 2>&1
/usr/share/tlp/func.d/35-tlp-func-batt: line 25:  8049 Killed                  $TPACPIBAT -g FD 1 > /dev/null 2>&1
```

**Expected behavior**

TLP allows me to recalibrate the battery.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. The recalibration problem only occurs on AC (recalibration isn't supported on battery), but the bash errors are reported any time a tlp command is run.
2. Install TLP 1.3.1 on Fedora 34 according to TLP instructions (install both packages, both repos, akmod-acpi_call and enable the testing repo)
3. Run any tlp command
4. `tlp-stat` output: https://gist.github.com/erazemk/ce7e6e985aa08a0a6c6daac514c0655b

**Additional context**

I tried following the Fedora instructions and separately tried masking the service and socket listed in the Arch instructions, both cases failed.

Based on the bash errors, I'm assuming the problem is with the bash script `/usr/share/tlp/func.d/35-tlp-func-batt` on line 153, where it doesn't find `tpacpi-bat`.

linrunner, 2021-08-09:
I guess it's SELinux (once again) that's causing trouble here. Try to invoke tpacpi-bat manually:

    sudo /usr/share/tlp/tpacpi-bat -g FD 1

If this works, please check for SELinux AVCs that occur when you run whatever command you use to reproduce the issue but forgot to tell.

erazemk, 2021-08-10:
> If this works, please check for SELinux AVCs that occur when you run whatever command you use to reproduce the issue but forgot to tell.

I'm guessing it working would mean it would exit with status 0, which it didn't in this case: `fish: Job 1, 'sudo /usr/share/tlp/tpacpi-bat…' terminated by signal SIGKILL (Forced quit)`

linrunner, 2021-08-12:
This looks even more like selinux killing the script. Enter the command above again, then show 
    
    sudo ausearch -ts recent

erazemk, 2021-08-16:
> This looks even more like selinux killing the script. Enter the above command again, then show
> 
> ```
> sudo ausearch -ts recent
> ```

I'm not sure which part is relevant for you, so here's the whole output: [ausearch.txt](https://github.com/linrunner/TLP/files/6990687/ausearch.txt).

linrunner, 2021-08-19:
Indeed the audit log reveals nothing specific. Another possibility is that your symptoms stem from kernel 5.13 and this is a duplicate of #565 . 

Please make shure you have installed acpi_call-1.2.1-1.fc34.noarch.rpm from the [tlp-updates-testing](https://linrunner.de/tlp/installation/fedora.html#thinkpads-only) repo and retest afterwards.

linrunner, 2021-08-19:
ps. at least the symptom would fit: https://github.com/nix-community/acpi_call/pull/16

erazemk, 2021-08-19:
> Please make shure you have installed acpi_call-**1.2.1-1.fc34**.noarch.rpm from the [tlp-updates-testing](https://linrunner.de/tlp/installation/fedora.html#thinkpads-only) repo and retest afterwards.

I thought I did, but apparently the `tlp-updates-testing` repo was not enabled, so I was using `akmod-acpi_call-1.1.2` instead of `1.2.1`. After enabling the repo and updating all needed packages, the problem isn't there anymore.

Thank you for your help.

closed 2021-08-19 by erazemk

popov895, 2021-09-15:
I have the same issue on openSUSE Tumbleweed. 
```
$ sudo /usr/share/tlp/tpacpi-bat -g FD 1
Killed
```
Is is SELinux? What can I do to fix this issue?

linrunner, 2021-09-15:
@popov895 : you need a recent version of acpi_call. Source code is here: https://github.com/nix-community/acpi_call

popov895, 2021-09-15:
@linrunner : Do you mean that this package is outdated and I need to build it from source?

linrunner, 2021-09-16:
@popov895 : right.

popov895, 2021-09-16:
@linrunner : BTW on openSUSE Tumbleweed I should install **acpi_call-kmp-default** package instead of (or in addition to?) **acpi_call**.

**acpi_call** installs the following files (`rpm -ql acpi_call`):
```
/usr/libexec/modules-load.d
/usr/libexec/modules-load.d/acpi_call.conf
/usr/share/doc/packages/acpi_call
/usr/share/doc/packages/acpi_call/README.md
```

**acpi_call-kmp-default** installs the following files (`rpm -ql acpi_call`):
```
/lib/modules/5.14.1-1-default
/lib/modules/5.14.1-1-default/updates
/lib/modules/5.14.1-1-default/updates/acpi_call.ko
```

linrunner, 2021-09-16:
@popov895 : that's out of my scope. If you should find out something, we can include it in the [documentation](https://linrunner.de/tlp/installation/opensuse.html).

popov895, 2021-09-16:
@linrunner : 
> we can include it in the documentation

Yes, that's why I gave you this info. But let's wait while I talk with the openSUSE maintainers, maybe they need to add **acpi_call-kmp-default** to the **acpi_call** dependencies. I'll let you know later.

linrunner, 2021-09-17:
@popov895 : send me a pull request for the docs --> https://github.com/linrunner/tlp-doc/blob/main/installation/opensuse.rst