linrunner/TLP #581: suport for lenovo 82LM IdeaPad 5 14ALC05

rezad1393, 2021-09-18:
**Is your feature request related to a problem? Please describe.**

I have a lenovo 82LM IdeaPad 5 14ALC05 laptop and with this apps 
https://github.com/0xless/battmngr
https://github.com/tildehacker/ideapad-conservation-mode
I can enable battery conservation mode.
it seem to only have one limit line which is like having the start and stop threshold of other ideapads combined and it seems to not be changeable and is constant at 60%.

**Describe the solution you'd like**
add support for this in tlp.

linrunner, 2021-09-18:
That is already implemented my dear --> https://gist.github.com/linrunner/2ead4b591eed33055cf86a38ccc73949

linrunner, 2021-09-18:
ps. more than the fixed 60% is not possible under Linux, unless you program the change to the `ideapad_laptop` driver yourself.

closed 2021-09-18 by linrunner

rezad1393, 2021-09-18:
should I install tlp1.4?
because in 1.3 I get this:
` sudo tlp-stat -b  
--- TLP 1.3.1 --------------------------------------------

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = inactive (laptop not supported)
tpacpi-bat = inactive (laptop not supported)
tp-smapi   = inactive (laptop not supported)

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = 0x43 0x65 0x6C 0x78 0x70 0x65 0x
/sys/class/power_supply/BAT1/model_name                     = 0x4C 0x31 0x39 0x43 0x33 0x50 0x5650
/sys/class/power_supply/BAT1/cycle_count                    =      7
/sys/class/power_supply/BAT1/energy_full_design             =  56500 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  56500 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  33630 [mWh]
/sys/class/power_supply/BAT1/power_now                      =      0 [mW]
/sys/class/power_supply/BAT1/status                         = Unknown

Charge                                                      =   59.5 [%]
Capacity                                                    =  100.0 [%]
`

and these two are absent
Driver usage:
* vendor (ideapad_wmi) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0: 0(off)..1(on) -- battery conservation mode

linrunner, 2021-09-19:
@rezad1393 : of course, you *must* install the 1.4 beta for that. Read the page behind my link.