linrunner/TLP #582: /tmp folder filled with tlp temporay files

AlecsFerra, 2021-09-20:
I noticed that leaving my laptop on for some days this software fills the ```/tmp``` directory with temporary files, how can I find the root of the issue?

linrunner, 2021-09-21:
Generally I expect that people provide actionable information in their issue in order to enable analysis; the issue template should help with this. Which amounts at least to the full output of

    sudo tlp-stat 

As the output is long, via [Gist](https://gist.github.com/) please.

In your case, of course, a listing of the directory in question is also required, e.g.

    ls -l /tmp

linrunner, 2021-09-21:
Also please show the output of

    sudo tlp start

linrunner, 2021-09-30:
I am missing your details.

AlecsFerra, 2021-10-01:
Sorry I was occupied with uni

https://gist.github.com/AlecsFerra/817683ab48a0f364aa9df61450dbd71b

linrunner, 2021-10-02:
Something or someone calls a part of TLP in user context (alecs). TLP stores the current configuration in /tmp and deletes it after the program ends. 

But the installation package only creates tasks that run with root rights, so these calls here must be set up by yourself.

So what exactly did you do?

AlecsFerra, 2021-10-02:
Can running tlp-stat without root priviliges run a new instance?

AlecsFerra, 2021-10-02:
Or changing the mode

linrunner, 2021-10-02:
The call of tlp-stat creates such a tempfile. Instead of counter-questions, I would now like to have precise information on how and with which parameters you call tlp-stat so that we can find out why the tempfile is not deleted.

AlecsFerra, 2021-10-02:
I run this command
```exec = sudo tlp-stat -s | grep -q 'Mode           = AC' && echo " " || echo "鈴  "```

linrunner, 2021-10-05:
Thats strange. tlp-stat `-s` does not require sudo, but if you call it with sudo, it writes the configuration but to **/tmp**, but to **/run/tlp/run.conf**. Is there perhaps another call you have set up?

Instead of scraping tlp-stat's output take a look at `run-on-ac` and `run-on-bat`. They are much leaner and do not read any configuration --> https://linrunner.de/tlp/usage/run-on.html

linrunner, 2021-12-07:
Any news on this?

linrunner, 2021-12-21:
I'll close here. Nevertheless, you are welcome to post new findings about your problem here.

closed 2021-12-21 by linrunner