linrunner/TLP #583: HP battery charge limit support

JulianGro, 2021-09-21:
**Is your feature request related to a problem? Please describe.**

Battery charge limit doesn't work on my HP 625.

**Describe the solution you'd like**

I would like to set a charge limit to not have the battery die when being connected to power for extended periods of time.

**Additional context**

I tried the TLP beta 2.

linrunner, 2021-09-25:
Hi, the magic word here is "suitable kernel driver" i.e. one that provides the necessary sysfiles

    /sys/class/power_supply/BAT0/charge_control_start_threshold
    /sys/class/power_supply/BAT0/charge_control_end_threshold

Which is not the case in the current driver _hp_wmi_ for HP laptops. So no way for TLP to implement a charge limit.

closed 2021-12-07 by linrunner

AnterCreeper, 2022-02-28:
I have find that there are some options hidden in the BIOS, which called:
One Of: Battery Care Function
 				One Of Option: 100%, Value (8 bit): 0x0 (default)
 				One Of Option: 80%, Value (8 bit): 0x1
 				One Of Option: 50%, Value (8 bit): 0x2
End One Of {29 02}
Here is the only way.