linrunner/TLP #584: metadata about all possible config values

attila-lendvai, 2021-09-22:
i'm looking at the Guix packaging of TLP, and it would benefit greatly if there was a formal specification that listed all the possible config values with their metadata.

this applies the same to NixOS.

**Describe the solution you'd like**

to have a file in the official repo with a fixed format that contains at least these:
 - name
 - type
 - default value
 - whether mandatory or not (this may not apply if everything is optional).
 - optionally: documentation.

a piece from the Guix package definition for reference (https://github.com/guix-mirror/guix/blob/master/gnu/services/pm.scm#L85):

```
(cpu-max-perf-on-bat
   (maybe-non-negative-integer 'disabled)
   "Same as @code{cpu-max-perf-on-ac} on BAT mode.")
```

notice how the type contains that it's optional.

this file should be used internally to generate various things, like parts of the documentation, otherwise it will get out of sync.

feel free to reject this, i can understand if it's not worth the trouble on your end.

**Context**

Guix has a declarative syntax to specify the configuration of your Linux install. this syntax is type checked. currently the TLP service description has to list all the possible TLP config values (names are automatically converted from `foo-bar` to `FOO_BAR`). if the requested metadata were available, then i could generate the config description with a piece of code, instead of maintaining it by hand.

linrunner, 2021-09-25:
Hi,

that would indeed be a considerable effort. In addition, for many parameters the effective value range still depends on hardware and kernel. A formal specification would be interesting if the TLP code would use it itself. But this is not the case (and not possible in shell code).

Instead, further redundancy would be created without any significant benefit to TLP itself. Today I already have to maintain the code itself, tlp.conf and the docsite. So I don't want to fulfill your wish.

But take a look at TLPUI, a specification is maintained there: https://github.com/d4nj1/TLPUI/tree/master/tlpui/configschema

attila-lendvai, 2021-09-25:
ok, that makes sense. thank you for the link! i may end up using that.

closed 2021-09-25 by attila-lendvai