linrunner/TLP #585: Pc never use AC power since TLP has been installed

o101010, 2021-09-25:
**Describe the bug**

Hi,
I recently install Linux on my laptop. There was no problems.
I have install tlp and acpi-call-dkms and configured it via tlpui. I just remove RADEON* flags and set CHARGE_THRESH_BAT0. After that, my computer never used AC power when I plug it. It doesn't charge and doesn't used it as power source. Only the battery.

Unlike others issue, AC is detect but not the plugin status.
I uninstall the components via `apt purge` but it change nothing. AC power is not used.

**Expected behavior**

The AC power is used and the battery charge depending on charge thresh limits.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
Both
2. Actions to reproduce the behaviour
Install tlp and acpi_call_dkms
Starting with or without charger change nothing.
3. Shell commands entered and their output

4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**
https://gist.github.com/o101010/c4b32fba9ef16117bf7b41deb2ff1ad0

**Additional context**

I'm on a Thinkpad X280 with Zorin Os 16 (based on Ubuntu 20.04.3 LTS)
power-profiles-daemon.service seem's not install

Thank you in advance for your help.

linrunner, 2021-09-25:
Symptoms like this are never caused by software. Charging must work even without the operating system. You have a hardware problem (charger or system board).

> I uninstall the components via `apt purge` but it change nothing. AC power is not used.
This simply means that TLP has nothing to do with your problem.

closed 2021-09-25 by linrunner

o101010, 2021-10-02:
Mea-culpa,

The charger dead at the same time that I install TLP. Sorry.