linrunner/TLP #586: Host power management not set when SATA disks not present

lbschenkel, 2021-10-05:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

I have the following on `tlp.conf`:
```
TLP_DEBUG="disk"
SATA_LINKPWR_ON_AC="max_performance med_power_with_dipm"
SATA_LINKPWR_ON_BAT="min_power med_power_with_dipm"
#SATA_LINKPWR_BLACKLIST="host1"
AHCI_RUNTIME_PM_ON_AC="on"
AHCI_RUNTIME_PM_ON_BAT="auto"
#AHCI_RUNTIME_PM_TIMEOUT=15
#PCIE_ASPM_ON_AC=default
PCIE_ASPM_ON_BAT="powersupersave"
```
But I only have a NVMe disk in my laptop, the SATA ports are unused.

**Expected behavior**

That SATA link and runtime PM settings are applied to the SATA ports, but they aren't:
```
# tlp-stat -d
+++ Storage Devices
Devices = nvme0n1 sda

/dev/nvme0n1:
  Type      = NVMe
  Model     = Micron_2210_MTFDHBA1T0QFD               
  Firmware  = P6MA000 
  Scheduler = [none] mq-deadline kyber bfq (multi queue)

  Runtime PM: control = auto, autosuspend_delay_ms = (not available)

  SMART info:
    [omitted, not relevant]

/dev/sda: not present.

+++ AHCI Link Power Management (ALPM)
/sys/class/scsi_host/host0/link_power_management_policy  = min_power
/sys/class/scsi_host/host1/link_power_management_policy  = min_power

+++ AHCI Host Controller Runtime Power Management
/sys/bus/pci/devices/0000:04:00.0/ata1/power/control = on
/sys/bus/pci/devices/0000:04:00.1/ata2/power/control = on

# tlp-stat -T
--- TLP 1.3.1 --------------------------------------------

-- Journal begins at Sun 2021-09-19 09:51:44 CEST, ends at Tue 2021-10-05 11:22:04 CEST. --
okt 05 11:19:54 laptop tlp[554]: set_disk_apm_level(1)
okt 05 11:19:54 laptop tlp[554]: get_disk_dev(nvme0n1): dev=nvme0n1; type=nvme; path=pci-0000:02:00.0-nvme-1; bus=; dpath=
okt 05 11:19:54 laptop tlp[554]: set_disk_apm_level(1): nvme0n1 [] 128; not supported
okt 05 11:19:54 laptop tlp[554]: get_disk_dev(sda).missing
okt 05 11:19:54 laptop tlp[554]: set_disk_spindown_timeout(1).not_configured
okt 05 11:19:54 laptop tlp[554]: set_disk_iosched
okt 05 11:19:54 laptop tlp[554]: get_disk_dev(nvme0n1): dev=nvme0n1; type=nvme; path=pci-0000:02:00.0-nvme-1; bus=; dpath=
okt 05 11:19:54 laptop tlp[554]: set_disk_iosched: nvme0n1 [] keep; keep as is
okt 05 11:19:54 laptop tlp[554]: get_disk_dev(sda).missing
okt 05 11:19:56 laptop tlp[554]: Applying power save settings...done.
okt 05 11:19:56 laptop tlp[554]: Setting battery charge thresholds...done.
```

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? **battery**
2. Actions to reproduce the behaviour: `tlp bat`
3. Shell commands entered and their output: **see above**
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**
See above

**Additional context**

Given the trace output, TLP does not detect any SATA (sd*) disks (correctly, since I don't have any) and therefore does not try to set the AHCI_RUNTIME_PM settings. However, it would be nice if it did that regardless to save power.

Powertop notices that the settings are not optimal, and I can change them there. Once set, `tlp` does not touch them on transitions AC<->BAT<->AC, most likely due to bypassing the whole logic since it does not detect SATA disks.

linrunner, 2021-10-05:
Hi, 

the AHCI runtime pm support in 1.3.1 is - let's say - unfinished. 

Upgrade to 1.4.0 will meet your expectation ([see code](https://github.com/linrunner/TLP/blob/1.4.0/func.d/15-tlp-func-disk#L554)).

lbschenkel, 2021-10-05:
All right! Upgraded TLP to 1.4.0 indeed fixed the issue. Thank you. Closing.

closed 2021-10-05 by lbschenkel