linrunner/TLP #587: [tlp 1.4.0-1] USB ports don't work after update

tomizius, 2021-10-05:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

After the update this morning from version tlp 1.3.1 to tlp 1.4.0 the usb ports don't work as expected.
With version 1.3.1 I never had any prolembs with my usb ports. With version 1.4.0 the usb ports works, if I have the power supplier plugged to my device. If I remove the power supplier, the usb ports stop working after few seconds.

I just replaced tlp version 1.4.0 with version 1.3.1 out of my pacman cache and everything works as expected again.

This problem is also at an other device by a fellow student.

**Expected behavior**

Fix, that the usb ports are also working in battery mode.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
only on battery mode
2. Actions to reproduce the behaviour
 Use tlp version 1.4.0 with a Lenovo notebook and try to use usb.
3. Shell commands entered and their output
none - but if you need some, I can do it
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**
--- TLP 1.4.0 --------------------------------------------

+++ Configured Settings:
defaults.conf L0004: TLP_ENABLE="1"
defaults.conf L0005: TLP_WARN_LEVEL="3"
defaults.conf L0006: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0008: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0010: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0012: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0013: SCHED_POWERSAVE_ON_AC="0"
defaults.conf L0014: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0015: NMI_WATCHDOG="0"
defaults.conf L0016: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0017: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0018: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0019: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0020: DISK_IOSCHED="keep keep"
defaults.conf L0021: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0022: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
defaults.conf L0023: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0025: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0026: PCIE_ASPM_ON_AC="default"
defaults.conf L0027: PCIE_ASPM_ON_BAT="default"
defaults.conf L0028: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0030: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0032: WIFI_PWR_ON_AC="off"
defaults.conf L0033: WIFI_PWR_ON_BAT="on"
defaults.conf L0034: WOL_DISABLE="Y"
defaults.conf L0035: SOUND_POWER_SAVE_ON_AC="1"
defaults.conf L0036: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0037: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0038: BAY_POWEROFF_ON_AC="0"
defaults.conf L0039: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0040: BAY_DEVICE="sr0"
defaults.conf L0041: RUNTIME_PM_ON_AC="on"
defaults.conf L0042: RUNTIME_PM_ON_BAT="auto"
[... 201 lines cut ...]
* natacpi (thinkpad_acpi) = active (charge thresholds)
* tpacpi-bat (acpi_call)  = active (recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = Celxpert
/sys/class/power_supply/BAT0/model_name                     = 5B10W138
/sys/class/power_supply/BAT0/cycle_count                    =    155
/sys/class/power_supply/BAT0/energy_full_design             =  45730 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  40640 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  26750 [mWh]
/sys/class/power_supply/BAT0/power_now                      =   9944 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     95 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   65.8 [%]
Capacity                                                    =   88.9 [%]

+++ Recommendations
* Install ethtool to disable Wake On LAN (WOL)

linrunner, 2021-10-05:
Hi,

are you absolutely sure that this did not already occur with 1.3.1? I ask because this is a known issue (-> [FAQ](https://linrunner.de/tlp/faq/usb.html#newly-inserted-usb-devices-not-recognized-in-battery-mode)) on AMD ThinkPads since 1.2.

Before you try the solution from the FAQ, be so kind, install 1.3.1, reboot, show the full output of `tlp-stat` - and check for the symptom of course.

tomizius, 2021-10-05:
Hello,

I am 100 % sure, that it not occur with 1.3.1, because right now I am just running 1.3.1 and it's working. 1.3.1 is also working at an other Lenovo notebook!

Here we go:
--- TLP 1.3.1 --------------------------------------------

+++ Configured Settings:
defaults.conf L0004: TLP_ENABLE="1"
defaults.conf L0005: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0006: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0008: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0010: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0012: SCHED_POWERSAVE_ON_AC="0"
defaults.conf L0013: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0014: NMI_WATCHDOG="0"
defaults.conf L0015: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0016: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0017: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0018: DISK_IOSCHED="keep keep"
defaults.conf L0019: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0020: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
defaults.conf L0021: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0022: PCIE_ASPM_ON_AC="default"
defaults.conf L0023: PCIE_ASPM_ON_BAT="default"
defaults.conf L0024: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0025: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0026: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0027: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0028: WIFI_PWR_ON_AC="off"
defaults.conf L0029: WIFI_PWR_ON_BAT="on"
defaults.conf L0030: WOL_DISABLE="Y"
defaults.conf L0031: SOUND_POWER_SAVE_ON_AC="0"
defaults.conf L0032: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0033: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0034: BAY_POWEROFF_ON_AC="0"
defaults.conf L0035: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0036: BAY_DEVICE="sr0"
defaults.conf L0037: RUNTIME_PM_ON_AC="on"
defaults.conf L0038: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0039: RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me nouveau nvidia pcieport radeon"
defaults.conf L0040: USB_AUTOSUSPEND="1"
defaults.conf L0041: USB_BLACKLIST_BTUSB="0"
defaults.conf L0042: USB_BLACKLIST_PHONE="0"
defaults.conf L0043: USB_BLACKLIST_PRINTER="1"
defaults.conf L0044: USB_BLACKLIST_WWAN="0"
defaults.conf L0045: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"
defaults.conf L0046: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0047: RESTORE_THRESHOLDS_ON_BAT="0"
defaults.conf L0048: NATACPI_ENABLE="1"
defaults.conf L0049: TPACPI_ENABLE="1"
defaults.conf L0050: TPSMAPI_ENABLE="1"

+++ System Info
System         = LENOVO ThinkPad E595 20NF0006GE
BIOS           = R11ET39W (1.19 )
Release        = "Arch Linux"
Kernel         = 5.14.9-arch2-1 #1 SMP PREEMPT Fri, 01 Oct 2021 19:03:20 +0000 x86_64
/proc/cmdline  = initrd=\initramfs-linux.img cryptdevice=/dev/nvme0n1p2:main root=/dev/mapper/main-home rw lang=de init=/usr/lib/systemd/systemd locale=de_DE.UTF-8
Init system    = systemd 
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = not installed
Last run       = 17:13:28,     29 sec(s) ago
Mode           = AC
[... 201 lines cut ...]

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = active (data, thresholds)
tpacpi-bat = active (recalibrate)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = Celxpert
/sys/class/power_supply/BAT0/model_name                     = 5B10W138
/sys/class/power_supply/BAT0/cycle_count                    =    155
/sys/class/power_supply/BAT0/energy_full_design             =  45730 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  40640 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  20540 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  33825 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

/sys/class/power_supply/BAT0/charge_start_threshold         =     95 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   50.5 [%]
Capacity                                                    =   88.9 [%]

+++ Recommendations
* Install ethtool to disable Wake On LAN (WOL)

linrunner, 2021-10-05:
Your output is on AC power, battery is needed for comparison with 1.4.0.

tomizius, 2021-10-05:
Sorry for that.

Here we go again:
--- TLP 1.3.1 --------------------------------------------

+++ Configured Settings:
defaults.conf L0004: TLP_ENABLE="1"
defaults.conf L0005: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0006: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0008: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0010: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0012: SCHED_POWERSAVE_ON_AC="0"
defaults.conf L0013: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0014: NMI_WATCHDOG="0"
defaults.conf L0015: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0016: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0017: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0018: DISK_IOSCHED="keep keep"
defaults.conf L0019: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0020: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
defaults.conf L0021: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0022: PCIE_ASPM_ON_AC="default"
defaults.conf L0023: PCIE_ASPM_ON_BAT="default"
defaults.conf L0024: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0025: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0026: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0027: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0028: WIFI_PWR_ON_AC="off"
defaults.conf L0029: WIFI_PWR_ON_BAT="on"
defaults.conf L0030: WOL_DISABLE="Y"
defaults.conf L0031: SOUND_POWER_SAVE_ON_AC="0"
defaults.conf L0032: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0033: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0034: BAY_POWEROFF_ON_AC="0"
defaults.conf L0035: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0036: BAY_DEVICE="sr0"
defaults.conf L0037: RUNTIME_PM_ON_AC="on"
defaults.conf L0038: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0039: RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me nouveau nvidia pcieport radeon"
defaults.conf L0040: USB_AUTOSUSPEND="1"
defaults.conf L0041: USB_BLACKLIST_BTUSB="0"
defaults.conf L0042: USB_BLACKLIST_PHONE="0"
defaults.conf L0043: USB_BLACKLIST_PRINTER="1"
defaults.conf L0044: USB_BLACKLIST_WWAN="0"
defaults.conf L0045: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"
defaults.conf L0046: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0047: RESTORE_THRESHOLDS_ON_BAT="0"
defaults.conf L0048: NATACPI_ENABLE="1"
defaults.conf L0049: TPACPI_ENABLE="1"
defaults.conf L0050: TPSMAPI_ENABLE="1"

+++ System Info
System         = LENOVO ThinkPad E595 20NF0006GE
BIOS           = R11ET39W (1.19 )
Release        = "Arch Linux"
Kernel         = 5.14.9-arch2-1 #1 SMP PREEMPT Fri, 01 Oct 2021 19:03:20 +0000 x86_64
/proc/cmdline  = initrd=\initramfs-linux.img cryptdevice=/dev/nvme0n1p2:main root=/dev/mapper/main-home rw lang=de init=/usr/lib/systemd/systemd locale=de_DE.UTF-8
Init system    = systemd 
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = not installed
Last run       = 17:19:53,      8 sec(s) ago
Mode           = battery
Power source   = battery

[... 192 lines cut ...]

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = active (data, thresholds)
tpacpi-bat = active (recalibrate)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = Celxpert
/sys/class/power_supply/BAT0/model_name                     = 5B10W138
/sys/class/power_supply/BAT0/cycle_count                    =    155
/sys/class/power_supply/BAT0/energy_full_design             =  45730 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  40640 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  23370 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  12318 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

/sys/class/power_supply/BAT0/charge_start_threshold         =     95 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   57.5 [%]
Capacity                                                    =   88.9 [%]

+++ Recommendations
* Install ethtool to disable Wake On LAN (WOL)

linrunner, 2021-10-05:
The runtime pm settings of the PCIe devices are identical. At the moment I am at a loss as to what the cause could be.

Please try the workaround from the FAQ with 1.4.0 now.

linrunner, 2021-10-07:
Well ... let me think about that. Stay tuned please.

linrunner, 2021-10-07:
You rebooted for every test, right?

linrunner, 2021-10-07:
Next try with 1.4.0: please edit **/usr/sbin/tlp** function [*apply_common_settings()*](https://github.com/linrunner/TLP/blob/main/tlp.in#L20), moving line 20

         set_runtime_pm $1

after line 45

        set_sound_power_mode $1 

Reboot.

tomizius, 2021-10-07:
Sorry, but I don't get you.
Both is set at this file.
```set_runtimer_pm $1```is in the code at line 34 and ```set_sound_power_mode $1```is at line 45 in my file.
What do you want, should I remove them? 
Or move the postition in the bash code?

linrunner, 2021-10-07:
OK. Lets try to disable PCIe runtime pm completely:

```
RUNTIME_PM_ON_AC=
RUNTIME_PM_ON_BAT=
```

Reboot :)

linrunner, 2021-10-12:
Good. It would be great if you could narrow down which driver or device is to blame. To do this you need to re-enable runtime pm and then denylist individual devices or drivers step by step --> see the [Troubleshooting Guide](https://linrunner.de/tlp/support/troubleshooting.html#step-6-isolate-the-causing-device).

linrunner, 2021-11-05:
> There is one other problem which occurs since version 1.4.0 and not with 1.3.1. If my notebook goes in suspend to RAM I need a hard reset with 1.4.0, it doesn't come back by itself.
> Should I open for that another issue? Or is that simply irrelevant for you?
Similar problem here: https://github.com/linrunner/TLP/issues/593

linrunner, 2021-11-05:
Because the ahci driver already causes trouble and you excluded it from runtime pm I'd try to disable ahci runtime pm as well with:

    AHCI_RUNTIME_PM_ON_AC=
    AHCI_RUNTIME_PM_ON_BAT=

Slater91, 2021-11-25:
I can confirm that this is an issue on my laptop as well (HP EliteBook 745 G5 with a Ryzen 5 2500U). I was thinking that the issue was with the hardware as it was present with older, known-working kernels as well as the latest ones! Interestingly enough, the problem kept on being there even when disabling the TLP service.
The proposed fixes do indeed work.

linrunner, 2021-12-08:
@tomizius : so how do you wan't to proceed? Close?

sleeyax, 2021-12-08:
> @tomizius : so how do you want to proceed? Close?

What's the actual fix to this though? I have the exact same issue with 0 additional configuration and by reading this thread I'm still not sure what the solution is.

Slater91, 2021-12-08:
> > @tomizius : so how do you want to proceed? Close?
> 
> What's the actual fix to this though? I have the exact same issue with 0 additional configuration and by reading this thread I'm still not sure what the solution is.

The solution is this:
`AHCI_RUNTIME_PM_ON_AC=
AHCI_RUNTIME_PM_ON_BAT=`
Just remove anything after those two values and things should start working again.

As this issue was not present on versions before 1.4, though, I would personally say that digging a bit deeper into the causes and actually fixing the problem would be ideal, also because people who are going to install TLP on their systems might not see this bug report (I have had to point at least two people to this in the past week alone).

linrunner, 2021-12-09:
@Slater91 : the bottom line is: TLP exposes a kernel issue in the _ahci_ driver. If you want to address it, open a kernel bug. 

In TLP, I could only disable the feature by default, as was the case until 1.3. But then there is no benefit, because the majority of users runs the default configuration.

tomizius, 2021-12-09:
I understand the procedure of linrunner, that you wont disable the feature by default.
For me it's also fine, if you close this issue, because there is a workaround. Unfortunately I had no time to open a kernel bug, but I hope I find the time during around the Christmas holidays. 

@sleeyax right now I'm just running Version 1.3.1 because it's easier for me to disable the update, then exclude all buggy kernel drivers

linrunner, 2021-12-14:
> There is one other problem which occurs since version 1.4.0 and not with 1.3.1. If my notebook goes in suspend to RAM I need a hard reset with 1.4.0, it doesn't come back by itself.

Workaround added to the [FAQ](https://linrunner.de/tlp/faq/operation.html#system-freezes-on-wakeup-from-suspend-on-battery).

linrunner, 2022-01-26:
@tomizius are there news from the kernel front?

liberodark, 2022-03-06:
Same for me on my side have revert to 1.3.1

linrunner, 2022-03-06:
@liberodark : Stating only "same for me" is just a waste of attention and bandwidth. 

Where is your `tlp-stat` output? 

Did you try _both_ workarounds from the FAQ:

1. https://linrunner.de/tlp/faq/usb.html#newly-inserted-usb-devices-not-recognized-in-battery-mode
2. https://linrunner.de/tlp/faq/operation.html#system-freezes-on-wakeup-from-suspend-on-battery

and what was the result?

linrunner, 2023-07-16:
Since there is no further feedback, I consider this issue to be solved either with the workaround or with the resulting change  (https://github.com/linrunner/TLP/commit/56bb6ce2a11ff3f2424be6452af73f7a1cabf158).

closed 2023-07-16 by linrunner

linrunner, 2023-07-22:
Hi @ALL : TLP 1.6 Beta 1 is out and contains a fix for this issue -> https://github.com/linrunner/TLP/issues/700

linrunner, 2024-09-05:
Hi https://github.com/ALL : TLP 1.7 Beta 2 is out and contains another fix for this issue -> https://github.com/linrunner/TLP/issues/760

Slater91, 2024-10-25:
Unfortunately it seems like version 1.7 re-introduces the problem, despite the fix and the workarounds. Version 1.6.1, which I was using, worked without a hitch; version 1.7 appears to not work and I have to manually reset the USB subsystem every single time the computer goes on battery power. Disabling TLP makes things work as expected just to be clear.

Please find my tlp-stat output here: 
[tlp-stat.txt](https://github.com/user-attachments/files/17521517/tlp-stat.txt)

linrunner, 2024-10-25:
@Slater91 Why do you show the output for AC if the problem occurs in BAT mode? I would also like to have the output for 1.6.1.

Slater91, 2024-10-25:
Because I had no idea the output would be different. I don't know the ins and outs of TLP, so please tone it down. Here is the output while on battery:
[tlp-stat.txt](https://github.com/user-attachments/files/17521931/tlp-stat.txt)
You may notice that I disabled USB autosuspend entirely in the meantime, unfortunately to no avail.

The configuration was identical. I'm not sure where to get version 1.6.1 as apt only reports 1.7.0 as available.

linrunner, 2024-10-25:
> so please tone it down.

The question was meant as a suggestion to think along, nothing more.

>  You may notice that I disabled USB autosuspend entirely in the meantime, unfortunately to no avail.

I need to compare the result of 1.7 and 1.6.1. Please show the 1.6.1 tlp-stat output with identical settings.