linrunner/TLP #595: [Lenovo ThinkPad T14s Gen 1 AMD] discharge BAT0 malfunction

ncfavier, 2021-11-04:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

```
$ sudo tlp recalibrate
Setting temporary charge thresholds for BAT0:
  stop  = 100
  start =  96
Initiating discharge of battery BAT0 ...............
Error: discharge BAT0 malfunction -- check your hardware (battery, charger).
Battery recalibration aborted.
```

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
Only relevant for AC.
2. Actions to reproduce the behaviour
See description
3. Shell commands entered and their output
See description
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**
https://gist.github.com/ncfavier/f578877763fc99afad32dde333ecb5f3

**Additional context**

The laptop and charger are brand new and otherwise working properly. The battery is internal.

`tlp discharge` seems to work:
```
Currently discharging battery BAT0:
voltage            =  11902 [mV]
remaining capacity =  35120 [mWh]
remaining percent  =     60 [%]
remaining time     =    312 [min]
power              =   6736 [mW]
state              = Discharging
force discharge    = 1
Press Ctrl+C to cancel.
```

linrunner, 2021-11-07:
Hi,

`tlp recalibrate` and `tlp discharge` use the same code section to enable discharge. A loop waits until the battery actually discharges. Every second it outputs a dot, for a maximum of 15 seconds.

With my X1C6 (Intel) I have observed very different times for the process. It takes up to about 10 seconds.

How many dots does _tlp discharge_ output when successful?

ncfavier, 2021-11-19:
Actually this issue only happens when the battery is charged above the start threshold: running `recalibrate` sets the threshold to 96 so the battery transitions from Unknown to Charging first, and the command fails. Running it again transitions from Charging to Discharging and works.

Now I'm hitting another issue mentioned in the FAQ: after reaching 0%, `recalibrate` fails with `Error: battery BAT0 was not discharged completely i.e. terminated by the firmware -- check your hardware.`. Same brand new hardware.

linrunner, 2021-11-21:
As said above: the output of `tlp recalibrate` in exactly the error situation would be helpful. 

About the 0%: everything is said in the FAQ, i do not intend to intercept every peculiarity (bug?) of the Lenovo firmware.

ncfavier, 2021-11-21:
```
$ # battery is at 80% Unknown
$ sudo tlp recalibrate
Setting temporary charge thresholds for BAT0:
  stop  = 100
  start =  96
Initiating discharge of battery BAT0 ...............
Error: discharge BAT0 malfunction -- check your hardware (battery, charger).
Battery recalibration aborted.
$ # battery is now at 80% Charging
$ sudo tlp recalibrate
Setting temporary charge thresholds for BAT0:
  start =  96 (no change)
  stop  = 100 (no change)
Initiating discharge of battery BAT0 ..
[screen clears]
Currently discharging battery BAT0:
voltage            =  12184 [mV]
remaining capacity =  47270 [mWh]
remaining percent  =     80 [%]
remaining time     =    217 [min]
power              =  13036 [mW]
state              = Discharging
force discharge    = 1
Press Ctrl+C to cancel.
```

linrunner, 2021-11-24:
I enhanced the trace output and would like you to retest this situation with the main branch. After installing, please enable trace mode by adding

    TLP_DEBUG="bat ps run"

to your configuration.

Then i need again the output of `tlp recalibrate` and the trace output:

    sudo tlp-stat -T

ncfavier, 2021-11-24:
On f7b7ff747ab1fb479cd83d6bc25902ab70ecdbf9:

<details>
<summary> tlp-recalibrate </summary>

```
Setting temporary charge thresholds for BAT0:
  stop  = 100
  start =  96
Initiating discharge of battery BAT0 ...............
Error: discharge BAT0 malfunction -- check your hardware (battery, charger).
Battery recalibration aborted.
```
</details>

<details>
<summary> tlp-stat -T </summary>

```
--- TLP 1.5.0-alpha.0 --------------------------------------------

-- Journal begins at Sat 2021-10-30 01:54:34 CEST, ends at Wed 2021-11-24 20:08:21 CET. --
Nov 24 05:01:08 no tlp[2745]: Applying power save settings...done.
Nov 24 05:01:08 no tlp[2745]: Setting battery charge thresholds...done.
Nov 24 20:07:10 no tlp[547697]: +++ init start (1.5.0-alpha.0) ++++++++++++++++++++++++++++++++++++++++
Nov 24 20:07:10 no tlp[547697]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 24 20:07:10 no tlp[547697]: power_source=ac
Nov 24 20:07:10 no tlp[547697]: manual_mode=none
Nov 24 20:07:10 no tlp[547697]: power_mode=ac
Nov 24 20:07:10 no tlp[547697]: compare_and_save_power_state(0).different: old=
Nov 24 20:07:10 no tlp[547697]: Applying power save settings...done.
Nov 24 20:07:10 no tlp[547697]: check_thinkpad: tpmodel=T14s Gen 1
Nov 24 20:07:10 no tlp[547697]: batdrv_init.thinkpad: batteries=BAT0; natacpi=1; tpacpi=0; tpsmapi=254
Nov 24 20:07:10 no tlp[547697]: batdrv_init.thinkpad: read=natacpi; thresh=natacpi; bn_start=charge_control_start_threshold; bn_stop=charge_control_end_threshold; dischg=tpacpi; bn_dischg=
Nov 24 20:07:10 no tlp[547697]: batdrv.thinkpad.select_battery(BAT0): bat_str=BAT0; bat_idx=1; bd_read=/sys/class/power_supply/BAT0; bf_start=/sys/class/power_supply/BAT0/charge_control_start_threshold; bf_stop=/sys/class/power_supply/BAT0/charge_control_end_threshold; bf_dischg=
Nov 24 20:07:10 no tlp[547697]: batdrv.thinkpad.read_threshold(start): bm_thresh=natacpi; bf=/sys/class/power_supply/BAT0/charge_control_start_threshold; bat_idx=1; out=75; rc=0
Nov 24 20:07:10 no tlp[547697]: batdrv.thinkpad.read_threshold(stop): bm_thresh=natacpi; bf=/sys/class/power_supply/BAT0/charge_control_end_threshold; bat_idx=1; out=80; rc=0
Nov 24 20:07:10 no tlp[547697]: batdrv.thinkpad.write_thresholds(75, 80, 1, BAT0).start.no_change: bat=BAT0; old=75; new=75
Nov 24 20:07:10 no tlp[547697]: batdrv.thinkpad.write_thresholds(75, 80, 1, BAT0).stop.no_change: bat=BAT0; old=80; new=80
Nov 24 20:07:10 no tlp[547697]: batdrv.thinkpad.write_thresholds(75, 80, 1, BAT0).complete: bat=BAT0; rc=0
Nov 24 20:07:10 no tlp[547697]: batdrv.thinkpad.select_battery(BAT1).not_present
Nov 24 20:07:10 no tlp[547697]: Setting battery charge thresholds...done.
Nov 24 20:07:10 no tlp[547697]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpIUkRqK -> /run/tlp/run.conf
Nov 24 20:07:59 no tlp[548384]: +++ recalibrate (1.5.0-alpha.0) ++++++++++++++++++++++++++++++++++++++++
Nov 24 20:07:59 no tlp[548384]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 24 20:07:59 no tlp[548384]: power_source=ac
Nov 24 20:07:59 no tlp[548384]: manual_mode=none
Nov 24 20:07:59 no tlp[548384]: power_mode=ac
Nov 24 20:07:59 no tlp[548384]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 24 20:07:59 no tlp[548384]: check_thinkpad: tpmodel=T14s Gen 1
Nov 24 20:07:59 no tlp[548384]: batdrv_init.thinkpad: batteries=BAT0; natacpi=1; tpacpi=0; tpsmapi=254
Nov 24 20:07:59 no tlp[548384]: batdrv_init.thinkpad: read=natacpi; thresh=natacpi; bn_start=charge_control_start_threshold; bn_stop=charge_control_end_threshold; dischg=tpacpi; bn_dischg=
Nov 24 20:07:59 no tlp[548384]: batdrv.thinkpad.select_battery(DEF): bat_str=BAT0; bat_idx=1; bd_read=/sys/class/power_supply/BAT0; bf_start=/sys/class/power_supply/BAT0/charge_control_start_threshold; bf_stop=/sys/class/power_supply/BAT0/charge_control_end_threshold; bf_dischg=
Nov 24 20:07:59 no tlp[548384]: batdrv.thinkpad.read_threshold(start): bm_thresh=natacpi; bf=/sys/class/power_supply/BAT0/charge_control_start_threshold; bat_idx=1; out=75; rc=0
Nov 24 20:07:59 no tlp[548384]: batdrv.thinkpad.read_threshold(stop): bm_thresh=natacpi; bf=/sys/class/power_supply/BAT0/charge_control_end_threshold; bat_idx=1; out=80; rc=0
Nov 24 20:07:59 no tlp[548384]: batdrv.thinkpad.write_thresholds(DEF, DEF, 2, ).stop.write: bat=BAT0; old=80; new=100; steprc=0
Nov 24 20:07:59 no tlp[548384]: batdrv.thinkpad.write_thresholds(DEF, DEF, 2, ).start.write: bat=BAT0; old=75; new=96; steprc=0
Nov 24 20:07:59 no tlp[548384]: batdrv.thinkpad.write_thresholds(DEF, DEF, 2, ).complete: bat=BAT0; rc=0
Nov 24 20:08:00 no tlp[548384]: batdrv.thinkpad.select_battery(DEF): bat_str=BAT0; bat_idx=1; bd_read=/sys/class/power_supply/BAT0; bf_start=/sys/class/power_supply/BAT0/charge_control_start_threshold; bf_stop=/sys/class/power_supply/BAT0/charge_control_end_threshold; bf_dischg=
Nov 24 20:08:00 no tlp[548384]: batdrv.thinkpad.write_force_discharge(BAT0, 1): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; rc=0
Nov 24 20:08:00 no tlp[548384]: batdrv.thinkpad.read_force_discharge(BAT0): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; out=0; rc=0
Nov 24 20:08:00 no tlp[548384]: batdrv.thinkpad.force_discharge_active(BAT0): rc=1
Nov 24 20:08:01 no tlp[548384]: batdrv.thinkpad.read_force_discharge(BAT0): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; out=0; rc=0
Nov 24 20:08:01 no tlp[548384]: batdrv.thinkpad.force_discharge_active(BAT0): rc=1
Nov 24 20:08:02 no tlp[548384]: batdrv.thinkpad.read_force_discharge(BAT0): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; out=0; rc=0
[... cut ...]
Nov 24 20:08:16 no tlp[548384]: batdrv.thinkpad.read_force_discharge(BAT0): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; out=0; rc=0
Nov 24 20:08:16 no tlp[548384]: batdrv.thinkpad.force_discharge_active(BAT0): rc=1
Nov 24 20:08:16 no tlp[548384]: batdrv.thinkpad.write_force_discharge(BAT0, 0): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; rc=0
Nov 24 20:08:16 no tlp[548384]: batdrv.thinkpad.discharge.malfunction(BAT0)
Nov 24 20:08:16 no tlp[548384]: batdrv.thinkpad.read_force_discharge(BAT0): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; out=0; rc=0
```
</details>

linrunner, 2021-11-24:
> Nov 24 20:08:00 no tlp[548384]: batdrv.thinkpad.write_force_discharge(BAT0, 1): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; **rc=0**

_tpacpi-bat_ returns success when setting force-discharge

>  Nov 24 20:08:16 no tlp[548384]: batdrv.thinkpad.read_force_discharge(BAT0): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; **out=0**; rc=0

but the actual state doesn't change. Looks like the EC firmware silently discards force-discharge when the battery status is "Charging". Lets try to reproduce it with _tpacpi-bat_ alone:

Starting point as before: 

- Charge < 96: 
- status  = Charging
- charge_control_start_threshold =  96
- charge_control_end_threshold  =  100
 
```
sudo tlp-stat -b; sudo /usr/share/tlp/tpacpi-bat -v -s FD 1 1; sudo /usr/share/tlp/tpacpi-bat -v -g FD 1; sudo tlp-stat -b

```

ncfavier, 2021-11-24:
When I ran the commands above, the status wasn't Charging, it was Idle/Unknown. It then transitioned to Charging after running `recalibrate` (as described in [this comment](https://github.com/linrunner/TLP/issues/595#issuecomment-974227801)).

I ran your commands twice:

<details>
<summary>Once in Idle status</summary>

```
~ $ sudo tlp-stat -b; sudo /nix/store/fgs2wpn3b97rh4m0pbxbir8c9dck7281-tlp-1.4.1pre/share/tlp/tpacpi-bat -v -s FD 1 1; sudo /nix/store/fgs2wpn3b97rh4m0pbxbir8c9dck7281-tlp-1.4.1pre/share/tlp/tpacpi-bat -v -g FD 1; sudo tlp-stat -b
--- TLP 1.5.0-alpha.0 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds)
* tpacpi-bat (acpi_call)  = active (recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 5B10W139
/sys/class/power_supply/BAT0/cycle_count                    =     12
/sys/class/power_supply/BAT0/energy_full_design             =  57020 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  58330 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  52130 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Idle

/sys/class/power_supply/BAT0/charge_control_start_threshold =     75 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     80 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   89.4 [%]
Capacity                                                    =  102.3 [%]

Call    : \_SB.PCI0.LPC0.EC0.HKEY.BDSS 0x101
Response: 0x0
Call    : \_SB.PCI0.LPC0.EC0.HKEY.BDSG 0x1
Response: 0x701
yes
--- TLP 1.5.0-alpha.0 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds)
* tpacpi-bat (acpi_call)  = active (recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 5B10W139
/sys/class/power_supply/BAT0/cycle_count                    =     12
/sys/class/power_supply/BAT0/energy_full_design             =  57020 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  58330 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  52130 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Idle

/sys/class/power_supply/BAT0/charge_control_start_threshold =     75 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     80 [%]
tpacpi-bat.BAT0.forceDischarge                              =      1

Charge                                                      =   89.4 [%]
[... 48 lines cut ...]
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds)
* tpacpi-bat (acpi_call)  = active (recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 5B10W139
/sys/class/power_supply/BAT0/cycle_count                    =     12
/sys/class/power_supply/BAT0/energy_full_design             =  57020 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  58330 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  52130 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  17319 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     96 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      1

Charge                                                      =   89.4 [%]
Capacity                                                    =  102.3 [%]
```
</details>

linrunner, 2021-11-25:
To be clear: this issue is not about `tlp recalibrate`. `tlp recalibrate` is just `tlp setcharge 96 100` (causing the "Charging")  followed by `tlp discharge`.

The problem is to actually start the forced discharge while "Charging". My last command above is just what `tlp discharge` does first, then it waits for "Discharging". 

Your outputs show that force-discharge is enabled correctly:

> /sys/class/power_supply/BAT0/status = Charging
> tpacpi-bat.BAT0.**forceDischarge =  1**

How long does it take afterwards until the battery turns to "Discharging"?

ncfavier, 2021-11-25:
> How long does it take afterwards until the battery turns to "Discharging"?

About 2-4 seconds.

Weirdly, `sudo tlp setcharge 96 100; sleep 1; sudo tlp discharge` works.

linrunner, 2021-11-25:
Almost. `recalibrate` does not have the `sleep 1`. I still don't get what really happens in TLP. Please post the trace output  for

    sudo tlp setcharge DEF DEF; sudo tlp discharge

ncfavier, 2021-11-25:
> Almost. `recalibrate` does not have the `sleep 1`.

https://github.com/linrunner/TLP/blob/f7b7ff747ab1fb479cd83d6bc25902ab70ecdbf9/tlp.in#L501

`sudo tlp setcharge DEF DEF; sudo tlp discharge` works too.

<details>
<summary>tlp-stat -T</summary>

```
Nov 25 20:30:38 no tlp[1234829]: +++ setcharge (1.5.0-alpha.0) ++++++++++++++++++++++++++++++++++++++++
Nov 25 20:30:38 no tlp[1234829]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 25 20:30:38 no tlp[1234829]: power_source=ac
Nov 25 20:30:38 no tlp[1234829]: manual_mode=none
Nov 25 20:30:38 no tlp[1234829]: power_mode=ac
Nov 25 20:30:38 no tlp[1234829]: check_thinkpad: tpmodel=T14s Gen 1
Nov 25 20:30:38 no tlp[1234829]: batdrv_init.thinkpad: batteries=BAT0; natacpi=1; tpacpi=0; tpsmapi=254
Nov 25 20:30:38 no tlp[1234829]: batdrv_init.thinkpad: read=natacpi; thresh=natacpi; bn_start=charge_control_start_threshold; bn_stop=charge_control_end_threshold; dischg=tpacpi; bn_dischg=
Nov 25 20:30:38 no tlp[1234829]: batdrv.thinkpad.select_battery(DEF): bat_str=BAT0; bat_idx=1; bd_read=/sys/class/power_supply/BAT0; bf_start=/sys/class/power_supply/BAT0/charge_control_start_threshold; bf_stop=/sys/class/power_supply/BAT0/charge_control_end_threshold; bf_dischg=
Nov 25 20:30:38 no tlp[1234829]: batdrv.thinkpad.read_threshold(start): bm_thresh=natacpi; bf=/sys/class/power_supply/BAT0/charge_control_start_threshold; bat_idx=1; out=75; rc=0
Nov 25 20:30:38 no tlp[1234829]: batdrv.thinkpad.read_threshold(stop): bm_thresh=natacpi; bf=/sys/class/power_supply/BAT0/charge_control_end_threshold; bat_idx=1; out=80; rc=0
Nov 25 20:30:38 no tlp[1234829]: batdrv.thinkpad.write_thresholds(DEF, DEF, 2, ).stop.write: bat=BAT0; old=80; new=100; steprc=0
Nov 25 20:30:38 no tlp[1234829]: batdrv.thinkpad.write_thresholds(DEF, DEF, 2, ).start.write: bat=BAT0; old=75; new=96; steprc=0
Nov 25 20:30:38 no tlp[1234829]: batdrv.thinkpad.write_thresholds(DEF, DEF, 2, ).complete: bat=BAT0; rc=0
Nov 25 20:30:38 no tlp[1234931]: +++ discharge (1.5.0-alpha.0) ++++++++++++++++++++++++++++++++++++++++
Nov 25 20:30:40 no tlp[1234931]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 25 20:30:40 no tlp[1234931]: power_source=ac
Nov 25 20:30:40 no tlp[1234931]: manual_mode=none
Nov 25 20:30:40 no tlp[1234931]: power_mode=ac
Nov 25 20:30:40 no tlp[1234931]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 25 20:30:40 no tlp[1234931]: check_thinkpad: tpmodel=T14s Gen 1
Nov 25 20:30:40 no tlp[1234931]: batdrv_init.thinkpad: batteries=BAT0; natacpi=1; tpacpi=0; tpsmapi=254
Nov 25 20:30:40 no tlp[1234931]: batdrv_init.thinkpad: read=natacpi; thresh=natacpi; bn_start=charge_control_start_threshold; bn_stop=charge_control_end_threshold; dischg=tpacpi; bn_dischg=
Nov 25 20:30:40 no tlp[1234931]: batdrv.thinkpad.select_battery(DEF): bat_str=BAT0; bat_idx=1; bd_read=/sys/class/power_supply/BAT0; bf_start=/sys/class/power_supply/BAT0/charge_control_start_threshold; bf_stop=/sys/class/power_supply/BAT0/charge_control_end_threshold; bf_dischg=
Nov 25 20:30:40 no tlp[1234931]: batdrv.thinkpad.write_force_discharge(BAT0, 1): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; rc=0
Nov 25 20:30:41 no tlp[1234931]: batdrv.thinkpad.read_force_discharge(BAT0): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; out=1; rc=0
Nov 25 20:30:41 no tlp[1234931]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 25 20:30:41 no tlp[1234931]: batdrv.thinkpad.force_discharge_active(BAT0): bm_read=natacpi; bf=/sys/class/power_supply/BAT0/status; st=Unknown; rc=1
Nov 25 20:30:41 no tlp[1235055]: +++ auto (1.5.0-alpha.0) ++++++++++++++++++++++++++++++++++++++++
Nov 25 20:30:41 no tlp[1235055]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 25 20:30:41 no tlp[1235055]: power_source=ac
Nov 25 20:30:41 no tlp[1235055]: manual_mode=none
Nov 25 20:30:41 no tlp[1235055]: power_mode=ac
Nov 25 20:30:41 no tlp[1235055]: compare_and_save_power_state(0).equal
Nov 25 20:30:42 no tlp[1234931]: batdrv.thinkpad.read_force_discharge(BAT0): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; out=1; rc=0
Nov 25 20:30:42 no tlp[1234931]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 25 20:30:42 no tlp[1234931]: batdrv.thinkpad.force_discharge_active(BAT0): bm_read=natacpi; bf=/sys/class/power_supply/BAT0/status; st=Not charging; rc=1
Nov 25 20:30:43 no tlp[1234931]: batdrv.thinkpad.read_force_discharge(BAT0): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; out=1; rc=0
Nov 25 20:30:43 no tlp[1234931]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 25 20:30:43 no tlp[1234931]: batdrv.thinkpad.force_discharge_active(BAT0): bm_read=natacpi; bf=/sys/class/power_supply/BAT0/status; st=Discharging; rc=0
Nov 25 20:30:43 no tlp[1234931]: batdrv.thinkpad.read_force_discharge(BAT0): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; out=1; rc=0
Nov 25 20:30:43 no tlp[1234931]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 25 20:30:43 no tlp[1234931]: batdrv.thinkpad.force_discharge_active(BAT0): bm_read=natacpi; bf=/sys/class/power_supply/BAT0/status; st=Discharging; rc=0
Nov 25 20:30:43 no tlp[1234931]: batdrv.thinkpad.discharge.running(BAT0)
Nov 25 20:30:43 no tlp[1234931]: batdrv.thinkpad.read_force_discharge(BAT0): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; out=1; rc=0
Nov 25 20:30:43 no tlp[1234931]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 25 20:30:43 no tlp[1234931]: batdrv.thinkpad.force_discharge_active(BAT0): bm_read=natacpi; bf=/sys/class/power_supply/BAT0/status; st=Discharging; rc=0
[... cut ...]
Nov 25 20:30:46 no tlp[1234931]: batdrv.thinkpad.write_force_discharge(BAT0, 0): bm_dischg=tpacpi; bf_dischg=; bat_idx=1; rc=0
Nov 25 20:30:46 no tlp[1234931]: batdrv.thinkpad.discharge.cancelled(BAT0)
Nov 25 20:30:47 no tlp[1235274]: +++ auto (1.5.0-alpha.0) ++++++++++++++++++++++++++++++++++++++++
Nov 25 20:30:47 no tlp[1235274]: get_sys_power_supply(AC).ac_online: syspwr=0
Nov 25 20:30:47 no tlp[1235274]: power_source=ac
Nov 25 20:30:47 no tlp[1235274]: manual_mode=none
Nov 25 20:30:47 no tlp[1235274]: power_mode=ac
Nov 25 20:30:47 no tlp[1235274]: compare_and_save_power_state(0).equal
```
</details>

linrunner, 2021-11-26:
> Nov 25 20:30:42 no tlp[1234931]: batdrv.thinkpad.force_discharge_active(BAT0): bm_read=natacpi; bf=/sys/class/power_supply/BAT0/status; st=Not charging; rc=1

An initial "Not charging" is not the necessary condition to reproduce the problem. The "malfunction" message happens when `setcharge DEF DEF` makes the battery "Charging".

We need a starting point like above

>  battery is at 80% Unknown

<= 94% is also sufficient.

Or does your hardware behave differently each time?

linrunner, 2021-11-26:
ps.

> https://github.com/linrunner/TLP/blob/f7b7ff747ab1fb479cd83d6bc25902ab70ecdbf9/tlp.in#L501

Oops. Maybe i should look into my code from time to time ;-). But the `sleep 1` doesn't cause the problem at hand.

ncfavier, 2021-11-26:
> We need a starting point like above
> 
> > battery is at 80% Unknown

That was the starting point! See the earlier line:

> Nov 25 20:30:41 no tlp[1234931]: batdrv.thinkpad.force_discharge_active(BAT0): bm_read=natacpi; bf=/sys/class/power_supply/BAT0/status; st=Unknown; rc=1

linrunner, 2021-11-26:
All right. But then we still have no trace to the actual error situation. 

My hypothesis for the occurrence of the error is:

1. `setcharge DEF DEF` results in battery status "Charging" 
1. `discharge` sets force-discharge for the battery
1. after 15 seconds the status still has not reached "Discharging"
1. waiting time expired resulting in an abort

Probably your hardware behaves differently sometimes. In the TLP logic, on the other hand, I can see no problem so far.

I suggest you make more attempts to get a trace that shows the timeout.

linrunner, 2021-12-21:
Any news on this?

ncfavier, 2021-12-21:
No, sorry. Feel free to close for now

closed 2022-01-04 by linrunner