linrunner/TLP #596: Battery care support for Acer laptops

wojpawlik, 2021-11-06:
I own Acer SF314-43. What tools can I use to investigate how [Acer Care Center
](https://www.acer.com/ac/en/US/content/software-acer-care-center) communicates with hardware?

linrunner, 2021-11-07:
I have no idea, but I leave it open here, in case someone who reads here knows it.

dxmuses, 2022-02-28:
@wojpawlik Hey, I have the exact same laptop! I'm running a windows/void linux dualboot and I have a curious observation to report. I limited the battery charging threshold to 80% using the acer care center and it works exactly as advertised on windows. But somehow, this setting persists even when running linux! So, the battery stops charging at 80 even inside linux. Acer care center is just a normal windows program but strangely its setting survives across different operating systems just as if it were a bios setting.

It would be great if this gets working in tlp so people won't have to set up a dualboot just to limit the battery charge percentage. Here's hoping someone figures this one out. Good luck folks!

Bartvelp, 2022-03-01:
Just to chime in, @dxmuses. Setting a battery charging limit in the Acer Care Center and then booting into linux worked for me as well. But at some time, it just stopped charging even when it was below 80%. This was actually really annoying since windows wanted to update and did not boot because my battery was below 25%, but I also couldn't charge it. So this method can actually be sort of "dangerous".
Another reason that getting it to work in TLP would be nice.

dxmuses, 2022-03-01:
@Bartvelp That sounds rough. Sorry if I landed you in some trouble! In my case though, I've had this setting turned on for about a week now and its been working fine between linux and windows so far. There could be so many variations between your setup and mine that its hard to pinpoint what could be causing this strange behavior [eg: might be a different (bios/kernel/windows version/some other setting?)].

But you're totally right, it'd be much safer and cleaner to have this working in tlp. Someday, I hope!

balthier82, 2022-03-03:
Hi guys, I have an Acer Nitro 5. I would really need the plugin for this laptop. I hope someone can figure out how to do this.

linrunner, 2022-03-06:
Please save your declarations of wanting to have. They do nothing at all here in the issue. One of you Acer owners has to go and program a corresponding extension of the [kernel driver](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/platform/x86/acer-wmi.c?id=HEAD). Writing the plugin is then trivial.

frederik-h, 2022-06-03:
> Please save your declarations of wanting to have. They do nothing at all here in the issue. One of you Acer owners has to go and program a corresponding extension of the [kernel [...]

I have implemented an experimental kernel [driver](https://github.com/frederik-h/acer-wmi-battery/) for Acer's WMI  battery control interface. So far I have only tested it on a Swift 3, but I would expect that Acer uses the same interface on other laptops as well. If you give it a try, please let me know if it works on your system.

linrunner, 2022-06-03:
@frederik-h  Great. It's nice when the light appears at the end of the tunnel. I suggest you include a note in your driver documentation about the percentage where charging stops when health mode is active. I will need the value for the TLP docs as well.

Are you planning to submit the driver for the mainline kernel?

frederik-h, 2022-06-03:
> @frederik-h Great. It's nice when the light appears at the end of the tunnel. I suggest you include a note in your driver documentation about the percentage where charging stops when health mode is active. I will need the value for the TLP docs as well.

I will add it. It stops at 79% on my system. I could not find any methods to modify this. From the Acer Care Center screenshots and documentation that I have seen, I assume that the threshold is fixed. I have no Windows installation to verify the actual settings that are available in the Care Center.

> Are you planning to submit the driver for the mainline kernel?

Yes, but it would be great to test it on some more systems beforehand. I am also a bit unsure if it wouldn't be better to offer the settings separately for all batteries found in the system. The WMI methods take the battery index as an argument, but all invocations of this method that I could find in the Care Center binaries seemed to call it for the first battery only. Hence, I decided to implement it like this as well.

Bartvelp, 2022-06-05:
> Yes, but it would be great to test it on some more systems beforehand. 

Just tested it out on a Acer Aspire 5 A515-45G-R5A1, works like a charm.

> I will add it. It stops at 79% on my system. I could not find any methods to modify this threshold. From the Acer Care Center screenshots and documentation that I have seen, I assume that the threshold is fixed. I have no Windows installation to verify the actual settings that are available in the Care Center.

Stops at 79% for me as well. I do have a windows install and can confirm the battery care toggle is modified when toggling using your kernel driver. There is no way to modify the threshold in Acer Care Center, as you already mentioned.

Thanks for your work @frederik-h , I am interested as to how you reverse engineered the care center, any plans on disclosing that?

linrunner, 2022-06-05:
OK. I assume the charge threshold is 80% and the 79% is due to rounding.

linrunner, 2022-06-07:
> The WMI methods take the battery index as an argument, but all invocations of this method that I could find in the Care Center binaries seemed to call it for the first battery only. Hence, I decided to implement it like this as well.

@frederik-h : On other laptops with a fixed threshold, it seems to apply to all batteries. For Lenovo Non-ThinkPads, LG, Samsung and Sony I just assumed so when creating their plugins, see [BC Vendor Specifics](https://linrunner.de/tlp/settings/bc-vendors.html). Apart from ThinkPads, devices with multiple batteries seem to be quite rare anyway.  

Btw: how can I check for an Acer laptop: probing for the **/sys/devices/platform/acer-wmi/** directory?

linrunner, 2022-06-08:
@frederik-h : I would like to first detect if it is an Acer (to load the plugin) and then if the threshold feature is present (e.g. for display in tlp-stat -b).

Have you thought about adding the functionality to _acer-wmi_ instead of creating your own driver?

frederik-h, 2022-06-09:
> @frederik-h : I would like to first detect if it is an Acer (to load the plugin) and then if health mode is present (e.g. for display in tlp-stat -b).

Right, probing for the `acer-wmi` directory seems reasonable to me then, although the module seems (according to the comments and quirks-handling in its source) to support some laptops by other vendors as well. 

> Have you thought about adding the functionality to _acer-wmi_ instead of creating your own driver?

Yes, but I have implemented it for myself in the first place and doing a standalone module seemed easier to me. Since there is some interest in the module, I will try to upstream it time permits. If it will be necessary to integrate it into `acer-wmi`, this should be easily possible, given how simple the module is. One potential advantage is that sysfs files could live under the same path as the other Acer-related files.

linrunner, 2022-06-09:
@frederik-h : To be honest, I don't really want to release the Acer plugin until the feature is in the mainline kernel. Otherwise, too much support for the external kernel module ends up here. I've had this experience with ThinkPads for > 10 years and I won't give it to myself again.

I'll think about it.

MatteoGalletta, 2022-08-13:
> If you give it a try, please let me know if it works on your system.

Thanks @frederik-h, the driver is working on my Acer Aspire 1 A114-33-C4FF, stops as well at 79%.
Tested on Manjaro (5.15.60-1), managed to install the kernel headers executing [this](https://forum.manjaro.org/t/how-do-i-update-my-kernel-header/93075).

wojpawlik, 2022-08-20:
I confirm the driver working on Acer Swift SF314-43. Any plans on upstreaming?

Sasikuttan2163, 2022-09-09:
@frederik-h the driver works well with on my  Acer Nitro 5 AN515-44. Stops charging at 80%.

Sasikuttan2163, 2022-09-09:
@frederik-h the driver works well on my Acer Nitro 5, AN515-44. Stops charging at 79%.

re-ynd, 2022-09-17:
On Acer Aspire 7 running linux 5.19.7-1-MANJARO, I am getting permission denied error
```
sudo echo 1 > /sys/bus/wmi/drivers/acer-wmi-battery/health_mode
zsh: permission denied: /sys/bus/wmi/drivers/acer-wmi-battery/health_mode
```

MatteoGalletta, 2022-09-17:
> On Acer Aspire 7 running linux 5.19.7-1-MANJARO, I am getting permission denied error
> ```
> sudo echo 1 > /sys/bus/wmi/drivers/acer-wmi-battery/health_mode
> zsh: permission denied: /sys/bus/wmi/drivers/acer-wmi-battery/health_mode
> ```

It's not related to the script itself, I faced this problem before and it's related to the shell sudo behaviour, check [https://stackoverflow.com/q/84882](this)

re-ynd, 2022-09-17:
Ah perfect! That was dumb on my part. Thanks! It works on Acer Aspire 7 running 5.19.7-1-MANJARO.

tarruda, 2022-09-20:
@frederik-h thank you, also working on Acer Predator Helios 300 (2021 model)

mnlipp, 2022-11-13:
I tried this on an Aspire 5 A515-45 and a Swift 1 SF113-31.

`echo 1 >  /sys/bus/wmi/drivers/acer-wmi-battery/health_mode` does not give an error, but 
`cat /sys/bus/wmi/drivers/acer-wmi-battery/health_mode` returns 0 (before and after the echo).

So I suppose it doesn't work. But as this device has never seen Windows, therefore I cannot tell if the hardware is supposed to support this (cannot check using "Acer Care Center").

frederik-h, 2022-11-14:
> I tried this on an Aspire 5 A515-45 and a Swift 1 SF113-31.
>
> `echo 1 > /sys/bus/wmi/drivers/acer-wmi-battery/health_mode` does not give an error, but `cat [...]

I have opened issue frederik-h/acer-wmi-battery#9 for further discussion on this.

Dmivaka, 2022-12-14:
@frederik-h Tried it on Acer Extensa EX215-22-R2H8. Since my OS is Windows (and for some unknown reasons battery care cannot be enabled on my machine here) I've booted into Ubuntu 20.04 from USB stick and enabled health mode with your module. After booting back into Windows the battery discharged to 80% and is still there. Thank you!

KrulesExGamer, 2023-03-03:
Worked like a charm on Acer Aspire A315-23G

Vinz-S, 2023-08-09:
> I'm getting `make[1]: *** /lib/modules/5.17.9-300.fc36.x86_64/build: No such file or directory. Stop. ` when trying to build the module on Fedora 36, despite having `kernel-headers` installed.

I had the same issue because I installed the kernel-devel package from dnf, which was for another kernel version. I got the correct version from https://koji.fedoraproject.org/koji/packageinfo?buildStart=200&packageID=8&buildOrder=-completion_time&tagOrder=name&tagStart=0#buildlist where you first have to find your kernel, for you that would be 5.17.9-300.fc36, and then the correct devel, installing that should fix that issue.

If you have secure boot enabled you'll also encounter an issue related to keys, which you can read more about here: https://docs.fedoraproject.org/en-US/fedora/latest/system-administrators-guide/kernel-module-driver-configuration/Working_with_Kernel_Modules/#sect-signing-kernel-modules-for-secure-boot , haven't gotten to doing it myself yet.

tien-vo, 2023-09-12:
@frederik-h Works on my Acer Swift Edge SFA16-41. Thanks for your work. Would love to see this upstream at some point!

ShayonRoy, 2024-07-10:
@frederik-h It works totally fine on my Acer Aspire 7 A715-42G. Installed using the script from Diman119's script. But to make the health mode stick more then say 10 15 days without having to reenable it manually i had to add "options acer-wmi-battery enable_health_mode=1" in the /etc/modprobe.d/acer-wmi-battery.conf file.