linrunner/TLP #597: Support for Thinkbooks

atar-axis, 2021-11-08:
**Is your feature request related to a problem? Please describe.**

TLP does not work on Thinkbooks, at leas not on 13s-IWL

**Describe the solution you'd like**

I would like to supply all necessary information to support this model too, just tell me what you need.

linrunner, 2021-11-09:
Generally for feature requests I expect that people at least make the effort to say which function is missing. "Does not work" is in no way sufficient.

If this is about charge thresholds: as long as there is no suitable kernel driver for ThinkBooks, there will be no support for it. 

If you mean that I want to develop a kernel driver with you by remote reengineering on your hardware: you must be joking. Besides, I don't do kernel development. TLP makes use of existing kernel drivers.

closed 2021-11-09 by linrunner

atar-axis, 2021-11-09:
> TLP makes use of existing kernel drivers.

Yeah, that's what I was missing. :) thanks!
I will then talk to those.

46cv8, 2022-08-29:
Hey @atar-axis , Did you find out any information regarding this elsewhere? If so would you care to link to other issues or forums threads. I'm curious on the status of this too. Also you don't happen to know will setting charge thresholds via "vantage" in windows update the firmware so they are respected when running linux? If so I will probably just keep a usb bootable windows handy to do this.

linrunner, 2022-08-30:
> EDIT: For anyone else who may find this, it seems the ideapad in vantage doesn't even allow for one to define custom charge thresholds, they only allow for a conservation mode that restricts battery charge to no more tahn 60-65% (I think). Luckily this is exposed by more recent linux kernels. The following post may be of assistance to anyone who wants to use it in linux. https://askubuntu.com/questions/1038471/problem-with-lenovo-battery-threshold

TLP supports Lenovo Non-ThinkPad series with the _ideapad_laptop_ driver since 1.4. Do your research first next time.

https://linrunner.de/tlp/settings/bc-vendors.html

46cv8, 2022-08-30:
> Do your research first next time.

Yeah I'm not sure how I missed that during my "research" glad to see it's supported even if it is just to expose that one toggle. 
I was just trying to share a relevant response that might help others. It's good to have the link to the official documentation though showing how TLP exposes the functionality. Not sure if this is a recent addition to TLP or not, but it seems like the only available functionality in the battery firmware "conservation mode" is already exposed by the kernel driver, so I'm not sure what exact functionality he expected them to add.

TurnOffNOD, 2024-08-26:
> TLP supports Lenovo Non-ThinkPad series with the _ideapad_laptop_ driver since 1.4. Do your research first next time.
>
> https://linrunner.de/tlp/settings/bc-vendors.html#lenovo-non-thinkpad-series

However, thinkbook series (and thinkbook+ series) are not ideapad, so this is not clearly enough.

linrunner, 2024-08-26:
@TurnOffNOD done.

TurnOffNOD, 2024-08-30:
> There will always be someone for whom even the obvious - “all NON-Thinkpad series” - is still not clear enough ;-).

Oh, ThinkBook series is not IdeaPad series, so we can't know that `deapad_laptop` driver supports ThinkBook series from driver's name. Besides, series IdeaPad is much more earlier than series ThinkBook, taking this into consideration, we are not sure whether `all NON-Thinkpad series` include ThinkBook series.

TurnOffNOD, 2024-08-30:
> @TurnOffNOD done.

Thanks for your work.