linrunner/TLP #598: Changes in /etc/tlp.conf don't affect tlp-stat -b on a Lenovo W530 with Xubuntu 20.04.3

melolontha-melolontha, 2021-11-18:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

I use a Lenovo Thinkpad W530 with Xubuntu Ubuntu 20.04.3 LTS currently using a kernel 5.4.0-90-generic x86_64.

Charging thresholds 75% / 90% have worked on this laptop before. Now I checked it again with `sudo tlp-stat -b` and I was surprised that they are reported as 0% / 100%.

Therefore I checked `/etc/tlp.conf` and I found out that the lines for the charge threshold were commented out (i.e. they started with a `#`). Also the values there were 75% / 80% rather than 75% / 90% which I had set before and of course it was different from what is effectively reported.

Therefore I removed the `#` in front of the two lines for `BAT0` first, changed the upper threshold to 90% and then I executed `sudo tlp start` and checked `sudo tlp-stat -b` again. But the charge thresholds did not change! 

I also tried a reboot, but I see no difference. It looks like tlp does not really obey to what is in `/etc/tlp.conf`.

To be on the safe side, I also removed the `#` sign in front of the two lines for `BAT1` and adjusted the thresholds for `BAT1` in the same way, even though I don't have an Ultrabay battery. Of course I executed `sudo tlp start` again and checked the result of `sudo tlp-stat -b`. But the reported charge thresholds did not change, even after a reboot (with the running AC adapter connected and also running on battery only)!

**Expected behavior**

I expect the changes of `/etc/tlp.conf` to be reflected in `sudo tlp-stat -b` after `sudo tlp start` has been executed.

**To Reproduce**

The unexpected behavior is reproducible on my laptop. It does not  depend on running on battery or AC adapter

See attached file with listing of `/etc/tlp.conf` and `tlp-bat`
[2021-11-18_Files-for-tlp-issue.txt](https://github.com/linrunner/TLP/files/7561673/2021-11-18_Files-for-tlp-issue.txt)
 commands

**Additional context**

Add any other context about the problem here.

linrunner, 2021-11-18:
Hi, 

parameter lines with comments are be ignored:

     START_CHARGE_THRESH_BAT0=75          # war am 2021-11-17 auskommentiert
     STOP_CHARGE_THRESH_BAT0=90           # dito, von 80 auf 90 erhöht.

Refer to [Settings](https://linrunner.de/tlp/settings/introduction.html#parameter-syntax)

> Comments after parameters are not allowed, the whole line will be silently ignored:
>    EXAMPLE="dont use like this" # Parameter in front is ignored!

closed 2021-11-18 by linrunner

melolontha-melolontha, 2021-11-19:
linrunner, 
thank you very much for your advice. 
Indeed, now it works as expected. 
It was my fault not to read this part of settings.
However, one might consider to stop parsing a line from the config file with a # sign (including it) rather than discard it altogether.
Then one might annotate changes in the config file in the same manner as in bash scripts, something very well known to everyone.

linrunner, 2022-06-10:
Good news: I modified the parser to handle trailing comments.

linrunner, 2023-07-22:
Hi @ALL : TLP 1.6 Beta 1 is out and contains a fix for this issue -> https://github.com/linrunner/TLP/issues/700

melolontha-melolontha, 2023-07-24:
How can I verify the version which I have? I uninstalled the previous one and installed (hopefully!) the beta one, but trying comments in active lines of /etc/tlp.conf did not work. When I set them to separate lines instead, the lines were honoured.

melolontha-melolontha, 2023-07-24:
If `sudo tlp-stat` does the test: I geht

```
[sudo] Passwort für v: 
...
--- TLP 1.3.1 --------------------------------------------
```

What should I get after

```
sudo apt-get remove tlp
sudo add-apt-repository --remove ppa:linrunner/tlp
sudo add-apt-repository ppa:linrunner/tlp-beta
sudo apt install tlp tlp-rdw
```

This is, what I did.

linrunner, 2023-07-24:
Following the instructions _precisely_ is important. The [download page](https://download.linrunner.de/packages/) does *not* tell to uninstall anything or to remove the regular PPA. But that's not your real problem. 

You have overlooked the

    sudo apt update

after adding the Beta PPA. After you have made up for that, all you need to do is a

    sudo apt install tlp tlp-rdw

Good luck.