linrunner/TLP #599: Error: battery discharge/recalibrate not available

jJit0, 2021-11-18:
Hi,

I can't `tlp discharge` since version 1.4.0

I had a lot of kernel alerts when I do `dmesg -l alerts`

`BUG: kernel NULL pointer dereference, address: 0000000000000000`
`#PF: supervisor instruction fetch in kernel mode`
`#PF: error_code(0x0010) - not-present page`
`Oops: 0010 [#1] SMP NOPTI`
`CPU: 2 PID: 1473 Comm: tpacpi-bat Tainted: G S         OE     5.14.0-kali2-amd64 #1  Debian 5.14.9-2kali1`
`Hardware name: LENOVO 20HQS17409/20HQS17409, BIOS N1MET67W (1.52 ) 10/05/2021`
`RIP: 0010:0x0`
`Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.`
`RSP: 0018:ffffba9ac0843ef0 EFLAGS: 00010246`
`RAX: 0000000000000000 RBX: ffff931e42ca6cc0 RCX: 0000000000000001`
`RDX: 0000000000000001 RSI: 0000000000000000 RDI: ffff931e47ea4500`
`RBP: 0000000000000000 R08: 0000000000000003 R09: 0000000000000000`
`R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000001`
`R13: ffffffffffffffea R14: ffff931e47ea4500 R15: 0000000000000000`
`FS:  00007f000ae242c0(0000) GS:ffff9321d1700000(0000) knlGS:0000000000000000`
`CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033`
`CR2: ffffffffffffffd6 CR3: 000000010590a004 CR4: 00000000003706e0`
`DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000`
`DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400`
`Call Trace:`
` proc_reg_llseek+0x42/0x80`
` ksys_lseek+0x7a/0xb0`
` do_syscall_64+0x38/0xc0`
` entry_SYSCALL_64_after_hwframe+0x44/0xae`

And that's the `tlp-stat --battery` output
`+++ Battery Care`
`Plugin: thinkpad`
`Supported features: charge thresholds`
`Driver usage:`
`* natacpi (thinkpad_acpi) = active (charge thresholds)`
`* tpacpi-bat (acpi_call)  = inactive (kernel error)`
`Parameter value ranges:`
`* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99`
`* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)`

Any ideas ?

linrunner, 2021-11-18:
TLP 1.4 is by no means the cause. tlp-stat 1.4 even shows you the real cause:

> tpacpi-bat (acpi_call) = inactive **(kernel error)**

You need a newer version of _acpi_call_ -> https://github.com/nix-community/acpi_call/issues/15

closed 2021-11-18 by linrunner

jJit0, 2021-11-19:
Thanks @linrunner

The new version is working GREAT!!!😄