linrunner/TLP #601: batterycare/thinkpad: Use feature-detection instead of Model whitelist

t-8ch, 2021-11-24:
**Is your feature request related to a problem? Please describe.**

I have a T400 running with Coreboot. Coreboot does not support SMAPI but instead exposes the battery controls via ACPI methods that are handled by thinkpad_acpi.
Currently `bat.d/05-thinkpad` has a list of devices that are hardlocked to SMAPI instead of ACPI which includes the T400.
This leads to battery controls being unavailable in TLP.

**Describe the solution you'd like**

It would be nice if TLP could first check the availability of the actual sysfs files for natacpi to detect its availability and only afterwards consult the hardcoded list.

linrunner, 2021-12-06:
Oh well, yet another special case ...

Please test and report

    tlp-stat -b

t-8ch, 2021-12-06:
Thanks, will do! Unfortunately I only will have access to that device sometime next week, so it be a while.

t-8ch, 2021-12-12:
I can confirm this works, thanks a lot!

closed 2021-12-12 by t-8ch

linrunner, 2021-12-12:
Great. Thanks for testing.

linrunner, 2021-12-21:
TLP 1.5 Beta 1 is out -> https://github.com/linrunner/TLP/issues/609

linrunner, 2022-01-07:
Beta test is completed and TLP 1.5.0 is [out](https://linrunner.de/tlp/news.html). Have fun!