linrunner/TLP #602: [Asus laptop] Battery stop charge threshold isn't set at boot.

GWFrank, 2021-12-06:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**System info**

```
OS: Manjaro Linux x86_64 
Host: ZenBook UX425EA_UX425EA 1.0 
Kernel: 5.15.2-2-MANJARO 
DE: Plasma 5.23.3 
CPU: 11th Gen Intel i5-1135G7 (8) @ 4.200GHz 
GPU: Intel TigerLake-LP GT2 [Iris Xe Graphics] 
Memory: 2379MiB / 15698MiB
```

**Describe the bug**

Battery stop charge threshold isn't set at boot. But manually running `tlp start`  or restarting/reloading `tlp.service` correctly sets the threshold.

**Expected behavior**

After boot the stop charge threshold is what I set (80).

**To Reproduce**

0. Occur on both battery and AC.
1. Edit these two entries in `/etc/tlp.conf`
```
START_CHARGE_THRESH_BAT0=0
STOP_CHARGE_THRESH_BAT0=80
```
2. Use `tlp setcharge` to verify settings.
```
$ sudo tlp setcharge    
Setting temporary charge threshold for BAT0:
  stop =  80
```
3. Check `tlp-stat -b` output
```
$ sudo tlp-stat -b | grep charge_control
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     80 [%]
```
4. Reboot
5. Check `tlp-stat -b` output
```
$ sudo tlp-stat -b | grep charge_control
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
```
6. Reload tlp
```
$ sudo systemctl reload tlp
```
7. Check `tlp-stat -b` output
```
$ sudo tlp-stat -b | grep charge_control
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     80 [%]
```

[tlp-stat output](https://gist.github.com/GWFrank/b9148385b4645d981c7a83c91789a77c)

**Additional context**

I had been using [this tool](https://github.com/tshakalekholoane/bat) to set charge threshold, and I had removed changes done by it before configuring tlp's battery care settings.

linrunner, 2021-12-07:
Hi,

I need trace outputs from the boot process. Please add the line

    TLP_DEBUG="bat ps run"

to your configuration file, reboot and post the output of

    sudo tlp-stat -T

GWFrank, 2021-12-07:
Here is the output
```
--- TLP 1.4.0 --------------------------------------------

-- Journal begins at Wed 2021-10-20 22:29:38 CST, ends at Wed 2021-12-08 01:23:40 CST. --
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: +++ init start (1.4.0) ++++++++++++++++++++++++++++++++++++++++
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: get_sys_power_supply(BAT0).bat_discharging: syspwr=1; wait=
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: power_source=bat
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: manual_mode=none
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: power_mode=bat
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: compare_and_save_power_state(1).different: old=
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: Applying power save settings...done.
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: check_thinkpad.not_a_thinkpad: model=1.0
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: batdrv_init.thinkpad.not_a_thinkpad
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: check_thinkpad.not_a_thinkpad: model=1.0
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: batdrv_init.thinkpad-legacy.not_a_thinkpad
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: batdrv_init.asus.no_match
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: batdrv_init.huawei.no_match
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: batdrv_init.lenovo.no_match
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: batdrv_init.samsung.no_match
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: batdrv_init.lg.no_match
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: batdrv_init.generic: batteries=BAT0
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: set_charge_thresholds.no_method
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: Setting battery charge thresholds...done.
12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpJ2Qn99 -> /run/tlp/run.conf
12月 08 01:23:06 Frank-UX425EA-Linux tlp[1928]: +++ auto (1.4.0) ++++++++++++++++++++++++++++++++++++++++
12月 08 01:23:06 Frank-UX425EA-Linux tlp[1928]: get_sys_power_supply(BAT0).bat_discharging: syspwr=1; wait=
12月 08 01:23:06 Frank-UX425EA-Linux tlp[1928]: power_source=bat
12月 08 01:23:06 Frank-UX425EA-Linux tlp[1928]: manual_mode=none
12月 08 01:23:06 Frank-UX425EA-Linux tlp[1928]: power_mode=bat
12月 08 01:23:06 Frank-UX425EA-Linux tlp[1928]: compare_and_save_power_state(1).equal
12月 08 01:23:06 Frank-UX425EA-Linux tlp[1960]: +++ auto (1.4.0) ++++++++++++++++++++++++++++++++++++++++
12月 08 01:23:06 Frank-UX425EA-Linux tlp[1960]: get_sys_power_supply(BAT0).bat_discharging: syspwr=1; wait=
12月 08 01:23:06 Frank-UX425EA-Linux tlp[1960]: power_source=bat
12月 08 01:23:06 Frank-UX425EA-Linux tlp[1960]: manual_mode=none
12月 08 01:23:06 Frank-UX425EA-Linux tlp[1960]: power_mode=bat
12月 08 01:23:06 Frank-UX425EA-Linux tlp[1960]: compare_and_save_power_state(1).equal
```

linrunner, 2021-12-07:
> 12月 08 01:22:34 Frank-UX425EA-Linux tlp[484]: batdrv_init.asus.no_match

Looks like the kernel module `asus_wmi` is not loaded or not initialized yet when `tlp.service` starts, which in fact happens quite late in the boot process (`After=multi-user.target`).

I'd try to add the line

    asus_wmi

into **/etc/modules** and also make shure the module is loaded by the Initramfs.

GWFrank, 2021-12-08:
Adding `asus_wmi` into the initramfs's modules does fix the problem.

**Steps**
1. Edit this part of `/etc/mkinitcpio.conf`:
```
MODULES=(<some other modules> asus_wmi)
```
2. Build initramfs:
```
$ sudo mkinitcpio -P
```
3. Reboot

closed 2021-12-08 by GWFrank

linrunner, 2021-12-08:
Thanks for sharing your solution.

johncf, 2024-06-09:
I have the same problem, with my Debian 12 installation in an Asus laptop. I tried adding `asus_wmi` to `/etc/modules`, and this seems to confirm that `tlp.service` starts after `asus_wmi` is loaded:

```
% sudo journalctl -o short-monotonic -b -0 | rg -i 'asus_wmi|TLP'
[    4.675482] impossideb systemd-modules-load[429]: Inserted module 'asus_wmi'
[    4.675746] impossideb kernel: asus_wmi: ASUS WMI generic driver loaded
[    4.864396] impossideb kernel: asus_wmi: Initialization: 0x1
[    4.865371] impossideb kernel: asus_wmi: SFUN value: 0x21
[    8.430513] impossideb systemd[1]: Starting tlp.service - TLP system startup/shutdown...
[    8.542447] impossideb tlp[2024]: Applying power save settings...done.
[    8.552630] impossideb tlp[2024]: Setting battery charge thresholds...done.
[    8.558849] impossideb systemd[1]: Finished tlp.service - TLP system startup/shutdown.
```

I have set the configuration correctly:
```
% tlp-stat -c                                                                                                        ! 
--- TLP 1.5.0 --------------------------------------------

+++ Configured Settings:
defaults.conf L0004: TLP_ENABLE="1"
defaults.conf L0005: TLP_WARN_LEVEL="3"
defaults.conf L0006: TLP_PERSISTENT_DEFAULT="0"
[...]
defaults.conf L0051: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0052: RESTORE_THRESHOLDS_ON_BAT="0"
defaults.conf L0053: NATACPI_ENABLE="1"
defaults.conf L0054: TPACPI_ENABLE="1"
defaults.conf L0055: TPSMAPI_ENABLE="1"
/etc/tlp.d/01-zenbook.conf L0006: CPU_BOOST_ON_AC="1"
/etc/tlp.d/01-zenbook.conf L0007: CPU_BOOST_ON_BAT="0"
/etc/tlp.d/01-zenbook.conf L0028: START_CHARGE_THRESH_BATT="0"
/etc/tlp.d/01-zenbook.conf L0029: STOP_CHARGE_THRESH_BATT="60"
```

However, after startup, the charge threshold is still not set correctly:

```
% sudo tlp-stat -b                                               
--- TLP 1.5.0 --------------------------------------------

+++ Battery Care
Plugin: asus
Supported features: charge threshold
Driver usage:
* natacpi (asus_wmi) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0/1: 0(off)..100(default)

+++ Battery Status: BATT
/sys/class/power_supply/BATT/manufacturer                   = ASUSTeK
/sys/class/power_supply/BATT/model_name                     = UM5302
/sys/class/power_supply/BATT/cycle_count                    =    123
/sys/class/power_supply/BATT/charge_full_design             =   4220 [mAh]
/sys/class/power_supply/BATT/charge_full                    =   3527 [mAh]
/sys/class/power_supply/BATT/charge_now                     =   1763 [mAh]
/sys/class/power_supply/BATT/current_now                    =   3085 [mA]
/sys/class/power_supply/BATT/status                         = Charging

/sys/class/power_supply/BATT/charge_control_end_threshold   =    100 [%]

Charge                                                      =   50.0 [%]
Capacity                                                    =   83.6 [%]
```

I also tried adding `asus_wmi` to `/etc/initramfs-tools/modules` and running `sudo update-initramfs -c -k all`. But that made no difference either.

linrunner, 2024-06-10:
> /etc/tlp.d/01-zenbook.conf L0028: START_CHARGE_THRESH_BAT**T**="0"
> /etc/tlp.d/01-zenbook.conf L0029: STOP_CHARGE_THRESH_BAT**T**="60"

@johncf the `*_BATT` parameters you use do not exist. Use
```
START_CHARGE_THRESH_BAT0="0"
STOP_CHARGE_THRESH_BAT0="60"
```
instead. The [documentation](https://linrunner.de/tlp/settings/bc-vendors.html#asus) states the following:

>  Batteries BAT0, BATC and **BATT** share the STOP_CHARGE_THRESH_**BAT0** parameter
> Battery BAT1 uses the STOP_CHARGE_THRESH_BAT1 parameter

abhijeeth-babu, 2025-01-15:
> has anyone applied such change in fedora? `/lib/dracut/modules.d/` seems too hard for me to configure.

Do not mess with /lib/
[relevant arch wiki]( https://wiki.archlinux.org/title/Dracut#Early_kernel_module_loading)

Create a conf file in `/etc/dracut.conf.d/`, name it myflags.conf or anything else.
Add the following to it `force_drivers+=" asus_wmi "` (the spaces before and after module name are important)
And then run `dracut -f` (with sudo ofc)

BentoTPG, 2025-04-26:
I got the same problem with mint too
I set charge `sudo tlp setcharge 70 80 BAT1` and check in `sudo tlp-stat -b` it said
```
--- TLP 1.8.0 --------------------------------------------

+++ Battery Care
Plugin: asus
Supported features: charge threshold
Driver usage:
* natacpi (asus_wmi) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0/1: 1..100(default)

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = ASUS 
/sys/class/power_supply/BAT1/model_name                     = A32-K55
/sys/class/power_supply/BAT1/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT1/charge_full_design             =   4110 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   2598 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   2240 [mAh]
/sys/class/power_supply/BAT1/current_now                    =   1906 [mA]
/sys/class/power_supply/BAT1/status                         = Discharging

/sys/class/power_supply/BAT1/charge_control_end_threshold   =    80 [%]

Charge                                                      =   86.2 [%]
Capacity                                                    =   63.2 [%]

```
after i reboot i use `sudo tlp-stat -b`
```
--- TLP 1.8.0 --------------------------------------------

+++ Battery Care
Plugin: asus
Supported features: charge threshold
Driver usage:
* natacpi (asus_wmi) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0/1: 1..100(default)

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = ASUS 
/sys/class/power_supply/BAT1/model_name                     = A32-K55
/sys/class/power_supply/BAT1/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT1/charge_full_design             =   4110 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   2598 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   2240 [mAh]
/sys/class/power_supply/BAT1/current_now                    =   1984 [mA]
/sys/class/power_supply/BAT1/status                         = Discharging

/sys/class/power_supply/BAT1/charge_control_end_threshold   =    100 [%]

Charge                                                      =   86.2 [%]
Capacity                                                    =   63.2 [%]

```
I use `sudo tlp-stat -T`
```
--- TLP 1.8.0 --------------------------------------------

Apr 26 09:30:16 FX506LH tlp[1564]: Applying power save settings...done.
Apr 26 09:30:16 FX506LH tlp[1564]: Setting battery charge thresholds...done.
```
I don't know how to solve it in mint.I'm just using it for a few days