linrunner/TLP #603: [Thinkpad T580] Thresholds configuration is not respected, battery constantly kept full

mladenplavsic, 2021-12-07:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

On Thinkpad T580 battery charge thresholds are ignored. This seems to be started with update to 1.4 which happened recently. Both batteries are kept full, although in `tlp.conf` values are set to 60/70. This doesn't change with reboot, AC reconnect, nor after running `tlp start`. Output of `tlp-stat` says:
```
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds)
* tpacpi-bat (acpi_call)  = active (recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)
```
but the values for thresholds (also present in the output of `tlp-stat`) say:
```
/sys/class/power_supply/BAT0/charge_control_start_threshold =     60 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     70 [%]
...
/sys/class/power_supply/BAT1/charge_control_start_threshold =     60 [%]
/sys/class/power_supply/BAT1/charge_control_end_threshold   =     70 [%]
```

**Expected behavior**

When battery thresholds are set in configuration, it should be respected and battery level kept between start and stop values.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? 
AC
2. Actions to reproduce the behaviour
Updating configuration files and running `tlp start` doesn't change the applied thresholds
3. Shell commands entered and their output
Not applicable
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**
https://gist.github.com/mladenplavsic/ab343a5b4bb1973558c01b828aa79268

**Additional context**

I was using `tlp` for a long time, working without any problem. Something was updated recently and I noticed that battery is at 100% all the time. After trying to resolve the problem I found out that there was a new version `1.4` released, and so I suspect this is related to it.

linrunner, 2021-12-07:
Hi, 

your tlp-stat output proves that TLP has successfully written the thresholds to the embedded controller (EC):

> /sys/class/power_supply/BAT0/charge_control_start_threshold =     60 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     70 [%]
...
/sys/class/power_supply/BAT1/charge_control_start_threshold =     60 [%]
/sys/class/power_supply/BAT1/charge_control_end_threshold   =     70 [%]

So the problem is not with TLP in general or with 1.4, but with the EC.

I would first try to reset the EC via the emergency reset hole on your T580.

Second i'd try an update of the UEFI and EC firmware via LVFS (fwupdmgr). Current UEFI version is 1.30, your 1.16 is way outdated.

You should also have a look at the TLP FAQ, which contains workarounds for similar problems, e.g.:

- https://linrunner.de/tlp/faq/battery.html#disabling-the-charge-thresholds-does-not-work
- https://linrunner.de/tlp/faq/battery.html#start-threshold-does-not-apply-after-change

linrunner, 2021-12-21:
I'll close here. Nevertheless, you are welcome to post new findings about your problem here.

closed 2021-12-21 by linrunner