linrunner/TLP #604: Overriding settings works only for the current power source

BrendonIrwan, 2021-12-07:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs via ~~[Gist](https://gist.github.com/) (see below)~~ an attachment.

**Describe the bug**

When overriding both the following settings via the command line:
- `CPU_MAX_PERF_ON_AC`
- `CPU_MAX_PERF_ON_BAT`
Only one of `_AC` or `_BAT` is applied if they are in use; if I override while on battery, then only `_BAT` applies; if I override on AC, then only `_AC` applies. Otherwise, TLP uses those from my config file.

**Expected behavior**

Both `_AC` or `_BAT` should be applied.

**To Reproduce**

Steps to reproduce the unexpected behavior:

0. In your config file:
```
CPU_MAX_PERF_ON_AC=100
CPU_MAX_PERF_ON_BAT=0
```
1. `# tlp start -- CPU_MAX_PERF_ON_AC=50 CPU_MAX_PERF_ON_BAT=25` or ones to your liking.
2. `# tlp-stat -p`
3. Change power source.
4. `# tlp-stat -p`
5. Oops.
[tlp-stat.txt](https://github.com/linrunner/TLP/files/7671472/tlp-stat.txt)

**Additional context**

`tlp-stat` does not print overridden settings.

linrunner, 2021-12-08:
Provide the outputs via [Gist](https://gist.github.com/) please, no attachments. Thanks.

BrendonIrwan, 2021-12-08:
https://gist.github.com/BrendonIrwan/a1cf37fb00fde8e868380955a32c3628

BrendonIrwan, 2021-12-08:
Update: The overridden settings seems to be discarded also after a suspend and wake-up.

linrunner, 2021-12-08:
Hi,

the reason for the behavior is that the intel_pstate kernel driver does not accept the value 0 for `max_perf_pct`. The lower limit seems to be 11 (guessed from `min_perf_pct =  11 [%]`).

linrunner, 2021-12-08:
ps. What you call "overriding settings" is just a test tool (for my personal pleasure). The manpage states _"for this program invocation only"_, which means that they are valid only for that single invocation of `tlp <command>` and do not persist thereafter.

BrendonIrwan, 2021-12-08:
So changing the power source, suspend and wake-up, etc. counts as a change in invocation?

BrendonIrwan, 2021-12-08:
> Hi,
> 
> the reason for the behavior is that the _intel_pstate_ kernel driver does not accept the value 0 for `max_perf_pct`. The lower limit seems to be 11 in your case (guessed from `min_perf_pct = 11 [%]`).

Oh, this behavior is fine; I just want the lowest possible value. I use the same config file on others also.

linrunner, 2021-12-08:
> So changing the power source, suspend and wake-up, etc. counts as a change in invocation?

All these events invoke `tlp` with specific parameters, each time the configuration is read again. There is no daemon keeping a persistent state of settings. See:

- https://linrunner.de/tlp/introduction.html#event-driven-architecture
- https://linrunner.de/tlp/developers/architecture.html

Bottom line: you may not override parameters for these background events, only when invoking `tlp` in a shell.

> Oh, this behavior is fine; I just want the lowest possible value. I use the same config file on others also.

That won't work for your Latitude 5480 because the driver ignores `CPU_MAX_PERF_ON_BAT=0`. I suggest you use laptop specific snippets in **/etc/tlp.d/**.

BrendonIrwan, 2021-12-08:
Got them. Cheers.

linrunner, 2021-12-08:
If this clarifies your points, please close.

closed 2021-12-09 by BrendonIrwan