linrunner/TLP #605: WOL_DISABLE=N still disables WOL after period of time

Natetronn, 2021-12-08:
Hi, 

Is there anything else I need to set in `/etc/tlp.conf` for WOL to stay enabled, even after a certain amount of time? I've made sure the service is enabled, running and restarted, as well as WOL_DISABLE=N is set etc.

```
TLP_ENABLE=1
WOL_DISABLE=N
```
I ask because every time I turn on Magic Packet (g) it works but, only for a short time. I leave my PC for a spell and when I come back it's been set to default (d) i.e. disabled again again. I have to `ethtool -s eth0 wol g` or use Advanced Network Configuration to turn on Magic Packet again, which kind of defeats the purpose of WOL. Then I get the following:

```
Supports Wake-on: pumbg
Wake-on: g
```

If I uninstall TLP I don't have this issue, so I'm fairly certain it's TLP related (my Bios is setup correctly etc.) Much google searching points to TLP and the vars above to turn off TLP for WOL but, for me that's not working for some reason.

I'm on Manjaro (Arch) and it's a desktop environment. 

Anyway, I thought I'd ask before I resort to uninstalling TLP.

Natetronn, 2021-12-08:
```
❯ tlp-stat -c
--- TLP 1.4.0 --------------------------------------------

+++ Configured Settings:
/etc/tlp.conf L0029: TLP_ENABLE="1"
defaults.conf L0005: TLP_WARN_LEVEL="3"
defaults.conf L0006: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0008: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0010: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0012: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0013: SCHED_POWERSAVE_ON_AC="0"
defaults.conf L0014: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0015: NMI_WATCHDOG="0"
defaults.conf L0016: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0017: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0018: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0019: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0020: DISK_IOSCHED="keep keep"
defaults.conf L0021: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0022: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
defaults.conf L0023: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0025: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0026: PCIE_ASPM_ON_AC="default"
defaults.conf L0027: PCIE_ASPM_ON_BAT="default"
defaults.conf L0028: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0030: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0032: WIFI_PWR_ON_AC="off"
defaults.conf L0033: WIFI_PWR_ON_BAT="on"
/etc/tlp.conf L0321: WOL_DISABLE="N"
defaults.conf L0035: SOUND_POWER_SAVE_ON_AC="1"
defaults.conf L0036: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0037: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0038: BAY_POWEROFF_ON_AC="0"
defaults.conf L0039: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0040: BAY_DEVICE="sr0"
defaults.conf L0041: RUNTIME_PM_ON_AC="on"
defaults.conf L0042: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0043: RUNTIME_PM_DRIVER_DENYLIST="mei_me nouveau radeon"
defaults.conf L0044: USB_AUTOSUSPEND="1"
defaults.conf L0045: USB_EXCLUDE_AUDIO="1"
defaults.conf L0046: USB_EXCLUDE_BTUSB="0"
defaults.conf L0047: USB_EXCLUDE_PHONE="0"
defaults.conf L0048: USB_EXCLUDE_PRINTER="1"
defaults.conf L0049: USB_EXCLUDE_WWAN="0"
defaults.conf L0050: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"
defaults.conf L0051: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0052: RESTORE_THRESHOLDS_ON_BAT="0"
defaults.conf L0053: NATACPI_ENABLE="1"
defaults.conf L0054: TPACPI_ENABLE="1"
defaults.conf L0055: TPSMAPI_ENABLE="1"
```

Natetronn, 2021-12-08:
```
❯ sudo systemctl status tlp
[sudo] password for username: 
● tlp.service - TLP system startup/shutdown
     Loaded: loaded (/usr/lib/systemd/system/tlp.service; enabled; vendor preset: disabled)
     Active: active (exited) since Tue 2021-12-07 21:10:45 PST; 48min left
       Docs: https://linrunner.de/tlp
    Process: 1290 ExecStart=/usr/bin/tlp init start (code=exited, status=0/SUCCESS)
   Main PID: 1290 (code=exited, status=0/SUCCESS)
        CPU: 162ms

Dec 07 21:10:45 username systemd[1]: Starting TLP system startup/shutdown...
Dec 07 21:10:45 username tlp[1290]: Applying power save settings...done.
Dec 07 21:10:45 username tlp[1290]: Setting battery charge thresholds...done.
Dec 07 21:10:45 username systemd[1]: Finished TLP system startup/shutdown.
```

Natetronn, 2021-12-08:
I hadn't read the support page:

https://linrunner.de/tlp/support/index.html

If this isn't a potential bug and you'd like me to seek help elsewhere, I can close and ask on another forum. Just let me know.

Thanks!

linrunner, 2021-12-08:
Hi,

unfortunately the [setting's doc](https://linrunner.de/tlp/settings/network.html#wol-disable) is misleading here. In fact `WOL_DISABLE=N` means that TLP doesn't touch the state of WOL at all. 

I suspect a problem in the kernel driver of the LAN adapter or some other TLP power saving setting that makes the driver activate WOL. Check the [troubleshooting guide](https://linrunner.de/tlp/support/troubleshooting.html).

Natetronn, 2021-12-08:
Thanks for the reply!

My LAN adapter:
```
❯ lspci 
05:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)
```

So I started playing around with various PCIe settings to see if any of those are switching the LAN adapter from Magic Packet to Default (disabled.)

```
+++ PCIe Runtime Power Management
Enable devices    = (disabled)
Disable devices   = 05:00.0
Device denylist   = 05:00.0
Driver denylist   = mei_me nouveau radeon r8169
```

The device does show up in the tlp-stat with the correct driver and device address

`/sys/bus/pci/devices/0000:05:00.0/power/control = on   (0x020000, Ethernet controller, r8169)`

Anyway, it's late so, I'll play with the PCIe stuff a bit more tomorrow.

Thanks again...

linrunner, 2021-12-21:
Any news on this?

closed 2022-01-04 by linrunner