linrunner/TLP #606: Can't resume from suspend if a specific disk is added to DISK_DEVICES with AHCI_RUNTIME_PM_ON_BAT enabled

luckv, 2021-12-09:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/)

**Describe the bug**
If the parameter `DISK_DEVICES` is specifically set with one of my disks (sdb) and `AHCI_RUNTIME_PM_ON_BAT` is set to `auto`, after suspend (`systemctl suspend`), the pc turn on but not completely. The screen turn on but remains blank (lit but black), and the wifi doesn't connect (I can't ping the pc after suspension).

I don't know much about the disk. Was shipped with the laptop and report this name: Micron 1100 MTFD

I excluded all other possible causes:
 - The issue doesn't happen without tlp, or with tlp disabled
 - The issue doesn't happen with tlp default configuration
 - The issue doesn't happen if that disk is excluded from `DISK_DEVICES` or `AHCI_RUNTIME_PM_ON_BAT` is set to `on`
 - The issue doesn't happen with all other settings enabled that you see in my `tlp-stat`. I enabled them one by one and rebooted to check where the problem was.
 - Gone through the [troubleshooting guide](https://linrunner.de/tlp/support/troubleshooting.html), disabled both `RUNTIME_PM_ON_BAT` and `USB_AUTOSUSPEND`, but doesn't work
 - Searched in the web for similar issues
 - Searched in this repository for similar issues

**Expected behavior**
The pc to return from suspension normally

**To Reproduce**
Steps to reproduce the unexpected behavior:
   - Set sdb in the list of devices

1. The problem occurs only in **battery mode**
4. **Full output of `tlp-stat`: [link](https://gist.github.com/luckv/909fcbe028d1cbf4d0d3be48275b575e). All customizations are in `luckv.conf`

**Additional context**
OS: Fedora 35 (Workstation Edition)
Kernel version (`uname -r`): 5.15.6-200.fc35.x86_64

I have also Pop!_OS version 21.04 (same kernel version, 5.15) installed on the same pc (also the same boot disk) but with TLP 1.3.1, the issue doesn't present with it.

linrunner, 2021-12-09:
This is probably related to https://github.com/linrunner/TLP/issues/587#issuecomment-947376033

The AHCI autosuspend sysfiles associated sdb are:

    /sys/block/sdb/power/control = on, autosuspend_delay_ms = -1
    /sys/bus/pci/devices/0000:00:17.0/ata2/power/control     = auto  -- sdb

The disk seems to refuse to accept `auto` to **/sys/block/sdb/power/control**. 

Does the problem disappear after writing *on* to **/sys/bus/pci/devices/0000:00:17.0/ata2/power/control**?

luckv, 2021-12-09:
@linrunner are you suggesting that should I try to write directly in **/sys/bus/pci/devices/0000:00:17.0/ata2/power/control**`?

linrunner, 2021-12-09:
Almost. You need a root shell:

    echo > /sys/bus/pci/devices/0000:00:17.0/ata2/power/control

luckv, 2021-12-09:
The situation has changed. Before I tried what you suggest I commented `AHCI_RUNTIME_PM_ON_BAT` in my configuration so its value was **auto**. 
Without doing a reboot I done `sudo tlp start`, then I checked tlp-stat and what I saw was:
```
/sys/block/sdb/power/control = auto, autosuspend_delay_ms = 15000
/sys/bus/pci/devices/0000:00:17.0/ata2/power/control     = auto  -- sdb
```
but the problem persists, also after reboot. After reboot I took tlp-stat. [See here](https://gist.github.com/luckv/2713b528bef90c01a648f031be426ac5)

linrunner, 2021-12-10:
You tested the opposite of what I need to isolate the cause. So again properly please:

    echo on > /sys/bus/pci/devices/0000:00:17.0/ata2/power/control
    echo on > /sys/block/sdb/power/control

Then suspend and check if the problem occurs again.

luckv, 2021-12-12:
Sorry @linrunner for answering only now. 
I do what you suggested and it worked! My laptop resume from suspension correctly. 
In tlp-stat the output remained the same as my last message (see [here](https://gist.github.com/luckv/ac1f0f1ec505a2906205d4d56234ca0e)).
But if I try to read those values with cat I obtain those results:
```
[root@fedora luckv]# cat /sys/block/sdb/power/control
on
[root@fedora luckv]# cat /sys/bus/pci/devices/0000:00:17.0/ata2/power/control
auto
```

Hope this can help you to resolve the problem

linrunner, 2021-12-12:
> In tlp-stat the output remained the same as my last message (see [here](https://gist.github.com/luckv/ac1f0f1ec505a2906205d4d56234ca0e)).
> But if I try to read those values with cat I obtain those results:

tlp-stat always displays the same values as cat, it even uses cat internally.

The deviation in your test case is due to the fact that tlp is called again after the resume and restores the state before your manual write operations.  I guess the next suspend will produce the problem again.

What I don't understand is why  **/sys/block/sdb/power/control** stays `on` after boot and resume but is `auto` after `tlp start`.

Let's test this a little more thoroughly:

1. Enable trace mode by adding the following line to your config file
```
TLP_DEBUG="disk pm ps run"
```

2. Reboot and show the output of
```
sudo tlp-stat -d
```

3. Make your manual changes (in a root shell)
```
echo on > /sys/bus/pci/devices/0000:00:17.0/ata2/power/control
echo on > /sys/block/sdb/power/control
```
and show the output of
```
sudo tlp-stat -d
```
again.

4. Suspend and resume your laptop. Show the output of
```
sudo tlp-stat -d
```
once again. Finallly the trace output for the whole business
```
sudo tlp-stat -T
```

luckv, 2021-12-12:
> The deviation in your test case is due to the fact that tlp is called again after the resume and restores the state before your manual write operations. I guess the next suspend will produce the problem again.

You are right, only the first time the pc suspend and resume correctly

luckv, 2021-12-12:
@linrunner I do what as you written.
I created  a Gist with all the output you requested [here](https://gist.github.com/luckv/c16f813ff2bce046ccda2b49ea5d3e1b). Pretty every number at the start of the file name is the number in the numbered list of the passages. I also included a `tlp-stat -c` taken at the end to recheck my configuration.

luckv, 2021-12-13:
> Is this still the same kernel version?

Absolutely yes. I show you the output of my `uname --all`:
```
Linux xxx-fedora 5.15.6-200.fc35.x86_64 #1 SMP Wed Dec 1 13:41:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

> In your start post the initial state is
>>/sys/block/sdb/power/control = on, autosuspend_delay_ms = -1
>>/sys/bus/pci/devices/0000:00:17.0/ata2/power/control = auto -- sdb

I wasn't able to obtain that output ever again. I don't know why onestly.

> Your outputs show strange deviations.

I know. In fact one time I don't try what you asked because I was confused of the results of my actions.
One very very strange thing it's that, even if tlp uses cat internally as you said, `tlp-stat -d` and `cat cat /sys/block/sdb/power/control` show 2 different results, on said that `/sys/block/sdb/power/control` is `on`, the second say that is `auto`... even if executed one after the other.

luckv, 2021-12-13:
While in battery mode I restored `AHCI_RUNTIME_PM_ON_BAT=on`, shutdown the pc, then booted on.
I opened a terminal and launched only this commands. The outputs for /sys/block/... differs from tlp-stat

Gist [here](https://gist.github.com/luckv/effcf6ab884ffaf7de3fb83bf1fcd28c)

linrunner, 2021-12-14:
Oops. In fact there is an error in the output. TLP internally reads and writes the correct path /sys/block/DISK/device/power/control, but `tlp-stat -d` displays it incorrectly as /sys/block/DISK/power/control.

I can't provide a Fedora package right away, so please edit the file [/usr/share/tlp/func.d/tlp-func-stat Line 446](https://github.com/linrunner/TLP/blob/1.4.0/func.d/tlp-func-stat#L446) by hand according to the [commit diff](https://github.com/linrunner/TLP/commit/86634789b0807de241f24cab476112af49b29a61#diff-cb855064fc69904f66d1bfa361903a6ff5e65638e241ffe1e7c561de7e14d913) please.

linrunner, 2021-12-14:
Workaround added to the [FAQ](https://linrunner.de/tlp/faq/operation.html#system-freezes-on-wakeup-from-suspend-on-battery).

luckv, 2021-12-14:
I edited my files. Do you want that I test something else for you @linrunner ?

luckv, 2021-12-14:
May be really a bug in the linux kernel??

linrunner, 2021-12-15:
Just for the record: with `AHCI_RUNTIME_PM_ON_BAT=on` suspend/resume works without freeze now? 

> May be really a bug in the linux kernel??

Of course. But we will not solve it here, you will have to create a kernel bug report.

> I edited my files. Do you want that I test something else for you @linrunner ?

I will let you know when the 1.5 beta is ready.

luckv, 2021-12-17:
> Just for the record: with `AHCI_RUNTIME_PM_ON_BAT=on` suspend/resume works without freeze now?
Yes
> Of course. But we will not solve it here, you will have to create a kernel bug report.
Ok
 > I will let you know when the 1.5 beta is ready.
Perfect

linrunner, 2021-12-22:
@luckv : Fedora F35 packages are [available now](https://download.linrunner.de/packages/). Enjoy.

closed 2022-09-25 by linrunner