linrunner/TLP #608: New T61 Lenovo battery doesn't report cycle counts while former one did report

eBug, 2021-12-10:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

I bought my Lenovo T480 with a 01AV490 which accurately reports cycle counts. After installing a new genuine 01AV452 battery (Lenovo sealed package) I get a cycle_count 'not supported'

A clear and concise description of what the bug is.

Both batteries should report cycle counts

A clear and concise description of what you expected to happen.

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? Both
2. Actions to reproduce the behaviour: swap 01AV490 and 01AV452 batteries
3. Shell commands entered and their output

```
# tlp-stat -b
--- TLP 1.3.1 --------------------------------------------

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = active (data, thresholds)
tpacpi-bat = active (recalibrate)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = LGC
/sys/class/power_supply/BAT1/model_name                     = 01AV490
/sys/class/power_supply/BAT1/cycle_count                    =    197
/sys/class/power_supply/BAT1/energy_full_design             =  23940 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  18880 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  17280 [mWh]
/sys/class/power_supply/BAT1/power_now                      =      0 [mW]
/sys/class/power_supply/BAT1/status                         = Unknown

/sys/class/power_supply/BAT1/charge_start_threshold         =     96 [%]
/sys/class/power_supply/BAT1/charge_stop_threshold          =    100 [%]

Charge                                                      =   91.5 [%]
Capacity                                                    =   78.9 [%]

+++ Charge total                                            =   95.7 [%]
```
And for the newer battery

```
+++ ThinkPad Battery Status: BAT1 (Ultrabay / Slice / Replaceable)
/sys/class/power_supply/BAT1/manufacturer                   = SMP
/sys/class/power_supply/BAT1/model_name                     = 01AV452
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  24000 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  24140 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =   6810 [mWh]
/sys/class/power_supply/BAT1/power_now                      =  10297 [mW]
/sys/class/power_supply/BAT1/status                         = Discharging

/sys/class/power_supply/BAT1/charge_start_threshold         =     96 [%]
/sys/class/power_supply/BAT1/charge_stop_threshold          =    100 [%]
tpacpi-bat.BAT1.forceDischarge                              =      0

Charge                                                      =   28.2 [%]
Capacity                                                    =  100.6 [%]

+++ Charge total                                            =   58.8 [%]
```

linrunner, 2021-12-11:
Keep your cool and wait until the battery has one cycle. From then on, the indicator works again.

https://linrunner.de/tlp/faq/battery.html#why-does-tlp-stat-b-display-cycle-count-not-supported

closed 2021-12-11 by linrunner

eBug, 2021-12-11:
Indeed.

```
+++ ThinkPad Battery Status: BAT1 (Ultrabay / Slice / Replaceable)
/sys/class/power_supply/BAT1/manufacturer                   = SMP
/sys/class/power_supply/BAT1/model_name                     = 01AV452
/sys/class/power_supply/BAT1/cycle_count                    =      1
/sys/class/power_supply/BAT1/energy_full_design             =  24000 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  24550 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  24550 [mWh]
/sys/class/power_supply/BAT1/power_now                      =      0 [mW]
/sys/class/power_supply/BAT1/status                         = Full

/sys/class/power_supply/BAT1/charge_start_threshold         =     96 [%]
/sys/class/power_supply/BAT1/charge_stop_threshold          =    100 [%]
tpacpi-bat.BAT1.forceDischarge                              =      0

Charge                                                      =  100.0 [%]
Capacity                                                    =  102.3 [%]
```