linrunner/TLP #609: [TLP 1.5 Beta] Call for testers

linrunner, 2021-12-20:
Hi @all,

as with every upcoming release I need your help to complete the test coverage.

**Test Objectives**
- Testing of new and changed features that interest you → see [release notes](https://github.com/linrunner/TLP/releases) and [changelog](https://github.com/linrunner/TLP/blob/main/changelog)
-  General regression testing (check if everything still works for you)
- Battery care
  -  Testing of charge thresholds for Sony laptops
  -  Regression testing of ThinkPad charge thresholds (check if everything still works for you)
  -  Optional: testing of ThinkPad battery recalibration via the new sysfs attribute *charge_behaviour* - needs a a self patched kernel (see [patches](https://lore.kernel.org/platform-driver-x86/20211123232704.25394-1-linux@weissschuh.net/))

**Beta Packages**
See the [download page](https://download.linrunner.de/packages/).

**What do I need from the testers?**
After installing please start the beta with

    sudo tlp start

I'm expecting your unabbreviated outputs of

    sudo tlp-stat

on battery and AC (via [Gist](https://gist.github.com/) please).

**Bug Reports**
If something does not work as expected → please open an [issue](https://github.com/linrunner/TLP/issues).

Thanks for testing!

BarbUk, 2021-12-21:
Hi !

Thanks for the new release.
[Ac and bat](https://gist.github.com/BarbUk/04db8ca9e963df98af3690ea4a884763).

t-8ch, 2021-12-22:
Tested NFC and `charge_behaviour` on a [X1C9](https://gist.github.com/t-8ch/bf8df4f1a7065e4ffaad58847c766cb1).

The Coreboot T400 natacpi logic was tested when that logic was added.
I'll only be able to test the full beta with the coreboot T400 and a (non-coreboot) T460s early next year.

linrunner, 2021-12-22:
Fedora F35 packages are [available now](https://download.linrunner.de/packages/). Enjoy.

luckv, 2021-12-22:
[Here](https://gist.github.com/luckv/c04c9b8615248422e2360df8015dae1e)

I know you haven't resolved it, but the issue #606 is still present.

linrunner, 2021-12-23:
@luckv : that's reassuring.

linrunner, 2022-01-07:
Beta test is completed and TLP 1.5.0 is [out](https://linrunner.de/tlp/news.html). Have fun!

closed 2022-01-07 by linrunner

linrunner, 2023-07-22:
Hi @ALL : TLP 1.6 Beta 1 is out. Testers wanted -> https://github.com/linrunner/TLP/issues/700

linrunner, 2024-09-05:
Hi @ALL : TLP 1.7 Beta 1 is out. Testers wanted -> https://github.com/linrunner/TLP/issues/760