linrunner/TLP #614: Exceptions from RUNTIME_PM_ON_*="" are not possible atm

Mintulix, 2022-01-26:
TLP 1.5.0

What I want:
Using the Bios defaults for all PCI devices in AC and BAT mode with one exception (Nvidia dGPU should always be in PM-mode, because the Intel iGPU alone is used).

How to accomplish that:

Method 1a (works):
```
RUNTIME_PM_ON_AC="auto"
RUNTIME_PM_ON_BAT="auto"
RUNTIME_PM_DISABLE="..." # exclude all devices from PM that aren't in PM-mode by Bios default except the dGPU -> list 10 devices
```
Method 1b (works):
```
RUNTIME_PM_ON_AC="on"
RUNTIME_PM_ON_BAT="on"
RUNTIME_PM_ENABLE="..." # include all devices that are in PM-mode by Bios default, adding the dGPU -> list 13 devices
```

Method 2 (preferred, much more elegant, but it doesn't work because RUNTIME_PM_ENABLE is ignored):
```
RUNTIME_PM_ON_AC="" # Bios defaults
RUNTIME_PM_ON_BAT="" # Bios defaults
RUNTIME_PM_ENABLE="01:00.0" # always enable PM for the dGPU -> list 1 device
```

The "faulty" code part is in /usr/share/tlp/func.d/05-tlp-func-pm:
```
    if [ -z "$ccontrol" ]; then
        # do nothing if unconfigured
        echo_debug "pm" "set_runtime_pm($1).not_configured"
        return 0
    fi
```
I'm sure it's there for a reason, but for me disabling the "return 0" does the trick.
What do you think? Should method 2 be allowed, or rather why must there be a return if RUNTIME_PM_ON_* is unconfigured?

linrunner, 2022-01-27:
Hi,

indeed a very special use case. Btw it's not BIOS defaults but _kernel_ driver defaults.

`RUNTIME_PM_ON_AC=""`, `RUNTIME_PM_ON_BAT=""` is explicitly meant to disable the feature. I will have to think about whether it is safe to include `RUNTIME_PM_ENABLE`, `RUNTIME_PM_DISABLE` in the check as well. 

In the meantime, you can trick the logic with an invalid value:
```
RUNTIME_PM_ON_AC="nil" 
RUNTIME_PM_ON_BAT="nil" 
RUNTIME_PM_ENABLE="01:00.0"
```

Mintulix, 2022-01-27:
Hi, thanks for your workaround. It works as expected, so you can leave the code unchanged. Perhaps this workaround should be documented?
My idea to leave as many settings unchanged from kernel driver as possible was, on AC the fans were driving me crazy and on BAT I had problems with some components after some time (still testing...).
Again, thank you for this program and your answer, for me this is fixed.

Mintulix, 2022-12-07:
Hey, within one year, what more could one want? ;-)
I didn't test much more since then, because your workaround worked so great. TLP was doing its job in background, so all good.
Tested your changes and it works like before (with the nil-solution), thanks a lot for solving this special use case.

linrunner, 2023-07-22:
Hi @ALL : TLP 1.6 Beta 1 is out and contains the change -> https://github.com/linrunner/TLP/issues/700

closed 2023-07-26 by linrunner