linrunner/TLP #615: System freezes and reboots on Ubuntu running on Lenovo X1 carbon gen 8

bpavacic, 2022-02-01:
I am using version 1.3.1-2  installed from Ubuntu 21.10 .deb package on Lenovo X1 Carbon gen 8.
The battery is kept charged between 75% and 85%, the settings I have set on some previous version of TLP and/or some older kernel.

The problem is that when I issue TLP commands the system freezes for 10 seconds or so and then reboots.

Anyone experiencing the same problems? Please let me know if you need more information to debug this issue.

```
$ uname -a 
Linux thinkpad 5.13.0-28-generic #31-Ubuntu SMP Thu Jan 13 17:41:06 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

$ grep -v "^#" /etc/tlp.conf |grep -v "^$"
START_CHARGE_THRESH_BAT0=75
STOP_CHARGE_THRESH_BAT0=85

$ service tlp status
○ tlp.service - TLP system startup/shutdown
     Loaded: loaded (/lib/systemd/system/tlp.service; enabled; vendor preset: enabled)
     Active: inactive (dead)
       Docs: https://linrunner.de/tlp

$ lsb_release -a 
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 21.10
Release:	21.10
Codename:	impish

```

bpavacic, 2022-02-01:
[image]

Attaching the output of "tlp chargeonce" run as root, seconds before the machine rebooted

linrunner, 2022-02-02:
Duplicate of #599

linrunner, 2022-02-02:
You need a newer version of acpi_call --> https://github.com/nix-community/acpi_call/issues/15

closed 2022-02-02 by linrunner

bpavacic, 2022-02-02:
Thank you.
Upgrading acpi_call from 1.1.0 to 1.2.2 fixed the problem. I have used the instructions [here](https://blog.monosoul.dev/2021/10/16/ubuntu-21-10-and-acpi-call-dkms-bug/)