linrunner/TLP #616: New feature about ignore LTR

biji, 2022-02-03:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I repeatedly ran into [...]

**Describe the solution you'd like**

Hi dev.. probably you can check this script for new feature/option: https://github.com/intel/S0ixSelftestTool
After running the script , my system can reach `Package C3` while screen on, thus saving more power

```
cat /sys/kernel/debug/pmc_core/package_cstate_show
Package C2 : 565365816
Package C3 : 1007659202
Package C6 : 0
Package C7 : 0
Package C8 : 0
Package C9 : 0
Package C10 : 41748004
```

I think it is because ignoring LTR

linrunner, 2022-02-03:
Hi, thanks for your hint.

If i understand correctly, you want me to put in something like

    echo <IP OFFSET>  > /sys/kernel/debug/pmc_core/ltr_ignore

as described in [Linux S0ix Troubleshooting](https://01.org/blogs/qwang59/2020/linux-s0ix-troubleshooting), section "Check device LTR value".

My assessment is that this is really only meant for debugging (and as a help for kernel bug reports), it is not safe for general operation. Therefore I will not implement it.

biji, 2022-02-03:
Using `PCIE_ASPM_ON_BAT=powersupersave` I can get same power consumption about 6.3 W, but system never reach `package C10` when entering suspend mode s2idle/s0ix (screen off, laptop become hot after 30 minutes)

using that parameter: `ltr_ignore` or commenting `PCIE_ASPM_ON_BAT` , system can enter `Packaged C10` without laptop become hot

linrunner, 2022-02-03:
I have already understood that it brings a significant improvement for your laptop. 

Nevertheless it is a "hackish" workaround for bugs in kernel drivers. This has to be fixed in the kernel.

closed 2022-02-03 by linrunner

AnterCreeper, 2022-02-28:
> I have already understood that it brings a significant improvement for your laptop.
> 
> Nevertheless it is a "hackish" workaround for bugs in kernel drivers. This has to be fixed in the kernel.

Can you offer a feature that running custom script?

linrunner, 2022-02-28:
In case of an error, the assignment of the cause is always tricky. I don't have time to analyze with users why their scripts don't produce the desired result. That's why I made the basic decision a long time ago not to support custom scripts.

biji, 2022-03-01:
@AnterCreeper Additional tips, i can get 3.6 W when idle :

* enable_psr
* enable_fbc
* intel_idle.no_acpi=Y
* enable_guc

* echo powersupersave > /sys/module/pcie_aspm/parameters/policy

when using discrete nvidia:
* options nvidia NVreg_PreserveVideoMemoryAllocations=0
* options nvidia NVreg_EnableS0ixPowerManagement=1

Try ignore all LTR, or better try one by one, my script on boot:

```
echo 8 >  /sys/kernel/debug/pmc_core/ltr_ignore
```