linrunner/TLP #618: Improve UDEV rule performance

benzea, 2022-02-16:
Hi, I was looking at libusb#237 and noticed that TLP is a huge offender with regard to USB device processing in udev.

The TLP udev rules need an extremely long time to process events. This is not really necessary, as there is no reason to block the rest of udev (and userspace!) in order to possibly enable USB autosuspend.

The obvious reason for it to go really horribly wrong is the sleep of 500ms in the codebase, see https://github.com/linrunner/TLP/blob/main/func.d/20-tlp-func-usb#L64. This sleep is probably just plain wrong, if subdevices are of such importance, you can also modify the parent device from a `DEVTYPE=usb_interface` rule.

But, equally bad is the fact that we even wait for this operation to complete. I am thinking the following solutions are possible (roughly in preferred order in my view):
 1. Have `udev` rules that do not require executing a script. You can test for file existence in udev, so writing out a file to e.g. `/run/tlp` and testing for that you can enable/disable rules. Not sure if one needs a script though, as autosuspend needs to be set on the parent device.
 2. Have a tiny daemon that monitors udev and reacts to the events.
 3. Push the script into a systemd (template) unit. i.e. add `TAG+="systemd"` and set `ENV{SYSTEMD_WANTS}+="tlp-usb-udev@.service"`. The service should be `Type=oneshot` with `ExecStart=/lib/udev/tlp-usb-udev usb %I` or something similar (did not test it).
    * To avoid the sleep, one could trigger it for each interface update. But, that is kind of ugly as it would create a *lot* of `.device` units.
    * I expect `udevadm settle` is enough though, so possibly just an `ExecStartPre-=/usr/sbin/udevadm settle`.
 4. Remove the sleep and run for `usb_interface` and update the parent; but that still means loading all the shell scripting.

To be honest, at the end of the day this might be sanest if it was just a small daemon that reacts to udev events.

linrunner, 2022-11-28:
Hi, 

thanks for your insights. I have (obviously) thought about your comments for a quite a time.

1) Sounds very complicated and I don't have the same control over what happens as in a script. Logging is even more difficult.
2) I would like to avoid implementing a daemon as a shell script.
3) This would allow TLP to be used only with systemd. 
4) Would mean backtracking down the whole tree searching for the parent multiple times, this also takes time.

I openly confess that my knowledge of the subject leaves a lot to be desired.

benzea, 2023-10-20:
Or maybe it would be enough to fork into the background so that udev can continue to process rules?

linrunner, 2023-10-30:
The udev manpage says:

> Starting daemons or other long-running processes is not allowed; the forked processes, detached or not, will be unconditionally > killed after the event handling has finished. 

I think I'll just give 4. a try as soon as I have time. The "backtracking" includes only one step to the direct parent.