linrunner/TLP #619: Ability to mute audio inputs and outputs on startup

drws, 2022-02-16:
**Is your feature request related to a problem? Please describe.**

Since muting audio channels [can save power](https://github.com/linrunner/TLP/issues/159#issuecomment-139293236), I propose "mute all audio inputs on startup" and "mute all audio outputs on startup" features for the TLP.

What is not completely in-scope of the project is the increased privacy and convenience such options also provide. I prefer that my machine boots silently since you never know when you need it. For additional convenince I propose that volume-setting options are considered to be included as well. While not strictly power saving-related, there might be some sense to adding this to TLP, similar to radio device management. I also assume that kernel power savings are activated for a muted audio channel even if its volume is non-zero. If I'm mistaken, this additional proposal makes no sense and can be skipped.

**Describe the solution you'd like**

Ability to mute and/or set volume of audio inputs and/or outputs on startup, similar to radio device management. Configuration-wise I'm thinking about 2+2 additional options such as:

    AUDIO_INPUT(S)_MUTE_ON_STARTUP=N|Y
    AUDIO_OUTPUT(S)_MUTE_ON_STARTUP=N|Y
    AUDIO_INPUT(S)_VOLUME_ON_STARTUP=100
    AUDIO_OUTPUT(S)_VOLUME_ON_STARTUP=100

**Describe alternatives you've considered**

This can be of course scripted manually with different tools available such as `pactl` but it could be a welcome addition to TLP.

linrunner, 2022-02-21:
Hi, thanks for your suggestion.

The tools available for audio mute in the various distributions and desktop environments are anything but uniform. An implementation would mean taking the entire environment into account. Who is supposed to implement and test these countless dependencies? I will not integrate something like that. TLP is intentionally implemented without dependencies to the desktop.

closed 2022-02-21 by linrunner

drws, 2022-02-22:
I understand what you mean. In fact, the diversity of audio-control tools lead me to suggest the feature for TLP. Since it is an I/O-related suggestion, I was thinking about talking to ALSA directly, bypassing higher audio routing layers, from deprecated JACK to shiny new PipeWire. In this way, TLP would only need to know a bit of ALSA to iterate through all available hardware inputs and outputs and set the volume and/or mute each one, just as it needs to know a bit of ACPI for example.

The funny part is that in all the voluminous software diversity around Linux audio, I can't find a simple tool that would be able to autonomously "mute everything".

Just wanted to put down a bit more developed idea in case someone stumbles upon it.