linrunner/TLP #622: CPU_ENERGY_PERF_POLICY is not picked up both default and configured

krykra7, 2022-03-05:
As i described in title despite the default values or values i set both in `etc/tlp.d/00-template.conf` and `/etc/tlp.conf` are not applied and `energy_performance_preference` stays stuck with `performance` mode.

**Expected behavior**

Expected behavior is that tlp would pick up those settings and change `energy_performance_preference`.

**To Reproduce**

I just started tlp with `tlp start` and my processor was operating at maximum frequency, despite there was 0 load. So just start tlp, according to documentation (https://linrunner.de/tlp/settings/processor.html) performance policy should be either `balance_performance` or `balance_power` but it is `performance` on AC and BAT.

Commands performed: `tlp start`, `tlp-stat` - output in gist. Also i reinstalled and installed tlp again and it didn't help.
tlp-state output: https://gist.github.com/krykra7/50f6ebad64b9071e037ec6553569202e

linrunner, 2022-03-05:
Hi,

your tlp-stat output indicates that you didn't configure anything at all: all settings are default, no mention of **/etc/tlp.conf**. I guess you missed to remove the leading '#'. But this is not the cause of your problem, because the TLP defaults do not work either.

The reason seems rather that the performance governor is active, which maximizes the frequencies:

 > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = **performance**

I don't know if this is the default with the Pop!_OS kernel or if some other Pop!_OS tool is at work. 

I'd try to activate the powersave governor via TLP:

    CPU_SCALING_GOVERNOR_ON_AC=powersave
    CPU_SCALING_GOVERNOR_ON_BAT=powersave

krykra7, 2022-03-05:
> Hi,
>
> your tlp-stat output indicates that you didn't configure anything at all: all settings are default, no mention of **/etc/tlp.conf**. I guess you missed to remove the leading '#'. But this is [...]

About custom configuration i just tried it and then cleared it, that's why you can't see it. I've tried you suggestion and now with `powersave` governor `energy_performance_preference` changed to `balance_performance` on AC and `balance_power` on BAT which is what i wanted, also frequencies are normal as before. I've been using TLP on pop for about 8 months now, and it was today i noticed that something changed with frequencies, maybe it's something with latest kernel updates. I'm not that experienced with linux to check that. 

Still, i'm not sure whether I should treat problem of not working defaults as resolved or not.

Another question if you could answer it, with `powersave` governor does cpu still has ability to raise performance on demand?

linrunner, 2022-03-05:
> About custom configuration i just tried it and then cleared it, that's why you can't see it.

Your claim includes that the custom setting does not work. Then you should show it completely. I can't give support for things I don't see. Please keep that in mind next time.

> Still, i'm not sure whether I should treat problem of not working defaults as resolved or not.

What you observe is not an issue with TLP. It is the performance governor working as designed.

> Another question if you could answer it, with `powersave` governor does cpu still has ability to raise performance on demand?

Yes, of course. _powersave_ is the default of most distribution kernels.

closed 2022-03-05 by linrunner

aacebedo, 2023-07-09:
Hi, is there an ETA for the 1.6 release?  I read the amd_pstate active mode is going to be the default one in the upcoming kernels so it will be more widely used and I saw that the current version stick with "performance" profile.