linrunner/TLP #623: Thinkpad E590 BAT0 in a not charging/waiting to discharge state.

EmilZahariev, 2022-03-06:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

Battery does not charge to the selected stop threshold. The tlp-stat -b output lists it as 
`/sys/class/power_supply/BAT0/status= Not charging`

Further, the panel applet, and neofetch, lists the battery as Waiting to discharge

**Expected behavior**

When the stop threshold was reached tlp-stat -b should list
`/sys/class/power_supply/BAT0/status= Full`

The panel applet and neofetch should return full as well

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? 
Answer: on AC
2. Actions to reproduce the behaviour 
Answer: Remove thresholds in tlp.conf, activate with tlp start and fullcharge.
3. Shell commands entered and their output
```
sudo tlp start
TLP started in AC mode (auto).

sudo tlp setcharge
no output

sudo tlp fullcharge
Setting temporary charge thresholds for BAT0:
  start =  96 (no change)
  stop  = 100 (no change)
Charging starts now, keep AC connected.

sudo tlp-stat -b   
--- TLP 1.5.0 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds)
* tpacpi-bat (acpi_call)  = active (recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 01AV445
/sys/class/power_supply/BAT0/cycle_count                    =     31
/sys/class/power_supply/BAT0/energy_full_design             =  45000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  42590 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  42580 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Not charging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     96 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =  100.0 [%]
Capacity                                                    =   94.6 [%]
```

5. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)

[tlp-stat output](https://gist.github.com/EmilZahariev/a3581b650718ea6ce1e85aa0338f913f)
[tlp.conf](https://gist.github.com/EmilZahariev/e828a1b719ecf6fa5f18629e5d06c80a)

**Additional context**

```
OS: Arch Linux x86_64 
Host: 20NB006WBM ThinkPad E590
Kernel: 5.16.11-arch1-2
DE: Xfce 4.16
WM: Xfwm4 
CPU: Intel i7-8565U (8) @ 4.600GHz
GPU: Intel WhiskeyLake-U GT2 [UHD Graphics 620] 
GPU: AMD ATI Radeon 540/540X/550/550X / RX 540X/550/550X
```
The battery only reaches the charging threshold and goes to the full state only after I recalibrate/fullcharge it. After that the problem described above happens. I have also removed the charging thresholds in the conf and issued a fullcharge. The BIOS was updated from pcsupport.lenovo.com yesterday

UEFI BIOS Version: ROYET48W (1.31)
Embedded Controller Version: ROYHT48W (1.23)

I also tested with several live USBs to ensure that there are not any other conflicting OS settings, but the problem persists.

Further to that, I ran a full diagnostic on the machine to ensure that there are no physical issues. The diagnostic confirmed that everything is fine with the hardware. I also sent the machine to the lenovo support service and even they confirmed that there are no issues with the machine. Also, I tested it with a different charger, this had no effects, the issue persists.

linrunner, 2022-03-06:
Hi, 

there is no problem at all, certainly not in TLP. The value **/sys/class/power_supply/BAT0/status** comes directly from the kernel. You will have to put up with it.

closed 2022-03-06 by linrunner

linrunner, 2022-03-06:
"Not charging" is, by the way, the state value that signals that a charging threshold takes effect on ThinkPads (the value was only recently added to the kernel). If the charge level is minimally below 100%, e.g. 99.99%, then the default start threshold of 96% takes effect.